BigBobbas/ESP32-S3-Box3-Custom-ESPHome #3: No sound/microphone

styphonthal, 2024-06-07:
I have no sound or microphone. I tried to comment out the areas mentioned in the guide, but that causes a whole arrange of other errors.

BigBobbas, 2024-06-08:
> I have no sound or microphone. I tried to comment out the areas mentioned in the guide, but that causes a whole arrange of other errors.

Please try the installation from the updated instructions. https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/instructions/installation%20guide.md

styphonthal, 2024-06-10:
I deleted the home assistant and esphome devices, then reinstalled it with above. Everything works great now

BigBobbas, 2024-06-10:
Great news, thanks for the update. I will hopefully be releasing an update later , which provides a switch to turn on or off a screen saver, which then dims the display, displays the time and temp, then after 3 mins the screen turns off. Wakes on touch or voice.  This can be turned on off from the device.

closed 2024-06-11 by BigBobbas