BigBobbas/ESP32-S3-Box3-Custom-ESPHome #4: Feature Request: Add Display Activation and Deactivation Based on User Interaction and Idle Period

scholzri, 2024-06-07:
Introduce a feature to automatically enable the display when the red button is touched or the wakeword is detected. Additionally, implement an automatic display deactivation after a predefined idle period.

**Display Activation:**
- Red Button: Enable the display immediately when the red button is touched.
- Wakeword Detection: Enable the display upon detecting a predefined wakeword.

**Display Deactivation:**
- Automatically disable the display after a specified period of inactivity or idle time to conserve power and to make the appearance of the device a little more subtle.

Thank you for considering this feature request!

BigBobbas, 2024-06-08:
I am releasing a variation of the config soon , which will include a dim 'screen saver' page displaying time and temp. This will timeout to a blank screen after x mins. I am still tidying up instructions and everything else, so it will be a few days before it is ready. I may even be able to incorporate it so that it can be enabled and disabled with a button from the device.

BigBobbas, 2024-06-10:
I have made an update to include this feature, this can be turned on and off from the 'info' page. I will work on user being able to select a timeout in the next update. This defaults to display time & temp after 30s and then after 3 mins screen turns off.
The ready to use config is s3b.yaml - cheers

youkorr, 2024-06-10:
bonjour pour la temperature peut on le remplacer par autre thermometre merci est comment??

BigBobbas, 2024-06-10:
> ![times sensor](https://private-user-images.githubusercontent.com/96953496/338285139-7e338ba0-d78f-474e-830f-986c30e5e984.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgwNTMwODMsIm5iZiI6MTcxODA1Mjc4MywicGF0aCI6Ii85Njk1MzQ5Ni8zMzgyODUxMzktN2UzMzhiYTAtZDc4Zi00NzRlLTgzMGYtOTg2YzMwZTVlOTg0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEwVDIwNTMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwYjFhM2UyMjUyNWFkMGMwMTc3OGRhY2QwNTFhMDhhYWNlMDVlYmQwYTkyZmExYjczN2YyNzkxMWJiN2UzOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bSN0uBbQWHtox7sakWlx-K36K0_ylFDeYE-b5GCQLt0) hello for the temperature can we replace it with another thermometer thank you how is it??

you would need to add a - homeassistant: sensor and then change this line - 1155 (or near)<br><br>

[image]<br>
to the id that you give to the -homeassistant sensor.

youkorr, 2024-06-10:
Thank you for your answer but as I start is seen your work on this firmware which is beautiful if I understand this you say:
it.printf (160, 180, id (my_fontseg2), light_blue, textalign :: center, "%.F ° C", id (s3temp) .state);

 by that of my temperature sensor:

it.printf (160, 180, id (my_fontseg2), light_blue, textalign :: center, "%.F ° C", id (sensor.temp_entree_temperature) .state);

But I end up with a compilation error need help thank you for the novice :)

youkorr, 2024-06-10:
Ok I had misread it was necessary to last this line:
You would need to add a - Homeassistant: Sensor
 - Platform: Homeassistant
 ID: Liv_temp
 entity_id: sensor.living_room_temp
 Internal: True

BigBobbas, 2024-06-10:
you would change to 
```yaml
it.printf (160, 180, id (my_fontseg2), light_blue, textalign :: center, "%.F ° C", id (liv_temp) .state);
```
use lower case for id: 's

youkorr, 2024-06-10:
it.printf(160, 180, id(my_fontseg2), light_blue, TextAlign::CENTER, "%.f°C", id(liv_temp).state);

Ok thank you it works

youkorr, 2024-06-10:
For the other functions, taking, air conditioner how they have them id's
[image]
[image]

BigBobbas, 2024-06-11:
follow the instruction [here](<https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/instructions/make%20it%20your%20own.md>)

closed 2024-06-11 by BigBobbas