BigBobbas/ESP32-S3-Box3-Custom-ESPHome #20: put a digital alarm or door??

youkorr, 2024-06-20:
[image]

Lin4fun, 2024-06-21:
> ![Capture d’écran 2024-06-20 235812](https://private-user-images.githubusercontent.com/96953496/341576096-3ea5ce2e-4735-4949-bbfe-8ace4b1948aa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5MjkzNzYsIm5iZiI6MTcxODkyOTA3NiwicGF0aCI6Ii85Njk1MzQ5Ni8zNDE1NzYwOTYtM2VhNWNlMmUtNDczNS00OTQ5LWJiZmUtOGFjZTRiMTk0OGFhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIxVDAwMTc1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE2MGQwZjg4NDIzMDEzOGMyNWQ1NjY1NGM5MmMyMDg4ODFhODM1ZDJiOTBmY2JlMzI0NWQ3MTY2OGI3Y2FjMTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QUo7zuzOvU3x2dqFYpizE700OJAHIv-LZ7A1zkpT64U)

This is no issue, is there in the documentation anywhere written that this configuration is for creating an alarm code panel?!

Close this  again and may add as suggestion in the discussion.

youkorr, 2024-06-21:
I wonder if it was possible to postpone my alarm sign in the box and it was just a suggestion