BigBobbas/ESP32-S3-Box3-Custom-ESPHome #25: Component touchscreen is marked FAILED after flashing latest s3b.yaml

krbrowning, 2024-06-23:
[E][component:082]:   Component touchscreen is marked FAILED

https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/commit/3dc18eb9ea1573a469541306ea1ba102f2f6de83

flashing results in [E][component:082]:   Component touchscreen is marked FAILED error
reverting to previous commit fixes the issue.

https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/commit/54651b9127d9f32747df63f0f7a6df2966252a55

BigBobbas, 2024-06-23:
thank you for letting me know, i will take a look into this 👍

krbrowning, 2024-06-23:
the same error happened to me when I tried to add a night mode switch

switch:
  - platform: template
    name: "Night Mode"
    id: night_mode_switch
    optimistic: true
    restore_mode: RESTORE_DEFAULT_OFF
    turn_on_action:
      - logger.log: "Night Mode ON"
      - light.turn_on:
          id: led
          brightness: 10%
    turn_off_action:
      - logger.log: "Night Mode OFF"
      - light.turn_on:
          id: led
          brightness: 100%

BigBobbas, 2024-06-23:
i've just tested with s3b.yaml and i don't get any errors and all is working ok. Had you made any edits to the config ? or was this after adding the above switch ?

krbrowning, 2024-06-23:
I had made the above changes, had the same error. pasted the newest s3b.yaml into the device yaml  ha -> esphome integration -> edit window -> paste -> save -> install. got the same error, moved to the previous commit of s3b.yaml and tried again, no error. I'll try reproducing it again.

krbrowning, 2024-06-23:
same error with latest s3b.yaml, including log file
[logs_esp32-s3box-4_run.txt](https://github.com/user-attachments/files/15946090/logs_esp32-s3box-4_run.txt)
[esp32-s3box-4(yaml).txt](https://github.com/user-attachments/files/15946095/esp32-s3box-4.yaml.txt)
using the previous commit seems to fix the error
[logs_esp32-s3box-4_run (1).txt](https://github.com/user-attachments/files/15946100/logs_esp32-s3box-4_run.1.txt)
[esp32-s3box-4yaml2.txt](https://github.com/user-attachments/files/15946101/esp32-s3box-4yaml2.txt)
could completely be something on my end, is this the correct way to flash?

youkorr, 2024-06-23:
Good evening I saw your comment me when small problems I restore it with the expressive application and then I flash it with esphome all checking the build to see if you don't have a compilation error

youkorr, 2024-06-23:
If you help with pleasure

BigBobbas, 2024-06-24:
> same error with latest s3b.yaml, including log file [logs_esp32-s3box-4_run.txt](https://github.com/user-attachments/files/15946090/logs_esp32-s3box-4_run.txt) [...]

you can flash manually by copying and pasting in the yaml to a device card in the ESPHome Add-on dashboard and save and install or you can use the web installer [here](<https://support.bbdl.co.uk>). It would be beneficial to delete any existing cards before starting the process, so that a new card can be adopted once initial setup is complete.  Also remove any instances of the device from the ESPHome integration in HA>>settings>>device & services>> esphome . Using the web installer will prompt for you to add the device to the dashboard.

BigBobbas, 2024-06-25:
With regards to making a 'night mode' for the screen saver, because the backlight is controlled by a script, you could add the switch without the actions and then add the switch state as a condition to the radar sensor automation. IF you need help doing this I am glad to help. 
I was thinking of splitting the info page up and creating a settings page , with screen  saver settings. I could include the switch in there . This could also be associated with a time schedule so that the action would only happen between certain hours . This would save having to remember to enable each night.
This will take me a little time to complete, but I will try and add it to the next update.

joelis10, 2024-06-27:
happens to me too, should be fixed with a press of the reset button (at least for now, if it is a bug)

closed 2024-07-18 by krbrowning