BigBobbas/ESP32-S3-Box3-Custom-ESPHome #27: ChatGPT's response ends early. Not possible to pronounce.

unreality81, 2024-06-27:
Hello, First of all, this is a great project of yours. I use an ESP32 S3 Box 3 with ChatGPT and I noticed that he can't finish his sentences before the dashboard is shown to me again and the Chatgpt announcement is also ended early. Can you set a different timeout here?

BigBobbas, 2024-06-27:
Hi there, thank you for trying out this project and for giving feedback on your issue. I haven't personally tested with ChatGPT but will see if we can figure it out. 
Can I ask what kind of length of time does it respond for before cutting off ? also if possible if you could get device logs covering the time from before, during and after an example event. You can view the logs from the ESPHome dashboard and they will be shown in realtime. This will help me to see what is happening on the device during that time.
Many thanks 
Rob

unreality81, 2024-06-27:
[logs_esp32-s3box-3_logs (1).txt](https://github.com/user-attachments/files/16009273/logs_esp32-s3box-3_logs.1.txt)

Thank you for your quick answer :) Here is the log file! I asked for a roll recipe, after about 5 seconds the dashboard appeared again. The announcement stops at around 18 seconds! However, this has also happened at 15 seconds. I do not know why.

BigBobbas, 2024-06-27:
Thanks for the logs and info, if you open this url in a browser does it play the full response http://192.168.2.154:8123/api/tts_proxy/3090767fd7bcbc2fdba80d15c3aad451042b1b0b_de-de_70263bfdba_tts.home_assistant_cloud.mp3

unreality81, 2024-06-27:
Yes, the announcement will be played in full.

unreality81, 2024-06-27:
Hello, the display conversation now remains! That's pretty cool but unfortunately the announcement keeps stopping :/

gnumpi, 2024-07-03:
It's probably the timeout which stops other elements in the pipeline after the first element has finished playing/streaming. Depending on the mp3 bitrate and chosen buffer size it might happen that the time between http stream end and playing all buffered content needs to be longer than this timeout. I will check this...

gnumpi, 2024-07-03:
I uploaded a fix which should resolve the buffer size vs timeout mismatch. Can you check if this solves your issue?
(note: the branch name has a typo, sorry for that)
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: fix_buffer_size_tiemout_mismatch
    components: [ adf_pipeline, i2s_audio ]
    refresh: 0s
```

unreality81, 2024-07-05:
Hello, sorry, I've been working a lot lately. I'll test it today and let you know if it solves my problem. Thank you for that

unreality81, 2024-07-05:
It worked wonderfully. Thank you very much for that

gnumpi, 2024-07-05:
good to hear, thx for reporting and testing!

BigBobbas, 2024-07-05:
thanks for sorting that Mischa! 🙏

closed 2024-07-05 by BigBobbas

mikey60, 2024-07-10:
That also solved my Ollama local LLM response ending early.  Thanks!