BigBobbas/ESP32-S3-Box3-Custom-ESPHome #30: Help with compiling from windows

jolbi, 2024-06-28:
Hi,

Thank you for this great project! I am trying to compile your esphome code on windows because it fails on my 4GB RAM HA instance. This is my first time trying this, and I would appreciate some help.

I used the web installer, connected esp-box to wifi and added it to home assistant. I installed esphome on windows with pip, downloaded the esp32-s3box-3.yaml to windows and ran `esphome config esp32-s3box-3.yaml` .
I changed the wifi !secret to plain text, because it does not have access to the secrets on windows. This can be changed after, right?

The compiler starts but stops with error (last 7 lines:
```
  File "C:\Users\TimG\AppData\Local\Programs\Python\Python311\Lib\pathlib.py", line 1067, in write_bytes
    with self.open(mode='wb') as f:
         ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TimG\AppData\Local\Programs\Python\Python311\Lib\pathlib.py", line 1044, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: '.esphome\\font\\c9ee33c7\\font.ttf'
```
The file is there and the path is correct. I am running cmd as administrator. Do you have some ideas about what could be wrong?

Thanks in advance.

BigBobbas, 2024-06-28:
Hi there , 
the file you need to compile is this one https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b.yaml
you can add a secrets.yaml file to the same directory that you are running esphome from and it should include the wifi creds for you from the secrets.yaml.
Secrets.yaml should contain the following.
```yaml
wifi_ssid: "your_ssid"
wifi_password: "your_password"
```
from the command line if you use `esphome run s3b.yaml` it will then compile and start the upload too. 
I hope that helps
Cheers 
Rob

jolbi, 2024-06-28:
Thanks for lighting fast reply! I copied my secrets yaml to Windows and this now works. I tried your command with the yaml from your repo, but I get the same error. Seems more like there is a problem with the esphome python code or dependencies or some Windows-specific problem. I'll try later the same thing on linux, I hope that will work...

BigBobbas, 2024-06-28:
i run esphome cli from linux in windows (wsl) and it works well, running directly in windows can be a bit 'flaky' with dependencies etc so linux is a good option 👍 good luck and if you still have probs, let us know and we can see what we can do

jolbi, 2024-06-28:
I compiled it on a NUC running Debian and it works! (I removed wsl from my pc recently because it took too much space... But that would probably be faster - on my old NUC it took >20 min).
Thank you for the help!

The only thing I couldn't figure out was uploading binary through OTA - It tried to upload to some other IP address. I then flashed it via USB port and it worked. I found out after, that the IP address of the device is specified in the yaml... Just putting this here if someone else has a similar problem. :)

I tested it a bit and something odd is happening with voice. If I use Microsoft Azure TTS the assistant has a high-pitched voice. Doesn't matter the language or voice I choose. If I use Google's TTS the voice sounds normal. This was not the case with the original ESPhome voice assistant. I don't have piper currently running, so I cannot test that one. Sounds like the sound sent to esp-box is in "different format". Has anyone else noticed something similar?

BigBobbas, 2024-06-28:
great stuff... glad you have managed to get it up and running . 
My BAD - i have to use the `use_address` line when testing because i've got multiple devices stuff gets confusing lol ... so that shouldn't be there. i will get that altered. it should just resolve to the device name.

you can change the sample rate in the config which will alter the playback speed of tts shown below: just need to match the sample rate of the service being used. 
[image]

jolbi, 2024-06-29:
I changed the sample rate to 42KHz and now sounds great.
Thank you again for all the help! I am closing the issue.

closed 2024-06-29 by jolbi

BigBobbas, 2024-06-29:
Brilliant , glad you have it all sorted and thank you for your support and for trying out this project.