BigBobbas/ESP32-S3-Box3-Custom-ESPHome #38: sensor config error in display lambda

BigBobbas, 2024-07-08:
after having done several tests nothing is conclusive with this code
 if(id(tv).state == "on"){
 it.printf(120, 75, id(icon_font_80), yellow,"\U000F07F4");
 }

esphome gives me a lot of errors
with this code one can modify = 'on'

 if(id(tv).state == 'on'){
 it.printf(120, 75, id(icon_font_80), yellow,"\U000F07F4");
 }

the compilation is done well but the icon surely appears something wrong

_Originally posted by @youkorr in https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/33#discussioncomment-9982040_

BigBobbas, 2024-07-08:
@youkorr can you paste your full sensor block for this sensor please?  I'm assuming you have this setup under `text_sensor:`
thank you

youkorr, 2024-07-08:
- platform: homeassistant
    id: living_room_state
    entity_id: switch.prise_tv_mss315_main_channel
    internal: true

youkorr, 2024-07-08:
I didn't put anything on text sensor

youkorr, 2024-07-08:
[image]

youkorr, 2024-07-08:
I haven't found how to switch to other entities: tv, microwave, lighting, etc.

BigBobbas, 2024-07-08:
you could make the time smaller and then fit the icons all on the saver page of the entities that are on... i'm not really sure what you want to do

youkorr, 2024-07-08:
if(id(microwave).state == "on"){ 
         it.printf(125, 116, id(icon_font_80), yellow,"\U000F0C99");
         }   
         if(id(shield).state == "on"){ 
         it.printf(125, 116, id(icon_font_80), yellow,"\U000F09B1");
         }   
         if(id(alarm_armed).state == "on"){ 
         it.printf(125, 116, id(icon_font_80), yellow,"\U000F099D");
         }

youkorr, 2024-07-08:
thank you for your response if I reduce the icon it will affect the other icons in the firmware

[image]

youkorr, 2024-07-08:
I wanted a view of the entities activated in the screensaver

BigBobbas, 2024-07-08:
you can copy and paste a font block and change the id: of the font and the size, and then use that. As long as each font has an id: then you just use the id: of the font with the size you want.

BigBobbas, 2024-07-11:
have you managed to get this working now? I am looking into the issue with screen dimming now and others.and will update in the relevant post.

youkorr, 2024-07-11:
Sorry I was very busy on another Google nest hub Max project, yes I managed to get it working thank you very much for your advice

closed 2024-07-12 by BigBobbas