BigBobbas/ESP32-S3-Box3-Custom-ESPHome #41: Wake-up by Presence sensor or Wake-up Word does not work

Tobster77, 2024-07-17:
Hi,

thanks for providing this useful piece of software!

I am assuming the following logics:
I would assume that after some inactivity, the screen saver should get active. If chosen, the screen should go black.
Now the device should wake up when the wake-up word is received, or if the presence sensor is triggered. After some time, the screen would enter the screensaver again and optionally go black again.

Can you confirm this is the expected behaviour? Because I observed that the screen goes black, but it is not waking up again. Note, the box is still executing the programm, so it's is not crashing.

Thanks and best wishes,
Tobias

BigBobbas, 2024-07-17:
Hi ,
I will be adding an updated config later today which should resolve this and also include some new features.  I will add an announcement when this has been finished and available.

Many Thanks

Rob

BigBobbas, 2024-07-19:
Hi , has the update now resolved this issue?
Cheers
Rob

Tobster77, 2024-07-21:
Dear Rob,

unfortunately not. The screen stays black once it is disabled. Neither the presence sensor nor the wake up word enable it again.

Note, the screensaver is disabled, I directly let the screen go black. Maybe you can check if this can be reproduced on your side?

Thanks,
Tobias

BigBobbas, 2024-07-21:
Hi Tobias,
Thanks for getting back to me, I think I know what the problem will be, I hadn't considered having the screen set to go straight to a blank screen  rather than having the saver show first and then after 3mins the screen goes blank. So this will definately be an issue with the config and I will just need to change the screen saver script or wake screen trigger to enable this option. 
Thank you for pointing this out. Feedback is essential for making sure all eventualities are covered. 🙏 
This should be a relatively easy fix (fingers crossed) I will take a quick look into it this morning 👍 

All the best

Rob

BigBobbas, 2024-07-24:
Hi, I have had a look at this and it is a little more complex than I first thought and will require a few changes in order to have the device set to just go to blank screen, due to how the scripts work.
If you enable screen saver and timeout to blank, the time will be displayed for 30seconds, however you can change this for your use case by editing the following lines. These are under the `script:` section.
change 1. to 50ms
change 2. to However long you want before the screen goes blank.
change 3. to 50ms
you can play around with the delay times to tweak to suit your needs.
[image]

I hope this helps, please let me know if you still have issues.  

Best Wishes
Rob

Tobster77, 2024-07-24:
Thanks a lot; I'll check in four weeks after my vacations!

BigBobbas, 2024-07-24:
I am just making some changes today and have managed to now include a page on the device (also available from the integration in HA) 
to change the timeouts from the device, so no having to change code. If you are away for 4 weeks (Hope you have a great time) 
There will be an update waiting with some more new features when you return👍

BigBobbas, 2024-07-27:
timeout config on device now available and due to be released

closed 2024-07-27 by BigBobbas