BigBobbas/ESP32-S3-Box3-Custom-ESPHome #45: Can't connect to Home Assistant

Someguitarist, 2024-07-17:
So I was able to connect to the wifi, but I'm on the device setup page and don't know how to add it to HA.  If I click the link on the firmware installer, it goes to the ESPHome integration and asks for an IP.  I enter the IP on the screen and leave the port at the default 6053 and it doesn't connect.  I'm also unable to manually connect.

Am I missing something here?

Thanks.

pepe59, 2024-07-17:
Are you sure the IP is correct?
It happened to me during the first compilation that the IP remained default and I had to change the wifi connection again.

Someguitarist, 2024-07-17:
Awesome, thanks for the edit.  I'll give it some time and see what happens!

Someguitarist, 2024-07-17:
Okay, so first problem was that the installer couldn't install over ESPHome because it was using SSL.  Now I'm getting this message when trying to install.

INFO ESPHome 2024.7.0
INFO Reading configuration /config/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Detected timezone 'Etc/UTC'
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO46 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

micro_wake_word: [source /config/esp32-s3box-3.yaml:453]
  
  'models' is a required option for [micro_wake_word].
  
  The model parameter has moved to be a list element under the models parameter.
  model: hey_jarvis
  on_wake_word_detected: 
    - voice_assistant.start: 
        wake_word: !lambda |-
          return wake_word;

BigBobbas, 2024-07-17:
Thank you for pointing this out. I will get this fixed in the next couple of minutes 👍

BigBobbas, 2024-07-17:
this should now be fixed. 
If you already have the config in the dashboard you can change the following lines:
from:
```yaml
micro_wake_word:
  model: ${micro_wake_word_model}
```
to
```yaml
micro_wake_word:
  vad:
  models: 
    - model: ${micro_wake_word_model}
  ```

BigBobbas, 2024-07-19:
Hi , has this now resolved the issue ? @Someguitarist 👍

BigBobbas, 2024-07-27:
mmw2 now implemented in all configs

closed 2024-07-27 by BigBobbas