BigBobbas/ESP32-S3-Box3-Custom-ESPHome #47: Error compiling on latest -no-sensor

Kelerion-UK, 2024-07-17:
First off, thanks for remembering the no-sensor update this evening..  its well appreciated and keep up the great work!

after the midnight update after the esphome update today.. (33 mins ago)

I get this when compiling: 

INFO ESPHome 2024.7.0
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "/config/esphome/esp32-s3box-3.yaml", line 681, column 7
expected <block end>, but found '<block mapping start>'
  in "/config/esphome/esp32-s3box-3.yaml", line 943, column 8

I've done a clean rebuild.. still the same...

Thanks.

Lin4fun, 2024-07-17:
No error here on three devices. Delete the devices from esphome dashboard to clean the config and adopt them again, or compare the lines mentioned in the error message..

Kelerion-UK, 2024-07-18:
which are you using? the -dev one from 6 hours ago? or the -main one from an hour ago? (and that looks the wrong way around to me too)

BigBobbas, 2024-07-18:
this will be a yaml error. I have just done a straight copy and paste and overwritten the original config and it compiles ok.
there will most likely  be a `,` missing somewhere between the lines mentioned or an additional `"` if you can't find it, paste in that block and I can take a look 👍

BigBobbas, 2024-07-18:
Just realised that this may have been the config from main , which had an indent error, that is fixed. the only reason for the update in main was to add the support for microwakeword as per the breaking change.
The config with the latest features will always be the one in the 'Dev' branch. This will not get merged until it is a 'complete' config and confirmed to be working by you guys :)

BigBobbas, 2024-07-19:
is this working ok now ?
Cheers 
Rob

Kelerion-UK, 2024-07-20:
ooops.. forgot to close this! all working now! lol

closed 2024-07-20 by Kelerion-UK

BigBobbas, 2024-07-20:
cheers buddy 👍 just trying to keep on top of everything 🙃