BigBobbas/ESP32-S3-Box3-Custom-ESPHome #48: Audio Playback for Elevenlabs TTS very slow

cl0ud6uru, 2024-07-18:
I'm using elevenlabs for my TTS service and when it plays back on the box speakers it's slow and very low pitched do to the slowness. It plays fine on an external media player. Home assistant cloud TTS sounds fine. 
This issue happens on two different devices. 
I understand this may be out of scope, but has anyone else encountered issues like this?

Thanks!!

BigBobbas, 2024-07-18:
hi there,
 you can change the sample rate in the following block to match what the tts service you are using uses.
Hope this helps 👍 
[image]

cl0ud6uru, 2024-07-19:
> hi there, you can change the sample rate in the following block to match what the tts service you are using uses. Hope this helps 👍 [...]

That did the trick!! Thanks for the quick response!

closed 2024-07-19 by cl0ud6uru

Nitroedge, 2024-09-11:
For reference also, the Elevenlabs default sample rate is 44100 and I can confirm it works for me also.

BigBobbas, 2024-09-12:
I have added a variable for this to make it easier to change in upcoming release, including notes of which sample rates to use. Thanks for your feedback 👍