BigBobbas/ESP32-S3-Box3-Custom-ESPHome #50: Add spoken text as sensor

robgardien, 2024-07-18:
I would like te see the text translated with the STT engine to check if the device understood what I said and to display this text on my ulanzi clock :D

BigBobbas, 2024-07-18:
Hi, 
thank you for the suggestion, I will add this to the todo list 👍 

cheers 

Rob

BigBobbas, 2024-07-18:
I have now added this feature, you can turn conversation display on and off from the voice settings page 👍 
[Link at the bottom of this announcement](<https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/44>)

BigBobbas, 2024-07-18:
added to dev branch sensor and no sensor versions

closed 2024-07-18 by BigBobbas

Kelerion-UK, 2024-07-18:
compiled and working as expected on the no-sensor.. nice work!

BigBobbas, 2024-07-18:
cheers buddy , now to just tick off all the other things on my list 🤦 lol ... in between making the most outside whilst it's not raining!!! 🎉

robgardien, 2024-07-19:
Cheers, thank you!