BigBobbas/ESP32-S3-Box3-Custom-ESPHome #53: Compilation problem

JARDALN, 2024-07-22:
First please excuse my bad english.
I like this software, unfortunately my HA build is underpowered - RPI4/2GB RAM. A restart will occur during compilation. Would someone be willing to compile my yml file and send me the bin soubour ? Thank you for your time and any response.

BigBobbas, 2024-07-22:
Hi 
have you tried adding this to your config to the esphome: component near the top?
```yaml
esphome:
  compile_process_limit: 1
```

JARDALN, 2024-07-22:
Yes, I have this set up. Compilation does not complete, it always ends with a restart. In contrast, maybe an M5 Stack Atom Echo compilation will run, or anything probably less demanding.

BigBobbas, 2024-07-22:
i can compile this for you. if you can post your config

JARDALN, 2024-07-22:
Thank you very much. I will prepare my yml file and get back to you.

JARDALN, 2024-07-22:
I therefore allow myself to attach the yml file after our agreement. Thanks again.

JARDALN, 2024-07-22:
``

BigBobbas, 2024-07-22:
> ``

Hi , there is no file attached ?
Regards Rob

JARDALN, 2024-07-23:
Hello, I had some family concerns.
I can't save the yml file here. Should I rename it or send it to an e-mail?

BigBobbas, 2024-07-23:
if you could email to support@bbdl.co.uk and i can take a look later this evening  👍

JARDALN, 2024-07-23:
Sent to email. Thank you.

closed 2024-07-24 by BigBobbas