BigBobbas/ESP32-S3-Box3-Custom-ESPHome #61: No audio plays on ESP32-S3-Box-3B

raetha, 2024-07-28:
Hi @BigBobbas,

This looks like a fantastic project and I was excited to find it. I was able to successfully push versions of your config from the main and dev branches to my new ESP32-S3-Box-3B. The microphone portion works as expected and triggers an assist response, and I can see from the dev branch that a response is in fact returned.

However, no audio plays, neither assist responses or things sent via the media player entity in Home Assistant. I've tried both the config with and without the sensor features with no difference in audio.

I'm hoping I'm just somehow missing something, but since this is my first ESPHome device, I don't have the faintest idea how to troubleshoot. Any advice for things I might be missing, or ways I can provide more information would be greatly appreciated.

Thanks!

raetha, 2024-07-28:
I believe this is the relevant log lines from the console during a command trigger:
`[21:56:51][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from CREATED to PREPARING. (REQ: 0)
[21:56:51][I][adf_media_player:192]: got new pipeline state: 3, while in MP state IDLE
[21:56:51][D][adf_i2s_out:141]: Set final i2s settings: 16000
[21:56:51][D][esp_audio_processors:124]: Current settings: SRC: rate: 24000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[21:56:51][I][adf_media_player:256]: current mp state: ANNOUNCING
[21:56:51][I][adf_media_player:257]: anouncement: yes
[21:56:51][I][adf_media_player:258]: play_intent: false
[21:56:51][I][adf_media_player:259]: current_uri_: false
[21:56:51][D][esp_audio_sources:103]: Prepare elements called (initial_call)!
[21:56:51][D][esp_audio_sources:137]: Use fixed settings: yes
[21:56:51][D][esp_audio_sources:138]: Streamer status: 1
[21:56:51][D][esp_audio_sources:139]: decoder status: 1
[21:56:51][D][esp_audio_sources:140]: stream uri: https://home.ete.haus/api/tts_proxy/a15091c0ddfebe93c500a8815379589b04ffb10f_en-gb_183ef190f5_tts.piper.mp3
[21:56:51][D][adf_i2s_out:141]: Set final i2s settings: 16000
[21:56:51][D][esp_audio_processors:124]: Current settings: SRC: rate: 24000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[21:56:51][D][adf_audio_element:108]: Preparing [http]...
[21:56:51][D][esp-idf:000]: I (18837566) AUDIO_THREAD: The http task allocate stack on external memory

[21:56:51][D][esp-idf:000]: I (18837569) AUDIO_ELEMENT: [http-0x3d02d7c0] Element task created

[21:56:51][D][adf_audio_element:108]: Preparing [decoder]...
[21:56:51][D][esp-idf:000]: I (18837575) AUDIO_THREAD: The decoder task allocate stack on external memory

[21:56:51][D][esp-idf:000]: I (18837577) AUDIO_ELEMENT: [decoder-0x3d02d978] Element task created

[21:56:51][D][adf_audio_element:108]: Preparing [resampler]...
[21:56:51][D][esp-idf:000]: I (18837583) AUDIO_THREAD: The resampler task allocate stack on external memory

[21:56:51][D][esp-idf:000]: I (18837585) AUDIO_ELEMENT: [resampler-0x3d02db14] Element task created

[21:56:51][D][adf_audio_element:108]: Preparing [i2s_out]...
[21:56:51][D][esp-idf:000]: I (18837591) AUDIO_ELEMENT: [i2s_out-0x3d02dcd0] Element task created

[21:56:51][D][esp_audio_sources:193]: Preparation done!
[21:56:51][D][esp_adf_pipeline:342]: wait for preparation, done
[21:56:51][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[21:56:51][I][adf_media_player:192]: got new pipeline state: 5, while in MP state ANNOUNCING
[21:56:51][I][adf_media_player:256]: current mp state: ANNOUNCING
[21:56:51][I][adf_media_player:257]: anouncement: yes
[21:56:51][I][adf_media_player:258]: play_intent: false
[21:56:51][I][adf_media_player:259]: current_uri_: false
[21:56:51][D][adf_audio_element:165]: Resuming [http]...
[21:56:51][D][adf_audio_element:172]: [http] Sending resume command.
[21:56:51][D][adf_audio_element:165]: Resuming [decoder]...
[21:56:51][D][adf_audio_element:172]: [decoder] Sending resume command.
[21:56:51][D][adf_audio_element:165]: Resuming [resampler]...
[21:56:51][D][adf_audio_element:172]: [resampler] Sending resume command.
[21:56:51][D][esp-idf:000][resampler]: I (18837637) AUDIO_ELEMENT: [resampler] AEL_MSG_CMD_RESUME,state:1

[21:56:51][D][adf_audio_element:191]: [resampler] Checking State, got 72
[21:56:51][D][adf_audio_element:191]: [i2s_out] Checking State, got 74
[21:56:51][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[21:56:51][I][adf_media_player:192]: got new pipeline state: 6, while in MP state ANNOUNCING
[21:56:51][I][adf_media_player:256]: current mp state: ANNOUNCING
[21:56:51][I][adf_media_player:257]: anouncement: yes
[21:56:51][I][adf_media_player:258]: play_intent: false
[21:56:51][I][adf_media_player:259]: current_uri_: false
[21:56:51][I][esp_adf_pipeline:132]: [ resampler ] status: 12
[21:56:51][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:56:51][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[21:56:52][W][component:237]: Component interval took a long time for an operation (258 ms).
[21:56:52][W][component:238]: Components should block for at most 30 ms.
[21:56:52][D][sensor:094]: 'up_sens': Sending state 18837.53711 s with 0 decimals of accuracy
[21:56:52][D][text_sensor:064]: 'Device uptime': Sending state '000:05:13:57'
[21:56:53][W][component:237]: Component interval took a long time for an operation (256 ms).
[21:56:53][W][component:238]: Components should block for at most 30 ms.
[21:56:53][D][sensor:094]: 'up_sens': Sending state 18838.53711 s with 0 decimals of accuracy
[21:56:53][D][text_sensor:064]: 'Device uptime': Sending state '000:05:13:58'
[21:56:54][W][component:237]: Component interval took a long time for an operation (257 ms).
[... 68 lines cut ...]

[21:57:09][D][esp-idf:000][http]: E (18855781) esp-tls: Failed to open new connection

[21:57:09][D][esp-idf:000][http]: E (18855785) TRANSPORT_BASE: Failed to open a new connection

[21:57:09][D][esp-idf:000][http]: E (18855790) HTTP_CLIENT: Connection failed, sock < 0

[21:57:09][D][esp-idf:000][http]: E (18855794) HTTP_STREAM: Failed to open http stream

[21:57:09][D][esp-idf:000][http]: E (18855798) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,28674

[21:57:09][D][esp-idf:000][http]: W (18855802) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7

[... cut ...]
[21:57:10][D][esp-idf:000]: W (18856674) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:57:10][D][esp-idf:000]: W (18856676) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:57:10][D][esp-idf:000]: W (18856680) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:57:10][D][i2s_audio:117]: Other component hasn't released
[21:57:10][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 4)`

raetha, 2024-07-28:
I'm going to close this, caught the connection error at the very end of the logs this time. No idea why the ESPHome provided one was working, but I had a firewall rule in place that restricted what my IoT devices could get to on my LAN. SSL on port 443 wasn't being allowed and was blocked at the network level. Must be some difference in the way the other audio module works with HA, but I seem to be working now.

closed 2024-07-28 by raetha

BigBobbas, 2024-07-28:
glad you managed to figure it out. I hope you enjoy the project. I have some more features in the pipeline and will be coming soon.
Thank you for your support.

Best Wished
Rob 👍

raetha, 2024-07-29:
The project is great, I was looking for exactly this and had been hoping the one provided by the Nabu Casa team would have included the media player, amongst other things. It's also giving me a chance to learn a bit more about ESPHome, which has been on the list, but I simply hadn't gotten to yet.

I suspect I may have some PRs that I'd love to send your way eventually, including for some display optimization, adding USB audio support (if that works), and maybe some other optimizations. Do you have a preference for big/medium/small PRs, and against which branch if I get to that point?

BigBobbas, 2024-07-29:
Hi there ,
thank you for your support. 👍 
PR's are always welcome and the expertise of others is a fantastic contribution. 

I think PR's per type of implementation would be best and make it easier to manage in combination with things I am currently working on. If you could make PR's against s3b_dev.yaml in the dev branch that would be great. And maybe a good idea to ping me a message in advance to say you are going to look at XYZ just to save 2 people doing the same thing :) 

Cheers 
Rob