BigBobbas/ESP32-S3-Box3-Custom-ESPHome #71: error building from commandline

lecrams, 2024-08-05:
Hi, Due to limited HW resources I always build my larger configurations from the commandline.

When I run the build with esphome compile .\s3b.yaml I get the following error: 
Any idea what can be wrong:

ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\Scripts\esphome.exe\__main__.py", line 7, in <module>
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\__main__.py", line 1080, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\__main__.py", line 1058, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 1093, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 947, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 937, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 861, in validate_config
    result.run_validation_steps()
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 144, in run_validation_steps
    task.step.run(self)
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config.py", line 564, in run
    validated = schema(self.conf)
                ^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 251, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 379, in _exec
    v = func(path, v)
        ^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 625, in validate_dict
    return base_validate(path, data.items(), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\voluptuous_schema.py", line 148, in validate_mapping
    cval = cvalue(key_path, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 251, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 379, in _exec
    v = func(path, v)
        ^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\components\font\__init__.py", line 308, in _file_schema
    return validate_file_shorthand(value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\components\font\__init__.py", line 273, in validate_file_shorthand
    return FILE_SCHEMA(
           ^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 254, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 377, in _exec
    v = func(v)
        ^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\components\font\__init__.py", line 309, in _file_schema
    return TYPED_FILE_SCHEMA(value)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\config_validation.py", line 1607, in validator
    value = Schema(schemas[key_v])(value)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 251, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\validators.py", line 379, in _exec
    v = func(path, v)
        ^^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\voluptuous\schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\components\font\__init__.py", line 221, in download_web_font
    external_files.download_content(url, path)
  File "C:\Users\a5128563\AppData\Roaming\Python\Python312\site-packages\esphome\external_files.py", line 103, in download_content
    path.write_bytes(req.content)
  File "C:\Program Files\Python312\Lib\pathlib.py", line 1036, in write_bytes
    with self.open(mode='wb') as f:
         ^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\pathlib.py", line 1013, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: '.esphome\\font\\c9ee33c7\\font.ttf'
PS C:\esp_box\ESP32-S3-Box3-Custom-ESPHome-main>

lecrams, 2024-08-05:
solved by adapting the folders for the png's and fonts

closed 2024-08-05 by lecrams

BigBobbas, 2024-08-05:
that was a quick fix :) thanks for updating and closing 👍