BigBobbas/ESP32-S3-Box3-Custom-ESPHome #74: Error on Adoption Install | `src/esphome/components/adc/adc_sensor.cpp' not found

cornmacabre, 2024-08-10:
Hi! Firstly -- fantastic work here. I'm running into an issue at the final step of adoption & ESPhome installation. Flashing worked fine, and I have the device on-network and addressable by ESPhome. However, there seems to be a persistent issue when I do the final install step.

Using the non-sensor config. I also tried placing the compile=1 bit at the top of the esphome config. 

Rpi4 | ESPHome 2024.7.3
Core 2024.7.2
Supervisor 2024.08.0
Operating System 12.4

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Detected timezone 'America/Chicago'
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO46 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.0.0
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor.o] Source `src/esphome/components/adc/adc_sensor.cpp' not found, needed by target `.pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor.o'.
========================== [FAILED] Took 9.00 seconds ==========================
```

BigBobbas, 2024-08-10:
Hi there,  if you click the 3 dot menu on the device card in the esphome dashboard then click - clean build files then try install again. 👍

cornmacabre, 2024-08-10:
Really appreciate the timely response, literally actively solving configuring this -- so, great timing! 

I did as you suggested, and it did get the compile process moving! I also killed a bunch of add-ons to free up memory (monitoring usage, it doesn't top above 2gb of 8gb in my setup, so shouldn't be an issue. 

However, I'm getting persistently stuck now on a deeper compile step related to i2c/GPIO pins.

Here's where it failed a while later: (restart + clean install files done between takes.)

```
[...more stuff above]
Archiving .pioenvs/esp32-s3box-3/esp-idf/app_update/libapp_update.a
Compiling .pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board.o
Compiling .pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_i2c_pins':
components/audio_board/lyrat_v4_3/board_pins_config.c:39:34: error: 'GPIO_NUM_23' undeclared (first use in this function); did you mean 'GPIO_NUM_43'?
         i2c_config->scl_io_num = GPIO_NUM_23;
                                  ^~~~~~~~~~~
                                  GPIO_NUM_43
components/audio_board/lyrat_v4_3/board_pins_config.c:39:34: note: each undeclared identifier is reported only once for each function it appears in
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_i2s_pins':
components/audio_board/lyrat_v4_3/board_pins_config.c:54:33: error: 'GPIO_NUM_25' undeclared (first use in this function); did you mean 'GPIO_NUM_45'?
         i2s_config->ws_io_num = GPIO_NUM_25;
                                 ^~~~~~~~~~~
                                 GPIO_NUM_45
In file included from /data/cache/platformio/packages/framework-espidf/components/esp_rom/include/esp32s3/rom/ets_sys.h:19,
                 from /data/cache/platformio/packages/framework-espidf/components/log/include/esp_log.h:19,
                 from components/audio_board/lyrat_v4_3/board_pins_config.c:25:
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'i2s_mclk_gpio_select':
components/audio_board/lyrat_v4_3/board_pins_config.c:95:53: error: 'FUNC_GPIO0_CLK_OUT1' undeclared (first use in this function); did you mean 'FUNC_GPIO20_CLK_OUT1'?
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:137:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:95:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:98:53: error: 'FUNC_U0TXD_CLK_OUT3' undeclared (first use in this function); did you mean 'FUNC_U0TXD_CLK_OUT1'?
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:137:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:98:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
In file included from components/audio_board/lyrat_v4_3/board.h:29,
                 from components/audio_board/lyrat_v4_3/board_pins_config.c:28:
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_green_led_gpio':
components/audio_board/lyrat_v4_3/board_def.h:42:35: error: 'GPIO_NUM_22' undeclared (first use in this function); did you mean 'GPIO_NUM_42'?
 #define GREEN_LED_GPIO            GPIO_NUM_22
                                   ^~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:186:12: note: in expansion of macro 'GREEN_LED_GPIO'
     return GREEN_LED_GPIO;
            ^~~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:187:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
*** [.pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o] Error 1
========================= [FAILED] Took 485.84 seconds =========================

```

BigBobbas, 2024-08-10:
What version of esphome do you have installed ?

cornmacabre, 2024-08-10:
> What version of esphome do you have installed ?

ESPHome 2024.7.3

Kelerion-UK, 2024-08-10:
I'd recommend doing a clean first

BigBobbas, 2024-08-10:
are you running ESPHome as an HA addon? I have had the error in the past with `lyrat ` but that was when compiling in windows using the cli. change dto linux and then it worked. But the add-on should work. If you are running the add-on I would uninstall it and re-install it. 
Cheers
Rob

cornmacabre, 2024-08-10:
Hey Rob -- Thanks for the advice! I tried a good old fashioned clean re-flash, deleted the device and old config and attempted to run a clean install. Unfortunately, still hit that same issue. 

I'll try re-installing ESPHome tomorrow (yep! Using as the HA, no windows or anything unusual.) Still hit the same issue on a fresh attempt. To confirm, I should just be using the flash with the default config initially, right? (I'm not flashing the non-sensor version initially -- just using stock from your webtool. I'm updating the ESPhome config with that non-sensor config for final install.) 

Will try again tomorrow -- I'd absolutely love to get this thing working and support what you've got going here. Cheers, Doug.

// Here's the latest full dump of the error I hit. 
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Detected timezone 'America/Chicago'
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO46 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.0.0
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_element.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_process.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sinks.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sources.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_pipeline.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_pipeline_controller.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/media_player/adf_media_player.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/microphone/esp_adf_microphone.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/api_connection.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/api_server.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/list_entities.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/proto.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/api/user_services.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/button/button.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/display/display.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/display/display_buffer.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/display/rect.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/esphome/ota/ota_esphome.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/font/font.o
[... 127 lines cut ...]
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:137:45: note: in definition of macro 'REG_WRITE'
[... cut ...]
components/audio_board/lyrat_v4_3/board_pins_config.c:186:12: note: in expansion of macro 'GREEN_LED_GPIO'
     return GREEN_LED_GPIO;
            ^~~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:187:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
*** [.pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o] Error 1
Compiling .pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/headphone_detect.o
========================= [FAILED] Took 214.04 seconds =========================

BigBobbas, 2024-08-11:
Hi Doug 
just to double check .. this is the `esp32 s3 box 3` that you have? this is the one without any physical buttons on the front? The thing that the compile is failing on is related to the audio chip and pin definitions... which leads me to think it may be a different model.  I was on my mobile yesterday so couldn't read the errors fully. 
Cheers

cornmacabre, 2024-08-12:
Correct, it's the ESP32 Box 3B, the 'basic' non sensor edition without the blue dock. I've had it working with Home Assistants S3 Assistant module previously, with functional (if tiny) audio. Unsure why the audio specific portion is failing out here. Do the GPIO errors give a hint? That seems super weird. I'm going to try reinstalling ESPHome once I've got a confirmed backup working when on at my computer tonight, but yeah it's a bit of a head scratcher for me.

BigBobbas, 2024-08-12:
If it has been flashed previously this is possibly the cause. Those errors i beleive are related to the jesserocks adf external component on the original ESPHome firmware. It sounds like although you have done a 'clean build files' that some of the dependencies are still within the build files. re-installing the ESPHome add-on will possibly solve this. alternatively you can create a new device card from the dashboard - when it displays the API key at the end of the 'wizard' DO NOT click install - click 'skip' a new card will then be created. copy the config into the card and overwrite everything in there. Then change the device name at the top of the config and make sure that it is different to the previous installation. You can then try and compile and it should create a completely fresh build folder. Choose 'manual' rather than wirelessly for the install method. After compile (if it is successful) save the .bin file. Then go to https://web.esphome.io and connect the device, then install , browsing to the file. Before starting the process if you have the previously flashed device in HA>>settings>>devices & services then delete it and also any other copies of the device. Once flashed you should get a notification saying it has been discovered. Add this back to HA at this point, then click 'configure' next to the device in the list of devices in the integration and click to allow service calls.

Hopefully this will do the trick! if not please let me know 👍 

Cheers 
Rob

damanic, 2024-08-16:
@BigBobbas 
I am getting the same error on compile as @cornmacabre 

```
cc1: some warnings being treated as errors
*** [.pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o] Error 1
========================= [FAILED] Took 477.62 seconds =========================
```

This error was on first flash on brand new esp32-s3box-3 that arrived with the stock UI/demo.  Using ESPHome 2024.7.3  ,  HomeAssistant addon , 8GB raspberry Pi.

I used your web installer and instructions via https://support.bbdl.co.uk/

Clean build files and reruns always fails at same point.  

Added `compile_process_limit: 1` and still fails at same point.

BigBobbas, 2024-08-16:
> @BigBobbas I am getting the same error on compile as @cornmacabre
>
> ```
> cc1: some warnings being treated as errors
> *** [.pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o] [...]

Hi, 
Thanks for letting me know and i'm sorry to hear that you are having issues. Is it possible for you to post a photo of your actual device? 
As there are 2 instances of the same issue this is odd. The lyrat error I have come accross before but on a totally different setup. I will do some more digging.
I can compile the config for you which will allow you to flash it without compiling. However i'm concerned that when you come to make your own customisations then you will have the same issue.
Either way , we need to try and resolve this 👍

BigBobbas, 2024-08-16:
can you try removing the following from the config. Then 'save' and then do a clean build files followed by an install
```yaml
    components:
      - name: esp32_s3_box_3_board
        source: github://jesserockz/esp32-s3-box-3-board@main
        refresh: 0s
```

BigBobbas, 2024-08-16:
cheers for the pic. i can see it's the latest revision of the device. so all should be good, I have posted something above to try 👍

damanic, 2024-08-16:
@BigBobbas 
Removed

```
components:
      - name: esp32_s3_box_3_board
        source: github://jesserockz/esp32-s3-box-3-board@main
        refresh: 0s
```

saved, cleaned build files, install

Same lyrat compile error.

```
Compiling .pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board.o
Compiling .pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_i2c_pins':
components/audio_board/lyrat_v4_3/board_pins_config.c:39:34: error: 'GPIO_NUM_23' undeclared (first use in this function); did you mean 'GPIO_NUM_43'?
         i2c_config->scl_io_num = GPIO_NUM_23;
                                  ^~~~~~~~~~~
                                  GPIO_NUM_43
components/audio_board/lyrat_v4_3/board_pins_config.c:39:34: note: each undeclared identifier is reported only once for each function it appears in
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_i2s_pins':
components/audio_board/lyrat_v4_3/board_pins_config.c:54:33: error: 'GPIO_NUM_25' undeclared (first use in this function); did you mean 'GPIO_NUM_45'?
         i2s_config->ws_io_num = GPIO_NUM_25;
                                 ^~~~~~~~~~~
                                 GPIO_NUM_45
In file included from /data/cache/platformio/packages/framework-espidf/components/esp_rom/include/esp32s3/rom/ets_sys.h:19,
                 from /data/cache/platformio/packages/framework-espidf/components/log/include/esp_log.h:19,
                 from components/audio_board/lyrat_v4_3/board_pins_config.c:25:
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'i2s_mclk_gpio_select':
components/audio_board/lyrat_v4_3/board_pins_config.c:95:53: error: 'FUNC_GPIO0_CLK_OUT1' undeclared (first use in this function); did you mean 'FUNC_GPIO20_CLK_OUT1'?
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:137:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:95:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:98:53: error: 'FUNC_U0TXD_CLK_OUT3' undeclared (first use in this function); did you mean 'FUNC_U0TXD_CLK_OUT1'?
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:137:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
/data/cache/platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:98:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
In file included from components/audio_board/lyrat_v4_3/board.h:29,
                 from components/audio_board/lyrat_v4_3/board_pins_config.c:28:
components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_green_led_gpio':
components/audio_board/lyrat_v4_3/board_def.h:42:35: error: 'GPIO_NUM_22' undeclared (first use in this function); did you mean 'GPIO_NUM_42'?
 #define GREEN_LED_GPIO            GPIO_NUM_22
                                   ^~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:186:12: note: in expansion of macro 'GREEN_LED_GPIO'
     return GREEN_LED_GPIO;
            ^~~~~~~~~~~~~~
components/audio_board/lyrat_v4_3/board_pins_config.c:187:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
*** [.pioenvs/esp32-s3box-3/components/audio_board/lyrat_v4_3/board_pins_config.o] Error 1
========================= [FAILED] Took 474.26 seconds =========================
```

damanic, 2024-08-16:
@BigBobbas 
Search on line finds same/similar error here  -> https://community.home-assistant.io/t/esphome-voice-assistant-on-s3-devkitc-1/688139

All brand new to me, so not sure if it helps.

BigBobbas, 2024-08-16:
thanks for that. 
I have now managed to replicate the error. so am close to pinpointing the issue ;)

BigBobbas, 2024-08-16:
it is as if the compile isn't installing the board definitions and the component required that are all hosted externally. does the computer hosting your esphome installation have access to the internet without restrictions ?

damanic, 2024-08-16:
Yes it is effectively unrestricted using 8.8.8.8 for DNS.   Only firewall restrictions are inbound so there should be no issue downloading.  If there was a failed request for download during the build we would expect to see coms error.

youkorr, 2024-08-17:
Hi I just saw my little problem which will be resolved, in my case I created another network for homeassistant and the esp32 see what you can do

damanic, 2024-08-17:
I installed python3 and esphome on my PC (ubuntu)  an ran compile on `s3b_import.yaml` from command line. After a bunch of attempts it worked.

I had to clear a bunch of errors that related to missing PIPs and version mismatches. I also had to reinstate the jesserocks component.

So it seems at the very least there is an issue compiling on Raspberry Pi4 running HAOS and ESPHOME addon.  

It would be nice to have a simple installer method that works for everyone but if this build issue is indeed limited to  `Raspberry Pi4 running HAOS and ESPHOME addon` an acceptable solution might be to add a notice to the README along with a link to alternative install steps (via command line build).

BigBobbas, 2024-08-17:
thanks for the update.. 
This does seem to be isolated to a couple of instances so far, as I have tested and compiled on rpi4 with 8gb previously. 
I am due to merge configs from dev branch soon and will also be updating documents for installation etc. I will add your findings to the notes in the installation doc https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/instructions/installation%20guide.md#note
Thanks for your support, for reporting the issue and  for your patience with getting it up and running.
Cheers 
Rob

ZeniaBol, 2024-08-19:
Hi. Thank you for this project. However, I have similar problem with my 3b ( same like on the picture above)
This is the second already, I return first because after it's fail to compile, it's start losing connection to my PC over the USB-C cable and keep beeping. But second one now have a problem with installation. I try to follow your instruction, fail. I copy and paste a .yaml file ( no sensors to my adopted( partially ) box-3b, no luck
Here is a log from last try:
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/jesserockz/esp32-s3-box-3-board.git@main
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.0.0
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor.o] Source `src/esphome/components/adc/adc_sensor.cpp' not found, needed by target `.pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor.o'.
========================== [FAILED] Took 2.85 seconds ==========================

Looking for your help
Thank you

ZeniaBol, 2024-08-19:
Second try log:
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
INFO Detected timezone 'America/New_York'
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO46 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.0.0
Compiling .pioenvs/esp32-s3box-3/src/main.o
Compiling .pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/es8388.o
Compiling .pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/headphone_detect.o
Compiling .pioenvs/esp32-s3box-3/components/audio_hal/driver/es8311/es8311.o
components/audio_hal/driver/es8388/es8388.c:29:10: fatal error: board.h: No such file or directory

***************************************************************
* Looking for board.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:board.h"
* Web  > https://registry.platformio.org/search?q=header:board.h
*
***************************************************************

 #include "board.h"
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/es8388.o] Error 1
components/audio_hal/driver/es8388/headphone_detect.c:35:10: fatal error: board.h: No such file or directory

***************************************************************
* Looking for board.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:board.h"
* Web  > https://registry.platformio.org/search?q=header:board.h
*
***************************************************************

 #include "board.h"
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/headphone_detect.o] Error 1
components/audio_hal/driver/es8311/es8311.c:27:10: fatal error: board.h: No such file or directory

***************************************************************
* Looking for board.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:board.h"
* Web  > https://registry.platformio.org/search?q=header:board.h
*
***************************************************************

 #include "board.h"
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/components/audio_hal/driver/es8311/es8311.o] Error 1
========================= [FAILED] Took 23.73 seconds =========================

BigBobbas, 2024-08-19:
Hi , 
can you try cleaning build files. `esphome clean filename.yaml` or clean build files from the 3 dot menu of the device card in the dashboard'

ZeniaBol, 2024-08-19:
I forgot to mention, I did that before second try. Didn't help

damanic, 2024-08-19:
@BigBobbas 
I think you need to undo this commit -> https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/commit/b07d41f5881d70c0583e02a47fe4714537cf2967

I got same issues as @ZeniaBol until I reinstated jesserocks.

BigBobbas, 2024-08-19:
I have added this back in now , thank you for pointing that out @damanic , my error for not adding it back in. @ZeniaBol please copy the file https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b_import.yaml and try again. before installing , please do another clean.
thanks for your patience guys 👍

ZeniaBol, 2024-08-19:
Working great. Thank you. I will play with this box tomorrow 
[image]

cornmacabre, 2024-08-31:
> I installed python3 and esphome on my PC (ubuntu) an ran compile on `s3b_import.yaml` from command line. After a bunch of attempts it worked.
>
> I had to clear a bunch of errors that related to [...]

So sounds like the _lyrat_v4_3_ related compile errors are still present for some hardware? I tried again doing the usual clean-and still consistently hit the same issue, I have the same setup as you (8gb model). 

I started to go down the path with python3 and esphome on my windows PC, and it was just a nightmare of error after error attempting to compile (likely unrelated to the config here, just my lack of knowledge and familiarity in how to compile via python). Finally gave up when I was in the depths of installing all sorts of random functions and files and still hitting issues.

I have a separate pi5 that runs docker -- so I may attempt this later on different hardware that isn't windows.

BigBobbas, 2024-12-21:
closing due to age and no recent updates

closed 2024-12-21 by BigBobbas