BigBobbas/ESP32-S3-Box3-Custom-ESPHome #78: Error on adoption install - missing board.h?

gmalbert, 2024-08-19:
Regular flashing went fine. Went to the ESPHome dashboard to adopt and install. Got pretty far but kept getting stuck here with the error below. Multiple retries has not fixed it. I'm using the current version of ESP Home. 

```
components/audio_hal/driver/es8388/es8388.c:29:10: fatal error: board.h: No such file or directory

***************************************************************
* Looking for board.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:board.h"
* Web  > https://registry.platformio.org/search?q=header:board.h
*
***************************************************************

 #include "board.h"
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/components/audio_hal/driver/es8388/es8388.o] Error 1
```

damanic, 2024-08-20:
@gmalbert 
Probably the same issue as found here -> https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/issues/74#issuecomment-2297652473

Use the latest yaml https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b_import.yaml

gmalbert, 2024-08-20:
Correct. Works now. I'll close this. Thank you!

closed 2024-08-20 by gmalbert