BigBobbas/ESP32-S3-Box3-Custom-ESPHome #81: Missing esp_afe_sr_models.h

kevinbur, 2024-08-28:
Hi Guys, 

I wonder If you can help me past this problem.

I can get to step 8 of the installation guide, I've tried  many times and cleaned the build files but always get the same error.

[logs_esp32-s3box-3_run.txt](https://github.com/user-attachments/files/16785415/logs_esp32-s3box-3_run.txt)

Compiling .pioenvs/esp32-s3box-3/components/audio_recorder/recorder_sr.o
components/audio_recorder/recorder_sr.c:40:10: fatal error: esp_afe_sr_models.h: No such file or directory

***************************************************************************
* Looking for esp_afe_sr_models.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_afe_sr_models.h"
* Web  > https://registry.platformio.org/search?q=header:esp_afe_sr_models.h
*
***************************************************************************

 #include "esp_afe_sr_models.h"

Running.

Core
2024.8.3
Supervisor
2024.08.0
Operating System
13.1
Frontend
20240809.0
ESPHome
Current version: 2024.8.1 

Thanks

BigBobbas, 2024-08-28:
Hi there,
what hardware are  you running ESPHome on? and What OS or is it HA add-on .
Thanks 
Rob

kevinbur, 2024-08-28:
It's running on and intel i5 nuc with HA os.

Kelerion-UK, 2024-08-28:
I presume that means you are trying it on the HA addon... I know you said you'd done a clean.. but I'd try removing the device.. and in the integration (settings/devices/esphome)... and then re-creating it.

kevinbur, 2024-08-28:
I've given that a go several times.

BigBobbas, 2024-08-28:
have you tried uninstalling and re-installing the ESPHome add-on? some dependency is missing by the looks of things.

kevinbur, 2024-08-28:
Not yet, I'll give that a go tomorrow  and get back with the results.

BigBobbas, 2024-08-29:
Hi,
this is a bit of an odd one :) 
could you try with the config from here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/dev-live-wip/s3b_dev.yaml
you can copy and paste the entire contents into the device card in the ESPHome dashboard - overwriting everything - save and install
Cheers
Rob

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas