BigBobbas/ESP32-S3-Box3-Custom-ESPHome #82: Microphone not working

dsync89, 2024-08-30:
Hi guys, it's strange that the mic is not working when I flash this custom firmware on a new `ESP32-S3-BOX-3`.

Following is the software version I used:
- HA Core: 2024.8.3
- HA Supervisor: 2024.08.0
- HA OS: 13.1
- HA Frontend: 20240809.0
- ESPHome: 2024.8.1

Mic is working fine if I flash it using official firmware -> https://www.home-assistant.io/voice_control/s3_box_voice_assistant/

BigBobbas, 2024-08-30:
Hi ,
that is very strange. How are you installing the firmware? through the web-installer or by manually adding one of the configs? if manually adding, can you paste the link to the file you used.

Do you have the s3box set to use 'on device' or 'in home assistant' for the wakeword ?

Also if you can view the logs for the device from the ESPHome dashboard do you see any errors?

Many Thanks
Rob 👍

dsync89, 2024-08-30:
> Hi , that is very strange. How are you installing the firmware? through the web-installer or by manually adding one of the configs? if manually adding, can you paste the link to the file you used. [...]

Appreciate the prompt reply @BigBobbas .

I installed it through web-installer, and have tried both `on device` and `home assistant` for the wakeword but it didn't trigger the nabu face. When I manually click the bottom right icon to activate the face, it didn't seem to pick up any word I sent.

I also didn't see any error in the logs. Peculiar one indeed.

BigBobbas, 2024-08-31:
Hi, 
Is the Home Assistant logo in the top corner showing in blue or red ?
If this is red then there is a connection issue between HA and the S3 Box - I Would delete the device from HA>>Settings>>Devices & Services>> ESPHome - Delete any entries that are related to the S3Box and then re-add them. When asked for host, you can enter the IP address for the device, you should see this by tapping the `info` icon in the top right corner of the boxes screen.

To test that the microphone is functioning:
Set the wakeword location to 'on device'
Then open the ESPHome device logs from the ESPHome add-on dashboard and watch these as you say "ok Nabu"
There should be a change in the logs as it picks up and recognises the wakeword.  Can you please paste the logs so that I can take a look to see what is happening.

Thank you for your patience, i'm sure we will be able to get this up and running :)

Rob

ThirteenTX, 2024-09-02:
I am having the same issue. I am using a ESP32-S3-BOX-3b so let me know if I should delete this and create a new issue. I flashed mine from the site and then after it is added to ESPHome I install the s3b_no_sensors.yaml. It shows connected in the upper left of the screen. It doesn't seem to show anything in the logs when saying the wake word. If I press the push to talk button in the bottom right this appears in the logs.
```
[09:33:10][D][voice_assistant:221]: Starting Microphone
[09:33:10][D][esp_adf_pipeline.microphone:025]: start request while ine state 0
[09:33:10][D][esp_adf_pipeline:060]: Starting request, current state STOPPED
[09:33:10][D][voice_assistant:504]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[09:33:10][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from STOPPED to PREPARING. (REQ: 0)
[09:33:10][D][voice_assistant:504]: State changed from STARTING_MICROPHONE to STOP_MICROPHONE
[09:33:10][D][voice_assistant:510]: Desired state set to IDLE
[09:33:10][D][voice_assistant:504]: State changed from STOP_MICROPHONE to IDLE
[09:33:10][D][adf_audio_element:108]: Preparing [i2s_in]...
[09:33:10][D][adf_audio_element:108]: Preparing [resampler]...
[09:33:10][D][voice_assistant:504]: State changed from IDLE to START_MICROPHONE
[09:33:10][D][voice_assistant:510]: Desired state set to START_PIPELINE
[09:33:10][D][voice_assistant:221]: Starting Microphone
[09:33:10][D][esp_adf_pipeline.microphone:025]: start request while ine state 1
[09:33:10][D][esp_adf_pipeline:060]: Starting request, current state PREPARING
[09:33:10][D][voice_assistant:504]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[09:33:10][D][adf_audio_element:108]: Preparing [pcm_reader]...
[09:33:10][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 2 bits: 16, DST: rate: 16000, ch: 1, bits 16
[09:33:10][D][esp_adf_pipeline:342]: wait for preparation, done
[09:33:10][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[09:33:10][D][adf_audio_element:165]: Resuming [i2s_in]...
[09:33:10][D][adf_audio_element:172]: [i2s_in] Sending resume command.
[09:33:10][D][esp-idf:000][i2s_in]: I (92881) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1

[09:33:10][D][adf_audio_element:191]: [i2s_in] Checking State, got 78
[09:33:10][I][esp_adf_pipeline:132]: [ i2s_in ] status: 12
[09:33:10][D][adf_audio_element:191]: [resampler] Checking State, got 78
[09:33:10][I][esp_adf_pipeline:132]: [ resampler ] status: 12
[09:33:10][D][adf_audio_element:191]: [pcm_reader] Checking State, got 65
[09:33:10][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[09:33:10][D][voice_assistant:504]: State changed from STARTING_MICROPHONE to START_PIPELINE
[09:33:10][D][voice_assistant:275]: Requesting start...
[09:33:10][D][voice_assistant:504]: State changed from START_PIPELINE to STARTING_PIPELINE
[09:33:10][D][voice_assistant:525]: Client started, streaming microphone
[09:33:10][D][voice_assistant:504]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[09:33:10][D][voice_assistant:510]: Desired state set to STREAMING_MICROPHONE
[09:33:10][D][voice_assistant:627]: Event Type: 0
[09:33:10][D][voice_assistant:620]: Signaling stop...
[09:33:10][D][voice_assistant:504]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[09:33:10][D][voice_assistant:510]: Desired state set to IDLE
[09:33:10][D][esp_adf_pipeline:070]: Called 'stop' while in RUNNING state.
[09:33:10][D][voice_assistant:504]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[09:33:10][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from RUNNING to ABORTING. (REQ: 1)
[09:33:10][D][adf_audio_element:324]: [i2s_in] Checking State for stopping, got 3
[09:33:10][D][esp-idf:000][resampler]: W (92964) AUDIO_ELEMENT: IN-[resampler] AEL_IO_ABORT

[09:33:10][D][esp-idf:000][i2s_in]: W (92977) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[09:33:10][D][esp-idf:000][i2s_in]: W (92981) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[09:33:10][D][esp-idf:000][i2s_in]: W (92986) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[09:33:10][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from ABORTING to STOPPED. (REQ: 1)
[09:33:10][D][voice_assistant:504]: State changed from STOPPING_MICROPHONE to IDLE
[09:33:18][W][component:237]: Component interval took a long time for an operation (220 ms).
[09:33:18][W][component:238]: Components should block for at most 30 ms.
[09:33:38][W][component:237]: Component interval took a long time for an operation (219 ms).
[09:33:38][W][component:238]: Components should block for at most 30 ms.
```
Again it seems like our issues are the same but let me know if I should delete this and create a new issue. I am excited to get this going as it looks really cool. Thanks.

BigBobbas, 2024-09-02:
> I am having the same issue. I am using a ESP32-S3-BOX-3b so let me know if I should delete this and create a new issue. I flashed mine from the site and then after it is added to ESPHome I install [...]

Hi @ThirteenTX 
no it's fine to continue in this issue :)
the logs look like they are reporting correctly if these are from when you press the 'tap to talk' button.  Have you had the Box working previously with the stock ESPHome firmware or is this the first use ?
Many Thanks 
Rob

ThirteenTX, 2024-09-02:
I did have it working under the default Home Assistant installer where voice was working. 

In my last attempt to get this version working I deleted it from ESPHome and from the ESPHome integration and then loaded it back up.

BigBobbas, 2024-09-02:
on top of the device make sure that the orange light to the left of the button is not lit. if it is then press the top button and then restart the device by removing power.
If the light is not lit then that won't be an issue.
Do you have the box set to use 'on device' or 'home assistant' wake word?

ThirteenTX, 2024-09-02:
Only a green light is lit at the top.

Your mention of 'on device' or 'home assistant' made me decide to change the "Wake word engine location" to on device. That does allow for wake word detection and command execution which is a step in the right direction. 

I do notice that the 'tap to talk' still does not work though. Less of an issue but I imagine it is still intended to work. Not sure if something there needs to be adjusted on my end as well. 

Really appreciate the help!

BigBobbas, 2024-09-02:
with the 'tap to talk' you don't use a wakeword and just say the command, so not sure if you knew that.

With 'Home Assistant' as the wake word engine location this would require 'openwakeword' or a similar wakeword engine to be installed and configured in Home Assistant and as such constantly streams audio from the mic every 5 to HA in order for it to detect and process the wake word. With on device this is all carried out on the device and only streams audio to HA once it determines that the wakeword has been detected and confirmed.

Hope that helps :)

ThirteenTX, 2024-09-02:
Yes, when I press tap to talk it doesn't seem to register the same command it registers when I initiate it from the wake word detection. 

When using wake word I say "Okay nabu, turn on the computer room light."  and it turns it on

When using 'push to talk' I say "turn on the computer room light" and nothing happens and the smiling home icon stays on the screen until I press the red circle.

ThirteenTX, 2024-09-02:
Additionally I noticed when I say the wake word with it on device, the sensor for "Assist in progress" goes to on but when using tap to take it stays in the off state.

BigBobbas, 2024-09-03:
I will check that , it has been a while since I have used that config. The newer version doesn't have the tap to talk. This was removed to provide space for better functionality. my theory being that if you are at the device tapping the screen, then it is just as easy to tap the screen and tap a control to do the same controls as voice (if that makes sense)

ThirteenTX, 2024-09-03:
Totally makes sense. Since I am new to the project and just learning about these, would it make more sense to test with the Dev build? I also agree about the button, so not a game changer there if it doesn't work.

BigBobbas, 2024-09-03:
the dev build is pretty stable and only really waiting on me to finish the 'wip' version and then i will merge them both and add them as the 'stable' version. So yes it's well worth trying , all you need to do is copy and paste the config and overwrite the current config in the ESPHome dashboard, click save and install. 
this would be the one to use https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/dev/s3b_dev.yaml
👍

xsmrcek, 2024-09-05:
This may not be related, but I have played with official configuration, adding sensors etc. My microphone started working again when I removed this:
[image]
Then I read somewhere that adding this may cause instability, sorry I don;t have a link now. So maybe your config is right, but there is issue on ESP side which have to be solved first.

BigBobbas, 2024-09-05:
> This may not be related, but I have played with official configuration, adding sensors etc. My microphone started working again when I removed this: [...]

Thank you for your input,
This has not been an issue with this project with many users not experiencing this.
Prior to me starting the project this was an issue and having the i2c bus configured would prevent voice from functioning.  However this is not an issue for this particular firmware. 
Many Thanks
Rob

ThirteenTX, 2024-09-28:
Hey there, I recently bought a second voice assistant. The this time I got the ESP32-S3-BOX-3 with the blue base with sensors. Since flashing the newer version , it is back to not recognizing the mic. I guess I will try and go back to the archived build for now and see how that goes. Let me know if you want me to test anything on my end. I only have the green light lit and have never pressed the hardware mute on this new one.

BigBobbas, 2024-09-28:
> Hey there, I recently bought a second voice assistant. The this time I got the ESP32-S3-BOX-3 with the blue base with sensors. Since flashing the newer version , it is back to not recognizing the [...]

Hi There ,
firstly... i have to ask.. have you removed the clear plastic film from the front of the screen?

ThirteenTX, 2024-09-28:
I have. I tried the archived build and it also failed. It can see things in the logs like presence sensor etc but none of the wake words work in the new build for me as well as the Okay Nabu in the old build. I am currently loading up the official build to see if that works or not.

BigBobbas, 2024-09-28:
the 'on device' wakeword should work as it only relies on the device .  if you can make sure that 'on device' is selected by going into settings>>voice on the device 
By default 'ok nabu, Alexa, Hey Jarvis, & hey Mycroft should work'

ThirteenTX, 2024-09-28:
Ah that was it. It was on Home Assistant and that doesn't seem to be recognized.

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas

rojoricardo, 2026-11-02:
Hello, I just managed to flash this firmware and when trying to get the wake word detected nothing happends.
I checked the logs as suggested and it just keeps repeting this:

```

[00:00:23][E][i2s_audio.speaker.std:427]: Parent bus is busy
[00:00:23][E][i2s_audio.speaker:142]: Driver failed to start; retrying in 1 second
[00:00:23][E][component:297]: i2s_audio.speaker set Error flag: unspecified
[00:00:24][E][component:322]: i2s_audio.speaker cleared Error flag
[00:00:24][E][i2s_audio.speaker.std:427]: Parent bus is busy
[00:00:24][E][i2s_audio.speaker:142]: Driver failed to start; retrying in 1 second
[00:00:24][E][component:297]: i2s_audio.speaker set Error flag: unspecified
```