BigBobbas/ESP32-S3-Box3-Custom-ESPHome #94: The ESPHome in progress binary sensor binary_sensor.esp32_s3_box_3_assist_in_progress is deprecated.

ThirteenTX, 2024-10-03:
It looks like we have some time, but all the same, I wanted to get the word out here. Looks like they are changing the current implementation.
[image]

BigBobbas, 2024-10-04:
Hi - nothing to worry about - this is only on the HA side of things.
HA adds the entity 'assist in progress' to all ESPHome VA devices,, this has now changed to a sensor with multiple states, which will give better clarity to what state the 'VA' is currently in.
Clicking submit is all that you need to do. 
If you had any automations in HA using 'assist in progress' as a trigger , you would need to update these. 

Cheers
Rob 👍

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas