BigBobbas/ESP32-S3-Box3-Custom-ESPHome #97: Failure to Play Media on Box 3

yuuki94, 2024-10-13:
I installed this on 2 Box 3's and getting the same issue where the box was able to perform TTS but failed to play the media file on the device itself. Below the logs:

`[16:43:40][D][esp-idf:000]: I (50458) AUDIO_THREAD: The http task allocate stack on external memory

[16:43:40][D][esp-idf:000]: I (50460) AUDIO_ELEMENT: [http-0x3d030548] Element task created

[16:43:40][D][adf_audio_element:108]: Preparing [decoder]...
[16:43:40][D][esp-idf:000]: I (50463) AUDIO_THREAD: The decoder task allocate stack on external memory

[16:43:40][D][esp-idf:000]: I (50465) AUDIO_ELEMENT: [decoder-0x3d030700] Element task created

[16:43:40][D][adf_audio_element:108]: Preparing [resampler]...
[16:43:40][D][esp-idf:000]: I (50471) AUDIO_THREAD: The resampler task allocate stack on external memory

[16:43:40][D][esp-idf:000]: I (50472) AUDIO_ELEMENT: [resampler-0x3d03089c] Element task created

[16:43:40][D][adf_audio_element:108]: Preparing [i2s_out]...
[16:43:40][D][esp-idf:000]: I (50478) AUDIO_ELEMENT: [i2s_out-0x3d030a58] Element task created

[16:43:40][D][esp_audio_sources:193]: Preparation done!
[16:43:40][D][esp_adf_pipeline:342]: wait for preparation, done
[16:43:40][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[16:43:40][I][adf_media_player:192]: got new pipeline state: 5, while in MP state ANNOUNCING
[16:43:40][I][adf_media_player:256]: current mp state: ANNOUNCING
[16:43:40][I][adf_media_player:257]: anouncement: yes
[16:43:40][I][adf_media_player:258]: play_intent: false
[16:43:40][I][adf_media_player:259]: current_uri_: false
[16:43:40][D][adf_audio_element:165]: Resuming [http]...
[16:43:40][D][adf_audio_element:172]: [http] Sending resume command.
[16:43:40][D][adf_audio_element:165]: Resuming [decoder]...
[16:43:40][D][adf_audio_element:172]: [decoder] Sending resume command.
[16:43:40][D][esp-idf:000][decoder]: I (50512) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[16:43:40][D][esp-idf:000][decoder]: I (50521) MP3_DECODER: MP3 opened

[16:43:40][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[16:43:40][I][adf_media_player:192]: got new pipeline state: 6, while in MP state ANNOUNCING
[16:43:40][I][adf_media_player:256]: current mp state: ANNOUNCING
[16:43:40][I][adf_media_player:257]: anouncement: yes
[16:43:40][I][adf_media_player:258]: play_intent: false
[16:43:40][I][adf_media_player:259]: current_uri_: false
[16:43:41][D][esp-idf:000][http]: E (50745) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780

[16:43:41][D][esp-idf:000][http]: E (50748) esp-tls: Failed to open new connection

[16:43:41][D][esp-idf:000][http]: E (50752) TRANSPORT_BASE: Failed to open a new connection

[16:43:41][D][esp-idf:000][http]: E (50758) HTTP_CLIENT: Connection failed, sock < 0

[16:43:41][D][esp-idf:000][http]: E (50761) HTTP_STREAM: Failed to open http stream

[16:43:41][D][esp-idf:000][http]: E (50765) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,28674

[16:43:41][D][esp-idf:000][http]: W (50768) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7

[16:43:41][D][esp_adf_pipeline:518]: Called deinit_all
[16:43:41][D][esp-idf:000]: I (51533) AUDIO_PIPELINE: audio_pipeline_unlinked

[16:43:41][D][esp-idf:000]: W (51535) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE

[16:43:41][D][esp-idf:000]: W (51537) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[16:43:41][D][esp-idf:000]: W (51540) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE

[16:43:41][D][esp-idf:000]: W (51542) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[16:43:41][D][i2s_audio:117]: Other component hasn't released
[16:43:41][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 4)
[16:43:41][I][adf_media_player:192]: got new pipeline state: 0, while in MP state ANNOUNCING
[16:43:42][I][adf_media_player:256]: current mp state: IDLE
[16:43:42][I][adf_media_player:257]: anouncement: false
[16:43:42][I][adf_media_player:258]: play_intent: false
[16:43:42][I][adf_media_player:259]: current_uri_: false
[16:43:42][D][sensor:094]: 'S3 Temperature': Sending state 31.82564 °C with 2 decimals of accuracy`

BigBobbas, 2024-10-13:
Hi, have you enabled the dive to make action calls ? covered in the section here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/instructions/make%20it%20your%20own.md#getting-started , also have you given both devices unique names?
thanks
Rob

yuuki94, 2024-10-13:
Yes, I am able to control my HA devices via the 2 boxes. The box also displays the text response to my speech, however it was unable to play the mp3 media file from my HA tts API.

Both devices have unique names. When I started with 1 box, the issue of unabling to play the mp3 media file was present.

Using an external media player works but then I would like to use the Box 3 to also play media.

yuuki94, 2024-10-13:
I'm using Piper TTS on my HA.

BigBobbas, 2024-10-13:
ok thank you , I will test with piper this afternoon. I use Nabu Casa usually which works fine so will test and let you know 👍

yuuki94, 2024-10-13:
Thanks. External media player works with the same Piper TTS. Just that the box's own player doesn't.

Does the SSL/Http errors in the logs indicate any particular setting being the issue?

BigBobbas, 2024-10-13:
hi, sorry was on mobile earlier and couldn't have a good look at the logs, have you removed the url paths to the mp3 file created from the logs or is there not one showing in the logs when tts should be playing?
I have tested with piper and this is playing fine.
a few further questions:
1 . do you have the latest version of the firmware installed 
2. What do you have the sample rate set as in the substitutions at the top of the config?
3. What version of HA do you have installed?
4. Are your S3 box and HA both on the same vlan / subnet? 
5. How do you have HA installed ? eg. VM, bare metal, container etc?

yuuki94, 2024-10-13:
The Mp3 url was listed, I didnt include that part of the logs.
1. Yeah, I just installed yesterday.
2. tts_voice_speed: "24000"
3. HA Core 2024.10.2, HA OS 13.1
4. Yup
5. On HA Yellow.

yuuki94, 2024-10-13:
full logs. currently i've also set it to also play on external media player.

```[22:22:46][D][voice_assistant:637]: Event Type: 10
[22:22:46][D][voice_assistant:646]: Wake word detected
[22:22:46][D][media_player:061]: 's3-box_media_player' - Setting
[22:22:46][D][media_player:065]:   Command: STOP
[22:22:46][D][esp_adf_pipeline:070]: Called 'stop' while in UNINITIALIZED state.
[22:22:46][D][voice_assistant:637]: Event Type: 3
[22:22:46][D][voice_assistant:651]: STT started
[22:22:47][D][binary_sensor:036]: 'Presence detect': Sending state OFF
[22:22:47][D][voice_assistant:637]: Event Type: 11
[22:22:47][D][voice_assistant:793]: Starting STT by VAD
[22:22:50][D][voice_assistant:637]: Event Type: 12
[22:22:50][D][voice_assistant:797]: STT by VAD end
[22:22:50][D][voice_assistant:514]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[22:22:50][D][voice_assistant:520]: Desired state set to AWAITING_RESPONSE
[22:22:50][D][esp_adf_pipeline:070]: Called 'stop' while in RUNNING state.
[22:22:50][D][voice_assistant:514]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[22:22:50][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from RUNNING to ABORTING. (REQ: 1)
[22:22:51][D][adf_audio_element:324]: [i2s_in] Checking State for stopping, got 3
[22:22:51][D][esp-idf:000][i2s_in]: W (22030874) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[22:22:51][D][adf_audio_element:324]: [pcm_reader] Checking State for stopping, got 3
[22:22:51][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from ABORTING to STOPPED. (REQ: 1)
[22:22:51][D][voice_assistant:514]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[22:22:58][D][voice_assistant:637]: Event Type: 4
[22:22:58][D][voice_assistant:665]: Speech recognised as: " Hello"
[22:22:58][D][voice_assistant:637]: Event Type: 5
[22:22:58][D][voice_assistant:670]: Intent started
[22:22:58][D][sensor:094]: 'up_sens': Sending state 22037.24609 s with 0 decimals of accuracy
[22:23:02][D][voice_assistant:637]: Event Type: 6
[22:23:02][D][voice_assistant:637]: Event Type: 7
[22:23:02][D][voice_assistant:693]: Response: "H-hello! Um, it’s so nice to see you! How can I help you today? If that’s okay... "
[22:23:02][D][voice_assistant:637]: Event Type: 8
[22:23:02][D][voice_assistant:715]: Response URL: "https://ha.stcanvas.link/api/tts_proxy/c7e50ebfb02505b1606d279c5062a519a7d369bf_en-gb_e135174bb9_tts.piper.mp3"
[22:23:02][D][voice_assistant:514]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[22:23:02][D][voice_assistant:520]: Desired state set to STREAMING_RESPONSE
[22:23:02][D][media_player:061]: 's3-box_media_player' - Setting
[22:23:02][D][media_player:068]:   Media URL: https://ha.stcanvas.link/api/tts_proxy/c7e50ebfb02505b1606d279c5062a519a7d369bf_en-gb_e135174bb9_tts.piper.mp3
[22:23:02][D][media_player:074]:  Announcement: yes
[22:23:02][D][adf_media_player:057]: Got control call in state IDLE
[22:23:02][D][adf_media_player:058]: req_track stream uri: https://ha.stcanvas.link/api/tts_proxy/c7e50ebfb02505b1606d279c5062a519a7d369bf_en-gb_e135174bb9_tts.piper.mp3
[22:23:02][D][esp_adf_pipeline:060]: Starting request, current state UNINITIALIZED
[22:23:02][W][font:112]: Encountered character without representation in font: '\xe2'
[22:23:02][W][font:112]: Encountered character without representation in font: '\x80'
[22:23:02][W][font:112]: Encountered character without representation in font: '\x99'
[22:23:02][D][voice_assistant:637]: Event Type: 2
[22:23:02][D][voice_assistant:729]: Assist Pipeline ended
[22:23:02][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from UNINITIALIZED to INITIALIZING. (REQ: 0)
[22:23:02][I][adf_media_player:192]: got new pipeline state: 1, while in MP state IDLE
[22:23:02][I][adf_media_player:256]: current mp state: IDLE
[22:23:02][I][adf_media_player:257]: anouncement: yes
[22:23:02][I][adf_media_player:258]: play_intent: false
[22:23:02][I][adf_media_player:259]: current_uri_: false
[22:23:02][D][esp-idf:000]: I (22042134) MP3_DECODER: MP3 init

[22:23:02][D][DAC_ES8311:227]: ES8311 chip found.
[22:23:02][D][esp-idf:000]: I (22042160) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[22:23:02][D][i2s_audio:067]: Install driver requested by Writer
[22:23:02][D][DAC_ES8311:256]: Setting up ES8311...
[22:23:02][D][esp_adf_pipeline:497]: pipeline tag 0, http
[22:23:02][D][esp_adf_pipeline:497]: pipeline tag 1, decoder
[22:23:02][D][esp_adf_pipeline:497]: pipeline tag 2, resampler
[22:23:02][D][esp_adf_pipeline:497]: pipeline tag 3, i2s_out
[22:23:02][D][esp-idf:000]: I (22042181) AUDIO_PIPELINE: link el->rb, el:0x3d030718, tag:http, rb:0x3d030d70

[22:23:02][D][esp-idf:000]: I (22042184) AUDIO_PIPELINE: link el->rb, el:0x3d0308d0, tag:decoder, rb:0x3d031db0

[22:23:02][D][esp-idf:000]: I (22042187) AUDIO_PIPELINE: link el->rb, el:0x3d030a6c, tag:resampler, rb:0x3d032df0
[... 127 lines cut ...]
[22:23:05][D][adf_audio_element:165]: Resuming [pcm_reader]...
[22:23:05][I][esp_adf_pipeline:132]: [ i2s_in ] status: 12
[22:23:05][D][adf_audio_element:191]: [i2s_in] Checking State, got 78
[22:23:05][I][esp_adf_pipeline:132]: [ resampler ] status: 12
[22:23:05][D][adf_audio_element:191]: [resampler] Checking State, got 78
[22:23:05][D][adf_audio_element:191]: [pcm_reader] Checking State, got 65
[22:23:05][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[... cut ...]
[22:23:05][W][font:112]: Encountered character without representation in font: '\x99'
[22:23:06][D][homeassistant.text_sensor:017]: 'media_player.living_room_display': Got state 'playing'
[22:23:07][W][font:112]: Encountered character without representation in font: '\xe2'
[22:23:07][W][font:112]: Encountered character without representation in font: '\x80'
[22:23:07][W][font:112]: Encountered character without representation in font: '\x99'
[22:23:09][D][sensor:094]: 'WiFi db': Sending state -32.00000 dBm with 0 decimals of accuracy```

BigBobbas, 2024-10-13:
when set to output to an external media player, the audio isn't streamed to the box it needs to be off for the box to recieve responses.

yuuki94, 2024-10-14:
> when set to output to an external media player, the audio isn't streamed to the box it needs to be off for the box to recieve responses.

noted. but when output isn't set to external, the box doesnt stream the audio either.

BigBobbas, 2024-10-16:
Hi , looking at this further I think the issue may be due to the url's being created are https:// 
"https://ha.stcanvas.link/api/tts_proxy/c7e50ebfb02505b1606d279c5062a519a7d369bf_en-gb_e135174bb9_tts.piper.mp3"
if you paste that into a browser does it play?

yuuki94, 2024-10-17:
> Hi , looking at this further I think the issue may be due to the url's being created are https:// "https://ha.stcanvas.link/api/tts_proxy/c7e50ebfb02505b1606d279c5062a519a7d369bf_en-gb_e135174bb9_tts.piper.mp3" if you paste that into a browser does it play?

It plays when it's pasted onto the browser.
The URL is on a public domain so it should be accessible as long as the box is connected to internet.

BigBobbas, 2024-10-17:
in HA if you got to 'media' from the sidebar _> text to speech -> piper and have the media player for the s3box selected at the bottom right are you able to send and hear TTS from there?

yuuki94, 2024-10-17:
I wasn't able to hear the TTS when s3box was selected. Logs below:

```
[21:27:02][D][sensor:094]: 'WiFi db': Sending state -34.00000 dBm with 0 decimals of accuracy
[21:27:02][D][sensor:094]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[21:27:02][D][sensor:094]: 'up_sens': Sending state 187282.37500 s with 0 decimals of accuracy
[21:27:09][D][media_player:061]: 's3-box_media_player' - Setting
[21:27:09][D][media_player:068]:   Media URL: https://ha.stcanvas.link/api/tts_proxy/9e9b7ab4537900b00b43d9e42a52003f01a8e8f4_en-us_894b186240_tts.piper.mp3
[21:27:09][D][media_player:074]:  Announcement: yes
[21:27:09][D][adf_media_player:057]: Got control call in state IDLE
[21:27:10][D][adf_media_player:058]: req_track stream uri: https://ha.stcanvas.link/api/tts_proxy/9e9b7ab4537900b00b43d9e42a52003f01a8e8f4_en-us_894b186240_tts.piper.mp3
[21:27:10][D][esp_adf_pipeline:060]: Starting request, current state STOPPED
[21:27:10][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from STOPPED to PREPARING. (REQ: 0)
[21:27:10][I][adf_media_player:192]: got new pipeline state: 3, while in MP state IDLE
[21:27:10][D][adf_i2s_out:141]: Set final i2s settings: 16000
[21:27:10][D][esp_audio_processors:124]: Current settings: SRC: rate: 24000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[21:27:10][I][adf_media_player:256]: current mp state: ANNOUNCING
[21:27:10][I][adf_media_player:257]: anouncement: yes
[21:27:10][I][adf_media_player:258]: play_intent: false
[21:27:10][I][adf_media_player:259]: current_uri_: false
[21:27:10][D][esp_audio_sources:103]: Prepare elements called (initial_call)!
[21:27:13][D][esp_audio_sources:137]: Use fixed settings: yes
[21:27:14][D][esp-idf:000][http]: E (187294661) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00

[21:27:14][D][esp-idf:000][http]: E (187294663) esp-tls: create_ssl_handle failed

[21:27:14][D][esp-idf:000][http]: E (187294666) esp-tls: Failed to open new connection

[21:27:14][D][esp-idf:000][http]: E (187294668) TRANSPORT_BASE: Failed to open a new connection

[21:27:14][D][esp-idf:000][http]: E (187294672) HTTP_CLIENT: Connection failed, sock < 0

[21:27:14][D][esp-idf:000][http]: E (187294675) HTTP_STREAM: Failed to open http stream

[21:27:14][D][esp-idf:000][http]: E (187294678) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,28674

[21:27:14][D][esp-idf:000][http]: W (187294680) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7

[21:27:14][D][adf_audio_element:324]: [http] Checking State for stopping, got 7
[21:27:14][D][esp-idf:000]: E (187294685) AUDIO_ELEMENT: [http] Element already stopped

[21:27:14][E][esp_adf_pipeline:223]: HTTPStreamReader got in error state while STOPPING. Stopping pipeline!
[21:27:14][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from ABORTING to DESTROYING. (REQ: 4)
[21:27:14][I][adf_media_player:192]: got new pipeline state: 11, while in MP state ANNOUNCING
[21:27:14][I][adf_media_player:256]: current mp state: ANNOUNCING
[21:27:14][I][adf_media_player:257]: anouncement: yes
[21:27:15][I][adf_media_player:258]: play_intent: false
[21:27:15][D][esp-idf:000]: I (187295633) AUDIO_PIPELINE: audio_pipeline_unlinked

[21:27:15][D][esp-idf:000]: W (187295635) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:27:15][D][esp-idf:000]: W (187295637) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:27:15][D][esp-idf:000]: W (187295639) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:27:15][D][esp-idf:000]: W (187295642) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[21:27:15][D][i2s_audio:117]: Other component hasn't released
[21:27:15][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 4)
[21:27:15][I][adf_media_player:192]: got new pipeline state: 0, while in MP state ANNOUNCING
[21:27:15][I][adf_media_player:256]: current mp state: IDLE
[21:27:15][I][adf_media_player:257]: anouncement: false
[21:27:15][I][adf_media_player:258]: play_intent: false
[21:27:15][I][adf_media_player:259]: current_uri_: false
[21:27:18][D][sensor:094]: 'S3 Temperature': Sending state 32.80830 °C with 2 decimals of accuracy
[21:27:18][D][sensor:094]: 'S3 Humidity': Sending state 62.26091 % with 2 decimals of accuracy
[21:27:28][D][binary_sensor:036]: 'Presence detect': Sending state ON
[21:27:30][D][binary_sensor:036]: 'Presence detect': Sending state OFF
[21:27:32][D][sensor:094]: 'WiFi db': Sending state -39.00000 dBm with 0 decimals of accuracy
[21:27:32][D][sensor:094]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[21:27:32][D][sensor:094]: 'up_sens': Sending state 187312.37500 s with 0 decimals of accuracy
[21:27:38][D][sensor:094]: 'battery_voltage': Sending state 0.02874 V with 1 decimals of accuracy
[21:27:38][D][sensor:094]: 'Battery level': Sending state 0.00000 % with 0 decimals of accuracy
```

yuuki94, 2024-10-17:
I changed my HA setup and retried. No more connection errors but it still does not play on my s3 box. 
I copied the local TTS file to my browser and it was playable.

Logs below

```
[00:05:43][D][binary_sensor:036]: 'Presence detect': Sending state OFF
[00:05:46][D][sensor:094]: 'S3 Temperature': Sending state 32.50580 °C with 2 decimals of accuracy
[00:05:46][D][sensor:094]: 'S3 Humidity': Sending state 61.81355 % with 2 decimals of accuracy
[00:05:48][D][media_player:061]: 's3-box_media_player' - Setting
[00:05:48][D][media_player:068]:   Media URL: http://192.168.0.165:8123/api/tts_proxy/69342c5c39e5ae5f0077aecc32c0f81811fb8193_en-us_894b186240_tts.piper.mp3
[00:05:48][D][media_player:074]:  Announcement: yes
[00:05:48][D][adf_media_player:057]: Got control call in state IDLE
[00:05:48][D][adf_media_player:058]: req_track stream uri: http://192.168.0.165:8123/api/tts_proxy/69342c5c39e5ae5f0077aecc32c0f81811fb8193_en-us_894b186240_tts.piper.mp3
[00:05:48][D][esp_adf_pipeline:060]: Starting request, current state STOPPED
[00:05:48][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from STOPPED to PREPARING. (REQ: 0)
[00:05:48][I][adf_media_player:192]: got new pipeline state: 3, while in MP state IDLE
[00:05:48][D][adf_i2s_out:141]: Set final i2s settings: 16000
[00:05:48][D][esp_audio_processors:124]: Current settings: SRC: rate: 24000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[00:05:48][I][adf_media_player:256]: current mp state: ANNOUNCING
[00:05:48][I][adf_media_player:257]: anouncement: yes
[00:05:48][I][adf_media_player:258]: play_intent: yes
[00:05:48][I][adf_media_player:259]: current_uri_: false
[00:05:48][D][esp_audio_sources:103]: Prepare elements called (initial_call)!
[00:05:48][D][esp_audio_sources:137]: Use fixed settings: yes
[00:05:48][D][esp_audio_sources:138]: Streamer status: 6
[00:05:48][D][esp_audio_sources:139]: decoder status: 6
[00:05:48][D][esp_audio_sources:140]: stream uri: http://192.168.0.165:8123/api/tts_proxy/69342c5c39e5ae5f0077aecc32c0f81811fb8193_en-us_894b186240_tts.piper.mp3
[00:05:48][D][adf_i2s_out:141]: Set final i2s settings: 16000
[00:05:48][D][esp_audio_processors:124]: Current settings: SRC: rate: 24000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[00:05:48][D][adf_audio_element:108]: Preparing [http]...
[00:05:48][D][adf_audio_element:108]: Preparing [decoder]...
[00:05:48][D][adf_audio_element:108]: Preparing [resampler]...
[00:05:48][D][adf_audio_element:108]: Preparing [i2s_out]...
[00:05:48][D][esp_audio_sources:193]: Preparation done!
[00:05:48][D][esp_adf_pipeline:342]: wait for preparation, done
[00:05:48][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[00:05:48][I][adf_media_player:192]: got new pipeline state: 5, while in MP state ANNOUNCING
[00:05:48][I][adf_media_player:256]: current mp state: ANNOUNCING
[00:05:48][I][adf_media_player:257]: anouncement: yes
[00:05:48][I][adf_media_player:258]: play_intent: yes
[00:05:48][I][adf_media_player:259]: current_uri_: false
[00:05:48][D][adf_audio_element:165]: Resuming [http]...
[00:05:48][D][adf_audio_element:172]: [http] Sending resume command.
[00:05:48][D][adf_audio_element:165]: Resuming [decoder]...
[00:05:48][D][adf_audio_element:172]: [decoder] Sending resume command.
[00:05:48][D][adf_audio_element:165]: Resuming [resampler]...
[00:05:48][D][adf_audio_element:172]: [resampler] Sending resume command.
[00:05:48][D][esp-idf:000][resampler]: I (8146135) AUDIO_ELEMENT: [resampler] AEL_MSG_CMD_RESUME,state:1

[00:05:48][D][adf_audio_element:165]: Resuming [i2s_out]...
[00:05:48][D][adf_audio_element:172]: [i2s_out] Sending resume command.
[00:05:48][D][esp-idf:000][i2s_out]: I (8146144) AUDIO_ELEMENT: [i2s_out] AEL_MSG_CMD_RESUME,state:1

[00:05:48][D][adf_audio_element:191]: [resampler] Checking State, got 79
[00:05:48][I][esp_adf_pipeline:132]: [ i2s_out ] status: 12
[00:05:48][D][adf_audio_element:191]: [i2s_out] Checking State, got 74
[00:05:48][I][esp_adf_pipeline:132]: [ resampler ] status: 12
[00:05:48][D][adf_audio_element:191]: [http] Checking State, got 79
[00:05:48][D][adf_audio_element:191]: [decoder] Checking State, got 79
[00:05:48][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[00:05:48][I][adf_media_player:192]: got new pipeline state: 6, while in MP state ANNOUNCING
[00:05:48][I][adf_media_player:256]: current mp state: ANNOUNCING
[00:05:48][I][adf_media_player:257]: anouncement: yes
[00:05:48][I][adf_media_player:258]: play_intent: yes
[00:05:48][I][adf_media_player:259]: current_uri_: false
[00:05:48][I][HTTPStreamReader:230]: Codec Format reported: 3.
[00:05:48][I][esp_adf_pipeline:132]: [ http ] status: 12
[00:05:48][I][esp_adf_pipeline:132]: [ decoder ] status: 12
[00:05:48][I][HTTPStreamReader:240]: [ * ] Receive music info from decoder, sample_rates=16000, bits=16, ch=1
[... 16 lines cut ...]

[00:05:49][D][esp-idf:000][decoder]: I (8146729) MP3_DECODER: Closed

[00:05:49][I][esp_adf_pipeline:123]: [ decoder ] byte_pos: 0, total: -1125
[00:05:49][I][esp_adf_pipeline:132]: [ decoder ] status: 15
[00:05:49][I][esp_adf_pipeline:135]: current state: FINISHING
[00:05:49][D][esp-idf:000][resampler]: I (8146824) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[00:05:49][I][esp_adf_pipeline:132]: [ resampler ] status: 15
[00:05:49][I][esp_adf_pipeline:135]: current state: FINISHING
[... cut ...]
[00:05:49][I][adf_media_player:256]: current mp state: IDLE
[00:05:49][I][adf_media_player:257]: anouncement: false
[00:05:49][I][adf_media_player:258]: play_intent: yes
[00:05:49][I][adf_media_player:259]: current_uri_: false
[00:05:54][D][sensor:094]: 'up_sens': Sending state 8151.25586 s with 0 decimals of accuracy
[00:05:55][D][sensor:094]: 'battery_voltage': Sending state 0.03694 V with 1 decimals of accuracy
[00:05:55][D][sensor:094]: 'Battery level': Sending state 0.00000 % with 0 decimals of accuracy
```

yuuki94, 2024-10-17:
The issue might be with ```[00:05:48][D][esp-idf:000][http]: W (8146209) HTTP_STREAM: No more data,errno:0, total_bytes:7938, rlen = 0``` ?

yuuki94, 2024-10-18:
i found the issue. The media was now playing but the speaker volume was set too low

Setting the volume above 80% was necessarily to hear even a soft voice.

closed 2024-10-18 by yuuki94

BigBobbas, 2024-10-18:
glad you managed to sort it 👍

citgot, 2025-02-11:
As I am experiencing the same problem but can't resolve it, I'm tagging along in this closed issue instead of starting a new one. I have the same errors as @yuuki94 initially had, but my URL is local. Full log from when i'm trying to send a TTS to the internal speaker of the ESP S3 Box3. External speaker works as it should. The audio file works from a browser.

```
[12:39:26][D][media_player:065]: 's3-box_media_player' - Setting
[12:39:26][D][media_player:072]:   Media URL: http://10.0.40.20:8123/api/tts_proxy/pRNodsSRb0BaRsktA7mzqA.mp3
[12:39:26][D][media_player:078]:  Announcement: yes
[12:39:26][D][adf_media_player:057]: Got control call in state IDLE
[12:39:26][D][adf_media_player:058]: req_track stream uri: http://10.0.40.20:8123/api/tts_proxy/pRNodsSRb0BaRsktA7mzqA.mp3
[12:39:26][D][esp_adf_pipeline:060]: Starting request, current state UNINITIALIZED
[12:39:26][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from UNINITIALIZED to INITIALIZING. (REQ: 0)
[12:39:26][I][adf_media_player:192]: got new pipeline state: 1, while in MP state IDLE
[12:39:26][I][adf_media_player:256]: current mp state: IDLE
[12:39:26][I][adf_media_player:257]: anouncement: yes
[12:39:26][I][adf_media_player:258]: play_intent: yes
[12:39:26][I][adf_media_player:259]: current_uri_: false
[12:39:26][D][esp-idf:000]: I (1046441) MP3_DECODER: MP3 init

[12:39:26][D][DAC_ES8311:227]: ES8311 chip found.
[12:39:26][D][esp-idf:000]: I (1046468) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[12:39:26][D][i2s_audio:067]: Install driver requested by Writer
[12:39:26][D][DAC_ES8311:256]: Setting up ES8311...
[12:39:26][D][esp_adf_pipeline:497]: pipeline tag 0, http
[12:39:26][D][esp_adf_pipeline:497]: pipeline tag 1, decoder
[12:39:26][D][esp_adf_pipeline:497]: pipeline tag 2, resampler
[12:39:26][D][esp_adf_pipeline:497]: pipeline tag 3, i2s_out
[12:39:26][D][esp-idf:000]: I (1046492) AUDIO_PIPELINE: link el->rb, el:0x3d04a684, tag:http, rb:0x3d04acdc

[12:39:26][D][esp-idf:000]: I (1046495) AUDIO_PIPELINE: link el->rb, el:0x3d04a83c, tag:decoder, rb:0x3d04bd1c

[12:39:26][D][esp-idf:000]: I (1046497) AUDIO_PIPELINE: link el->rb, el:0x3d04a9d8, tag:resampler, rb:0x3d04cd5c

[12:39:26][D][esp_adf_pipeline:507]: Setting up event listener.
[12:39:26][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from INITIALIZING to CREATED. (REQ: 0)
[12:39:26][I][adf_media_player:192]: got new pipeline state: 2, while in MP state IDLE
[12:39:26][I][adf_media_player:256]: current mp state: IDLE
[12:39:26][I][adf_media_player:257]: anouncement: yes
[12:39:26][I][adf_media_player:258]: play_intent: yes
[12:39:26][I][adf_media_player:259]: current_uri_: false
[12:39:26][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from CREATED to PREPARING. (REQ: 0)
[12:39:26][I][adf_media_player:192]: got new pipeline state: 3, while in MP state IDLE
[12:39:26][D][adf_i2s_out:141]: Set final i2s settings: 16000
[12:39:26][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[12:39:26][I][adf_media_player:256]: current mp state: ANNOUNCING
[12:39:26][I][adf_media_player:257]: anouncement: yes
[12:39:26][I][adf_media_player:258]: play_intent: yes
[12:39:26][I][adf_media_player:259]: current_uri_: false
[12:39:26][D][esp_audio_sources:103]: Prepare elements called (initial_call)!
[12:39:26][D][esp_audio_sources:137]: Use fixed settings: yes
[12:39:26][D][esp_audio_sources:138]: Streamer status: 1
[12:39:26][D][esp_audio_sources:139]: decoder status: 1
[12:39:26][D][esp_audio_sources:140]: stream uri: http://10.0.40.20:8123/api/tts_proxy/pRNodsSRb0BaRsktA7mzqA.mp3
[12:39:26][D][adf_i2s_out:141]: Set final i2s settings: 16000
[12:39:26][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[12:39:26][D][adf_audio_element:108]: Preparing [http]...
[12:39:26][D][esp-idf:000]: I (1046556) AUDIO_THREAD: The http task allocate stack on external memory

[12:39:26][D][esp-idf:000]: I (1046559) AUDIO_ELEMENT: [http-0x3d04a684] Element task created

[12:39:26][D][adf_audio_element:108]: Preparing [decoder]...
[12:39:26][D][esp-idf:000]: I (1046564) AUDIO_THREAD: The decoder task allocate stack on external memory

[12:39:26][D][esp-idf:000]: I (1046566) AUDIO_ELEMENT: [decoder-0x3d04a83c] Element task created

[12:39:26][D][adf_audio_element:108]: Preparing [resampler]...
[12:39:26][D][esp-idf:000]: I (1046573) AUDIO_THREAD: The resampler task allocate stack on external memory

[12:39:26][D][esp-idf:000]: I (1046575) AUDIO_ELEMENT: [resampler-0x3d04a9d8] Element task created

[12:39:26][D][adf_audio_element:108]: Preparing [i2s_out]...
[... 74 lines cut ...]
[12:39:44][I][adf_media_player:256]: current mp state: ANNOUNCING
[12:39:44][I][adf_media_player:257]: anouncement: yes
[12:39:44][I][adf_media_player:258]: play_intent: yes
[12:39:44][I][adf_media_player:259]: current_uri_: false
[12:39:44][D][esp_adf_pipeline:518]: Called deinit_all
[12:39:45][D][esp-idf:000]: I (1065712) AUDIO_PIPELINE: audio_pipeline_unlinked

[12:39:45][D][esp-idf:000]: W (1065714) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE

[12:39:45][D][esp-idf:000]: W (1065716) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[12:39:45][D][esp-idf:000]: W (1065719) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE

[... cut ...]
[12:39:45][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 4)
[12:39:45][I][adf_media_player:192]: got new pipeline state: 0, while in MP state ANNOUNCING
[12:39:45][I][adf_media_player:256]: current mp state: IDLE
[12:39:45][I][adf_media_player:257]: anouncement: false
[12:39:45][I][adf_media_player:258]: play_intent: yes
[12:39:45][I][adf_media_player:259]: current_uri_: false
```

Do you have any more suggestions or can find something in the log that i've missed?