BigBobbas/ESP32-S3-Box3-Custom-ESPHome #99: Wake up sound not playing

badnetmask, 2024-10-19:
Beforehand, let me say: this is a great project. I've been having a lot of fun customizing it. Thank you!

I can't get the wake up sound to play, and I'm not sure where to look.
I know the file has been copied correctly to HA because I can play it on the browser.
Also the conversation itself is working (I can send voice commands, they work, and I get voice responses).
It's just the wake up sound that is not playing.
Here's the piece of the log that I think is relevant.
Let me know if you need something else.
```
[18:57:13][I][esp_audio_sources:033][http]: Receive http event: 2
[18:57:13][I][esp_audio_sources:033][http]: Receive http event: 4
[18:57:13][D][esp-idf:000][http]: I (3186298) HTTP_CLIENT: Body received in fetch header state, 0x3fcd5adb, 788
[18:57:13]
[18:57:13][D][esp-idf:000][http]: I (3186300) HTTP_STREAM: total_bytes=2556
[18:57:13]
[18:57:13][I][HTTPStreamReader:230]: Codec Format reported: 3.
[18:57:13][D][esp-idf:000][http]: W (3186306) HTTP_STREAM: No more data,errno:0, total_bytes:2556, rlen = 0
[18:57:13]
[18:57:13][I][esp_audio_sources:033][http]: Receive http event: 7
[18:57:13][D][esp-idf:000][http]: I (3186309) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
[18:57:13]
[18:57:13][I][HTTPStreamReader:240]: [ * ] Receive music info from decoder, sample_rates=16000, bits=16, ch=1
[18:57:13][I][HTTPStreamReader:243]: [ * ] Receive music info from decoder, codec_fmt=3, bps=71000, duration=288, bytes=0
[18:57:13][D][adf_i2s_out:141]: Set final i2s settings: 16000
[18:57:13][D][esp_audio_processors:108]: Received request from: HTTPStreamReader
[18:57:13][D][esp_audio_processors:113]: New settings: SRC: rate: 16000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[18:57:13][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 1 bits: 16, DST: rate: 16000, ch: 2, bits 16
[18:57:13][D][adf_audio_element:108]: Preparing [http]...
[18:57:13][D][adf_audio_element:108]: Preparing [decoder]...
[18:57:13][D][esp-idf:000][decoder]: W (3186329) AUDIO_ELEMENT: OUT-[decoder] AEL_IO_ABORT
[18:57:13]
[18:57:13][D][esp-idf:000][decoder]: W (3186330) MP3_DECODER: output aborted -3
[18:57:13]
[18:57:13][D][esp-idf:000][decoder]: I (3186332) MP3_DECODER: Closed
```

BigBobbas, 2024-10-19:
Hi,
do you have the sound files in config/www/sounds ?
also at the top of your device config have you set the url to your HA instance ?

Many Thanks
Rob

badnetmask, 2024-10-20:
Yes, this is all there. I can see the URL to the file on the logs, so I copy it then paste on the browser to make sure it works correctly, and it does.

jsiemonski, 2024-10-28:
@badnetmask one thing I just noticed on mine, the downloaded file is called sounds_awake.mp3 but the configuration is looking for awake.mp3. Renaming the file made it stop complaining, but I am still stuck getting audio out of it.

badnetmask, 2024-10-28:
Yeah, that's really not my case. The box is able to download the file, just doesn't play it.

BigBobbas, 2024-11-01:
> Hi @BigBobbas, is there any other information I can provide to help debug this? I am also on Discord, if that helps.

Hi there , 
if you go to 'media' from the HA sidebar and select the s3 box media player in the bottom right. are you able to play media to the box from HA?  feel free to post in our discord and ping me. https://discord.gg/wHcKFF8E

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas

badnetmask, 2025-03-06:
For posterity (in case someone finds this in the future): the recent update to 2025.2.0 made it work.