BigBobbas/ESP32-S3-Box3-Custom-ESPHome #103: Unable to get TTS from s3-box-3's internal speaker

thaynes43, 2024-11-06:
Hello,

Very happy I found this repo, it's a huge upgrade from the one on HA's page. I've got everything running and connected to home assistant but I'm unable to get responses out of the s3-box's internal speaker when "output audio externally" is toggled off. I did a fresh install of this to verify if was none of my customizations which is included below. I also tried a fresh install of that is on the home assistant guide and that basic version did play exclusively out of the internal speaker. 

Looking for next steps in troubleshooting. I am using the base with the sensors and battery. When I check "Output audio externally" the volume bar of the s3-box_media_player goes to full and when unchecked it goes empty. One interesting thing is I can play other sources through the S3-box_media_player like BBC radio.

Config here since it is too large for the issue -> https://pastebin.com/DK1wM3PQ

BigBobbas, 2024-11-06:
Hi there ,
Sorry that you are having an issue with this. I will take a look and see if I can figure out what is going on, and get back to you. 👍 

Many Thanks
Rob

BigBobbas, 2024-11-06:
I've just tested and when I switch 'Ext media player' to on the S3 Box media player volume level is set to 0 and vice versa so with it off it is set to the defauly level. Yours from your description appears to be backwards ... not quite sure why. do you have 'Mute Responses' turned off under 'Voice' in settings?

thaynes43, 2024-11-06:
> I've just tested and when I switch 'Ext media player' to on the S3 Box media player volume level is set to 0 and vice versa so with it off it is set to the defauly level. Yours from your description [...]

Oh sorry, I wrote that backwards. The behavior is as expected where the volume drops to 0 when I go external and back to 100. The internal speaker did work initially, but I wasn't using the sensor dock before. It was also extremely quiet so my first move was to add a bit to increase the volume which seemed to work. That was before I found this project which was way better. Maybe it broke somehow when I was trying to increase the volume, I can try to go back to the vanilla firmware or install the firmware I pastbin'd above on my second S3-Box3. Will let you know how it goes.

Thanks!

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas