BigBobbas/ESP32-S3-Box3-Custom-ESPHome #104: Can't connect to ESPHome API

Gazda86, 2024-11-07:
please see the error in the picture. I would like to find a solution for this, the device is offline, does not connect to wifi. the wifi access is correct in the secret.yaml. Thank you in advance
[image]

BigBobbas, 2024-11-07:
hi there , if you look in your routers interface do you see the device connected? and if so can you make a note of the IP address ?  andd add it to the config like so. (under wifi: replacing the xxx's)
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address:  xxx.xxx.xxx.xxx
  ```

Gazda86, 2024-11-07:
[image]
until the installation, I can see the device on the router as well, it connects to the wifi. after installation, unfortunately, it is already connected to wifi.
[image]
[image]

BigBobbas, 2024-11-07:
from https://web.esphome.io connect the device using usb, then select 'logs' when the blank log screen appears, click 'reset this device' at the bottom and you should see if it tries to connect to wifi and any possible errors.

Gazda86, 2024-11-08:
[image]

it can't connect to wifi, but my network can see it. the log file in the picture.

BigBobbas, 2024-11-08:
can you confirm that your secrets file looks like this and that your credentials are correct including any upper and lower case letters etc.

```yaml
# Your Wi-Fi SSID and password

wifi_ssid: "your_ssid"
wifi_password: "your_password"
```
is your wifi ssid configured to be hidden?
does the ssid have any spaces or charecters in it ?

Gazda86, 2024-11-08:
the ssid is hidden, yes, Gazda 2,4 but I have already tried Gazda 2.4.

BigBobbas, 2024-11-08:
can you change the `wifi:` section in your config to the following
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  ```

sudoxnym, 2024-11-30:
I'm gonna chime in because I spent 4 hours on this last night.
steps to resolve
1. find device mac
2. set static ip in router
3. find your wifi setting in the yaml and replace with
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: [same static ip]
    gateway: [router ip]
    subnet: 255.255.255.0
4. install esptool.py
5. python -m esptool --port COM11 erase_flash [replace COM11 with the proper port]
6. start over
7. if you don't see the wifi options on the install page after install - REBOOT using the lower left button or hotplug it.
  7.a. install again [wifi setup should appear now]
8. profit

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas