BigBobbas/ESP32-S3-Box3-Custom-ESPHome #106: model prompt not working in version 2024.09.01?

Makrokaba, 2024-11-27:
Hi,

I have two ESP32-S3-Box-3 devices, one is running BigBobbas version 2024.06.01, the other one is flashed with BigBobbas version 2024.09.01.

Both are connected via ESP-Home to Home assistant 2024.11.3.
Both are connected to the same home assistant - voice assistant.
The voice assistant is setup to use Ollama via the Home-LLM plugin with a model prompt that supplies the LLM with date, time and home assistant entities and stati.

So now when I ask both boxes about the current time, the old version gives me the correct time and the newer version doesn't know anything that was supplied in the model prompt. 

Any helpful hints? I'm running out of ideas...

BigBobbas, 2024-11-27:
Hi , 
Thank you for your support and using the project. Unfortunatley I don't have any knowledge of Ollama or LLM's. 
Do the 2 devices have unique names? This will be at the top of the config. If they  don't have different names then this could be the issue.  You would need to change the name of one of the devices and flash via USB. remove both devices from the HA integration and then add back in. The default name is shown below
[image]

BigBobbas, 2024-12-21:
closed due to age and no recent update

closed 2024-12-21 by BigBobbas