BigBobbas/ESP32-S3-Box3-Custom-ESPHome #113: = [FAILED] Took 240.17 seconds ====

atuhim, 2024-12-13:
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/circular_buffer.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/circular_buffer_common.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/comparisons.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/concatenation.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/conv_common.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/cumsum.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/depth_to_space.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/depthwise_conv_common.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/dequantize.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/dequantize_common.o
Compiling .pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/detection_postprocess.o
*** [.pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/div.o] Source `components/esp-tflite-micro/tensorflow/lite/micro/kernels/div.cc' not found, needed by target `.pioenvs/esp32-s3box-3/components/esp-tflite-micro/tensorflow/lite/micro/kernels/div.o'.
========================= [FAILED] Took 240.17 seconds =========================

BigBobbas, 2024-12-13:
what hardware do you have ESPHome installed on? and how much RAM does it have ? What is the version number of ESPHome you have installed ?

atuhim, 2024-12-13:
Hi,
Intel nuc I5, 16GB Ram, 2024.11.3
Thank you for helping me.

BigBobbas, 2024-12-13:
Is it a bare metal install, or container , VM ? ESPHome will need at least 8gb ram dedicated for this to compile comfortably.

atuhim, 2024-12-13:
Bare MetalVon meinem iPhone gesendet

BigBobbas, 2024-12-13:
Thank you I will do some tests to see if this is an issue with 2024.11.3

atuhim, 2024-12-13:
this is working but I like your version better
https://www.home-assistant.io/voice_control/s3_box_voice_assistant/

BigBobbas, 2024-12-13:
This compiles for me. Try uninstalling and re-installing the esphome add-on

https://www.home-assistant.io/voice_control/s3_box_voice_assistant/

atuhim, 2024-12-15:
Hi,
it's compiled.
On the display appears ESPHome S3 BX3 Loading

INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating ***@***.***
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Updating ***@***.***
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for
***@***.***
INFO Updating ***@***.***
INFO Updating ***@***.***
INFO Updating ***@***.***
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform:
***@***.***)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8)
 - tool-cmake @ 3.16.4
 - tool-ninja @ 1.7.1
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ESPMicroSpeechFeatures @ 1.1.0
Compiling .pioenvs/esp32-s3box-3/src/main.o
Linking .pioenvs/esp32-s3box-3/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld:
missing --end-group; added as last command line option
RAM:   [=         ]  11.7% (used 38268 bytes from 327680 bytes)
Flash: [=====     ]  47.1% (used 3829697 bytes from 8126464 bytes)
Building .pioenvs/esp32-s3box-3/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/esp32-s3box-3/firmware.bin"],
[".pioenvs/esp32-s3box-3/firmware.elf"])
Wrote 0x3b7130 bytes to file
/data/build/esp32-s3box-3/.pioenvs/esp32-s3box-3/firmware.factory.bin,
ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/esp32-s3box-3/firmware.bin"],
[".pioenvs/esp32-s3box-3/firmware.elf"])
======================== [SUCCESS] Took 148.94 seconds
========================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.179 port 3232...
INFO Connected to 192.168.178.179
INFO Uploading
/data/build/esp32-s3box-3/.pioenvs/esp32-s3box-3/firmware.bin (3830064
bytes)
Uploading: [============================================================]
100% Done...

INFO Upload took 80.96 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.178.179 using esphome API
WARNING Can't connect to ESPHome API for esp32-s3box-3 @ 192.168.178.179:
Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>,
type=<SocketKind.SOCK_STREAM: 1>, proto=6,
sockaddr=IPv4Sockaddr(address='192.168.178.179', port=6053))]
(TimeoutAPIError)
INFO Trying to connect to esp32-s3box-3 @ 192.168.178.179 in the background

BigBobbas, 2024-12-15:
Hi, have you added the box to the HA ESPHome integration in HA>>settings>>devices and services ?

atuhim, 2024-12-15:
Yes, please check the screenshots Von meinem iPad gesendet

BigBobbas, 2024-12-15:
sorry, i don't see any screenshots.
It may be easier to troubleshhot this if you joined us on Discord ? https://discord.gg/wdw9MKZH

closed 2025-04-01 by BigBobbas