BigBobbas/ESP32-S3-Box3-Custom-ESPHome #114: update

JanKraslice, 2024-12-13:
I have a device 
project:
    name: "BigBobbas.s3box"
    version: "2024.06.01"

How please the update of the release itself from you for example you state from October: V1.10.24.1

A few weeks ago the timer option was added as part of the classic voice control installation, will it be added automatically to your BigBobbas.s3box project? 

Thank you for your advice

BigBobbas, 2024-12-13:
Hi Timers are already included in the firmware https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b.yaml

JanKraslice, 2024-12-13:
That's why I asked for an update I got the version from you from when there was no timer.

 

Now the timers are already there how do I get them or how do I update?

JanKraslice, 2024-12-13:
What's the easiest thing to do with

project:

    name: "BigBobbas.s3box"

    version: "2024.06.01"

 

on the new one where you already have timers

 

project:

    name: "BigBobbas.s3box"

    version: "2024.09.01"

BigBobbas, 2024-12-13:
Hi the best way is to copy and paste the new yaml into a text editor.
Then copy any customised parts of your existing yaml into the new yaml file overwriting the defaults. 
Then you can paste the entire 'new' edited yaml into the device card in the ESPHome dashboard and install.

JanKraslice, 2024-12-13:
I'll give it a try.

Suppose I don't want to keep any of my settings, but I don't want to lose the already created link in ESPHome or add a re-create key etc.

 

So can I copy all the code from the first line or should I keep some of the first lines?

Tobster77, 2024-12-19:
Hi,

maybe this is a dumb question as the instructions above were quite clear (just copy the new s3b.yaml file for updates):

The github archive also contains other resources such as sounds or images... in case of updates: How should I receive them?

I remember there was a new feature of ESPhome that updates should be pulled from the source code on github automatically. Is this already working? There is an "Update" button on the S3 box when checking my ESPhome addon homepage - what does it do? Is this the automatic update feature?

Thanks,
Tobias

BigBobbas, 2024-12-19:
Hi @Tobster77 
the sounds and images will remain the same , images will be compiled in automatically and any changes to sounds will currently need to be manually copied to HA although there are no current changes to these.
Updates do need to be added manually as the automated method is not suited to this firmware as it currently stands.
The update button that you see in the add-on is for the core ESPHome component and is safe to upload if you wish, however this is not neccessary. I will always advise of any New Features or changes that need to be made to coinside with an ESPHome update. so have a look through the discussions section or on our discord server as information will be made available there.
I hope this answers your questions and if you need any help at all with your device please join us on discord  https://discord.gg/wdw9MKZH
Many Thanks & Seasons Greetings
Rob

closed 2025-04-01 by BigBobbas