BigBobbas/ESP32-S3-Box3-Custom-ESPHome #117: ESPHome Update 2024.12.0 - Failing to compile

TheSerapher, 2024-12-18:
**Describe the bug**
When trying to update my box using the latest version of ESPHome  it's running into a CMake error which I am not sure how to resolve on my end.

**To Reproduce**
Try to update your screen with the latest ESPHome version, fails every time even with cleaned build files.

**Expected behavior**
Build as usual.

**Desktop (please complete the following information):**
 - HA OS
 - Safari
 - 18.2

**Additional context**
```
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
/data/cache/platformio/packages/framework-espidf Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/esp32-s3box-3/.pioenvs/esp32-s3box-3/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)
```

BigBobbas, 2024-12-18:
Hi there ,
there is a pinned post in the discussions [here](<https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions>) that will resolve this.
Hope that helps 
Cheers

Rob

TheSerapher, 2024-12-18:
Dang, you are fast 🙂 Sorry for not checking before, checked issues but forgot to glance at the discussions. Will test and close this if it works!

closed 2024-12-18 by TheSerapher