BigBobbas/ESP32-S3-Box3-Custom-ESPHome #118: No Adopt and install, and config will not compile

YoshiFan501, 2024-12-21:
**Describe the bug**
There is no adopt and install on the dashboard, and using take control, and then uploading the yaml on the discovered device does not work.

**Expected behavior**
Successful Compile and / or showing up in the esphome dashboard

**Desktop (please complete the following information):**
 - OS: Homeassistant 
 Core: 2024.12.5
Supervisor: 2024.12.0
Operating System: 14.1
Frontend : 20241127.8

**Additional context**
Log

`INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 213, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 231, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 341, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 307, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 798, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 111, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: Unable to find current revision in submodule path 'components/esp-adf-libs'`

BigBobbas, 2024-12-21:
Hi there , thank you for pointing this out. there has been a recent change to ESPHome and 'take control' has replaced the previous 'adopt' naming.

The best and quickest way, would be to run ESPHome in ubuntu using WSL in windows. You can also add the s3b.yaml file into the dashboard by copying the file to HA /config/esphome the device card should then appear in the dashboard.

If you need further help with this , the easiest way to help troubleshoot is to join us on discord https://discord.gg/wdw9MKZH or alternatively reply here 👍

YoshiFan501, 2024-12-21:
Switched to ubuntu

Got different errors, this time with the YAML

(venv) ***@***.***:/mnt/c/Users/nickk/Desktop/s3box$ esphome run s3b.yaml
INFO Reading configuration s3b.yaml...
INFO Updating ***@***.***
Failed config

esp32: [source s3b.yaml:80]
  board: esp32s3box

  [flash_size] is an invalid option for [esp32]. Please check the
indentation.
  flash_size: 16MB
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_ESP32S3_DATA_CACHE_64KB: y
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: y
      CONFIG_AUDIO_BOARD_CUSTOM: y
      CONFIG_ESP32_S3_BOX_3_BOARD: y

    [components] is an invalid option for [framework]. Please check the
indentation.
    components:
      - name: esp32_s3_box_3_board
        source: ***@***.***
        refresh: 0s
(venv) ***@***.***:/mnt/c/Users/nickk/Desktop/s3box$

BigBobbas, 2024-12-21:
can you post the esp32: block from your editor. and format it as code by typing ```yaml on the line before the code followed by the same three ` ` ` after the code (no spaces)

YoshiFan501, 2024-12-21:
```yaml
esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
      CONFIG_ESP32_S3_BOX_3_BOARD: "y"
    components:
      - name: esp32_s3_box_3_board
        source: github://jesserockz/esp32-s3-box-3-board@main
        refresh: 0s
   ```

BigBobbas, 2024-12-21:
thats really strange, everything looks fine and compiles for me. what version of ESPHome is installed in ubuntu? and what is the editor that you are using ?
`esphome version`
can you attatch the full yaml file that you are compiling so i can check for other possible issues.

YoshiFan501, 2024-12-21:
it installed Version: 2022.9.4 and im trying to upgrade it (it keeps failing on unicodedata2) (and also set default python to 3.13 to no avail)

BigBobbas, 2024-12-21:
That will be the issue 👍 
how are you installing?
`pip install esphome==2024.12.1` should work

YoshiFan501, 2024-12-21:
i ran that and got the same error

Building wheels for collected packages: unicodedata2
  Building wheel for unicodedata2 (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for unicodedata2 (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      /tmp/pip-build-env-m48dvnwk/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'test_suite'
        warnings.warn(msg)
      running bdist_wheel
      running build
      running build_ext
      building 'unicodedata2' extension
      creating build/temp.linux-x86_64-cpython-313/unicodedata2
      x86_64-linux-gnu-gcc -pthread -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./unicodedata2/ -I/mnt/c/Users/nickk/Desktop/s3box/venv/include -I/usr/include/python3.13 -c ./unicodedata2/unicodectype.c -o build/temp.linux-x86_64-cpython-313/./unicodedata2/unicodectype.o
      error: command 'x86_64-linux-gnu-gcc' failed: No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for unicodedata2
Failed to build unicodedata2
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (unicodedata2)
(venv) nick@Pico:/mnt/c/Users/nickk/Desktop/s3box$

YoshiFan501, 2024-12-21:
i didn't have an  updated gcc

BigBobbas, 2024-12-21:
is it working now ?

YoshiFan501, 2024-12-21:
I have successfully flashed it! Now it's just sitting on connecting to
Wi-Fi and I don't know how long that's going to take. My secrets file is
formatted correctly and named correctly so hopefully it doesn't take too
long

BigBobbas, 2024-12-21:
it should be fairly quick tbh 
check to see if your router has given it an ip

YoshiFan501, 2024-12-21:
I got it to work! thank you so much! Last bug is to see why the assistant voice sounds like a chipmunk

BigBobbas, 2024-12-21:
change the sample rate in the substitutions at the top of the config to match your tts service :)
`  tts_voice_speed: "24000" #options #nabu casa "24000" #piper "16000" elevenlabs "44100" must include quotes`

YoshiFan501, 2024-12-21:
Thank you so much! I'm excited to tinker with this

BigBobbas, 2024-12-21:
great stuff, glad you're up and running , have fun and you know where to find us if you need any help 👍

boivinx7, 2025-01-01:
All works with ESPHome 2024.11
Its not the first one that doesnt work with .12

BigBobbas, 2025-01-02:
> All works with ESPHome 2024.11 Its not the first one that doesnt work with .12

it shuld work fine in 2024.12.x provided that you change the esp-idf version as described here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/116

closed 2025-04-01 by BigBobbas