BigBobbas/ESP32-S3-Box3-Custom-ESPHome #120: Delay for new wake word trigger

andreaconfa, 2025-01-11:
Hi, 
i noticed that, after a reply from the voice assistant, the screen turns back to idle but the actual wake word detection is not starting.
In fact i have to wait for about 10 seconds before trigger again the voice assistant.

This issue is not present in the "official" esphome voice assistant project.

andreaconfa, 2025-01-11:
> in the device configuration page, if you issue a command, does the assist satellite sensor return to idle quicker ?

No, it stays to responding for a while even if the response is ended

BigBobbas, 2025-01-11:
what service are you using for TTS ? and are you using the default conversation agent?

andreaconfa, 2025-01-11:
I’usig google translate tts
Inviato da iPhone

BigBobbas, 2025-01-15:
what happens if you test with a different tts service ?

closed 2025-04-01 by BigBobbas