BigBobbas/ESP32-S3-Box3-Custom-ESPHome #124: Cannot install a second ESP-S3-Box

janaxhell, 2025-02-05:
I have HA running on a RPi 4 - 8Gb. I have successfully installed a ESP32-S3Box wirelessly via ESPHome builder, but now I cannot  do the same with a second unit on the same machine/network. I have plugged it to my Windows PC, which is a i7-3770K 32 Gb and added the second unit as new device. But when I start install, it says it's offline. I have forced the IP that I found on the router as 

```
  manual_ip:
    static_ip: 192.168.1.196
    gateway: 192.168.1.1
    subnet: 255.255.255.0
```

Now it says it's online, but the yaml hasn't changed, all the info that were added to the yaml of the first unit have not been added to this one.
The log of the install is this:

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/voice-assistant-salotto.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing voice-assistant-salotto (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/voice-assistant-salotto/src/main.cpp.o
Linking .pioenvs/voice-assistant-salotto/firmware.elf
RAM:   [=         ]  12.5% (used 40904 bytes from 327680 bytes)
Flash: [=====     ]  46.5% (used 853585 bytes from 1835008 bytes)
Building .pioenvs/voice-assistant-salotto/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/voice-assistant-salotto/firmware.bin"], [".pioenvs/voice-assistant-salotto/firmware.elf"])
Wrote 0xe07d0 bytes to file /data/build/voice-assistant-salotto/.pioenvs/voice-assistant-salotto/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/voice-assistant-salotto/firmware.bin"], [".pioenvs/voice-assistant-salotto/firmware.elf"])
========================= [SUCCESS] Took 37.38 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.196 port 3232...
INFO Connected to 192.168.1.196
INFO Uploading /data/build/voice-assistant-salotto/.pioenvs/voice-assistant-salotto/firmware.bin (853968 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.25 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

janaxhell, 2025-02-05:
Thank you! I think I assumed the IP showing  on the router was from the second unit, but it was most likely from the first and I have flashed it with all the wrong stuff. The procedure you are suggesting is very different from the one I followed here 
https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/instructions/make%20it%20your%20own.md
anyway I'll follow your guide.

janaxhell, 2025-02-05:
Bumping here just in case you don't receive notifications for edits.

BigBobbas, 2025-02-05:
taking a look

BigBobbas, 2025-02-05:
just tested and this compiles fine for me using the config here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b.yaml
make sure that you remove everything from the device card firs or highlight the entire existing code prior to pasting this config in

janaxhell, 2025-02-05:
Mmm, I have done that twice and even rebooted. Please keep in mind that the first unit was bricked, maybe I have to reset/format it?

BigBobbas, 2025-02-05:
the firmware is compiled prior to uploading to the device, so this makes no difference. It should still compile regardless. do you have the latest version of esphome installed 2024.12.4 i think we are on

janaxhell, 2025-02-05:
Yes, 2024.12.4. I have installed the whole HA about a month ago.

janaxhell, 2025-02-05:
Ok, got it: copying the raw yaml via Windows broke encoding. I used Total Commander and this time I could compile and download Factory. I could flash only with Chromium, as neither Vivaldi, nor Firefox could connect.

janaxhell, 2025-02-06:
So, I managed to flash both units, but while one was discovered, the other wasn't.
They both appear as Online in ESPHome Builder
Only one appears as Device in ESPHome Integration and if I try to add a new device, I'm asked for an IP of ESPhome node, which I hadn't to set for the other unit, nor I know which is.
Tried to reboot, and also plug the second unit directly to the Raspberry where HA runs with no effect.
The units are both working, I can go through the menu in the touchscreen.
The only thing I changed in the YAMLs is name and friendly name.
How do I force discovery?

BigBobbas, 2025-02-06:
if you go into settings menu and info do they both have different hostname. you can also get the ip address from here

janaxhell, 2025-02-06:
I was about to post that this is the strange thing appearing in Devices:

[image]

For some reason the unit is misconfigured as my router. I have no idea why this happened, what should I do?

BigBobbas, 2025-02-06:
can you tell me what hostname is shown on each device and the ip address as per https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/issues/124#issuecomment-2639340085

BigBobbas, 2025-02-06:
there is no API key unless you specically add one to the ESPHome config 

using the device ui and looking at the info page will always give the correct status of the device, as will tapping the wifi icon on the top bar.

janaxhell, 2025-02-06:
Ok, thanks for all the help!

closed 2025-04-01 by BigBobbas