BigBobbas/ESP32-S3-Box3-Custom-ESPHome #127: Failed config

bduijnhouwer, 2025-02-19:
After updating ESPHome, I cannot install anymore.
I receive the following errors:

```
Failed config

image: [source /config/esphome/my-satellite.yaml:1120]
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/error.png
    id: error_img
    resize: 320x240
    
    Image type RGB24 is removed; replace with 'type: RGB'.
    type: RGB24
    
    [use_transparency] is an invalid option for [image]. Did you mean [transparency]?
    use_transparency: True
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/idle.png
    id: idle_img
    resize: 320x240
```

ndelahun, 2025-02-19:
I changed the RGB24 to RGB in all cases
There is also an issue with use_transparency: true line to transparency: alpha_channel
that got past the initial errors but now I get the below when compiling

`
Generating assembly for .pioenvs/esp32-s3box-3/duer_profile.S
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor_common.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor_esp32.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor_esp8266.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor_libretiny.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adc/adc_sensor_rp2040.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_element.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_process.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sinks.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sources.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_pipeline.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_pipeline_controller.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/media_player/adf_media_player.o
Compiling .pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/microphone/esp_adf_microphone.o
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp: In member function 'virtual void esphome::esp_adf::ADFMediaPlayer::control(const esphome::media_player::MediaPlayerCall&)':
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_ENQUEUE' not handled in switch [-Werror=switch]
     switch (call.get_command().value()) {
            ^
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_ONE' not handled in switch [-Werror=switch]
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_OFF' not handled in switch [-Werror=switch]
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_CLEAR_PLAYLIST' not handled in switch [-Werror=switch]
cc1plus: some warnings being treated as errors
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/media_player/adf_media_player.o] Error 1
========================= [FAILED] Took 101.58 seconds =========================

`

BigBobbas, 2025-02-19:
Hi guys, there are a couple of changes required in the latest ESPHome version, however .. updating will not give any benefits currently so it is just as well to not update for the sake of it. As there is a new config due to drop any time soon for the S3 box stock firmware to utilise the new media components etc I have held back from making any changes until this has been completed, to avoid updating and then having to re-work the update ;).
The new media player etc will mean that we will no longer rely on the current ASDF external component and all components will be core ESPHome. 

Obviously if you are going to be making  your own changes to impliment new features in this ESPHome release then you will need to make the neccessary changes as per the release notes and any ESPHome related documents.

Cheers 

Rob

BigBobbas, 2025-02-20:
I am currently working on it, having a few issues so currently trying to troubleshoot 👍

smartmatic, 2025-02-20:
I have already updated to the latest ESP Home Version and run into the same issue as initially reported. Except downgrading ESP Home Builder i habe now no chance to complete the ESP32 S3 Box3 setup until a new release is available right?

youkorr, 2025-02-20:
what problem do you have smartmatic

BigBobbas, 2025-02-21:
please see https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/128 👍

BigBobbas, 2025-02-21:
> I have already updated to the latest ESP Home Version and run into the same issue as initially reported. Except downgrading ESP Home Builder i habe now no chance to complete the ESP32 S3 Box3 setup until a new release is available right?

if you have downgraded and running 2024.12.x then you should still be able to compile and install 👍

smartmatic, 2025-02-21:
> what problem do you have smartmatic

This one like the TC has

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/esp32-s3-box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
Failed config

image: [source /config/esphome/esp32-s3-box-3.yaml:1120]
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/error.png
    id: error_img
    resize: 320x240
    
    Image type RGB24 is removed; replace with 'type: RGB'.
    type: RGB24
    
    [use_transparency] is an invalid option for [image]. Did you mean [transparency]?
    use_transparency: True
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/idle.png
    id: idle_img
    resize: 320x240

smartmatic, 2025-02-21:
> if you have downgraded and running 2024.12.x then you should still be able to compile and install 👍

I've already updated all my ESP Devices to latest the latest ESP Home version therefore i won't downgrade.

youkorr, 2025-02-21:
https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/tree/dev

ajtatum, 2025-02-25:
I'm having issues with the firmware from the dev branch. The device turns on and shows connected to the WiFi, but in ESPHome, it shows as offline.

There's also this warning in Repairs:

```
The ESPHome in progress binary sensor binary_sensor.bedroom_home_assistant_voice_assist_in_progress is deprecated.

Migrate your configuration to use the corresponding assist_satellite entity and then click SUBMIT to disable the in progress binary sensor and fix this issue.
```

Attached are logs from the device.

Any help would be greatly appreciated!

[esp32-s3-box-3-01_logs.txt](https://github.com/user-attachments/files/18952173/esp32-s3-box-3-01_logs.txt)

BigBobbas, 2025-02-25:
There are a few reasons why the ESPHome dashboard would show as offline, the main one being mDNS. The device should be reachable by pinging `esp32-s3box-3-01.local` 
It can also be down to device naming, if you have more than 1 device with the same name. If HA and your ESPHome server are on different subnets or containers this may also be an issue. 
In order to get the device showing online so that you can edit and upload firmware then you can add the following into the device config and save. (the use_address line with your ip for the device)
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.1.122
```

You don't need to upload this change as it is simply an instruction to let the server know the address of the device so it can see it via ip.
With regards to the repair this is just to advise that the sensor that reports the voice status of the device has been updated and the old standard has been changed, if you have any automations using that sensor then change them to use the new sensor. This sensor entity is automatically added by HA to voice satellites. You can click submit to clear the repair.

I hope that helps

Cheers 

Rob

smartmatic, 2025-02-25:
Question to the round:
When i play media i am not able to control the media player via voice e.g. pause the music.
Does anybody experience the same behaviour?

closed 2025-04-01 by BigBobbas