BigBobbas/ESP32-S3-Box3-Custom-ESPHome #132: Upgradability with customized YAML data

TheSerapher, 2025-03-02:
Hey @BigBobbas ,

With your recent work to get the newest ESPHome version to work on your dev branch I realized how hard it is to keep my own customized version of your base code updated.

Here the things I have adjusted which made updating quite hard and a copy and paste galore to get your diffs into my yaml:

* Removed some pages I didn’t use
* added new pages and subpages for solar sensors
* Added all required sensors for those pages

Overall these few changes moved code around quite extensively so simply applying patches wasn’t possible without manual work. I am not familiar with ESPhome and how yaml files could be included to import more customizations but maybe you have an idea how updates could be applied for heavily customized yamls.

If there isn’t a good way I’ll continue copy and pasting things, I was hoping for some „include these yaml for sensors“ and even a „this game describes pages“. That would split customizable code from your base code that needs upgrading 🙂 

Thanks for you great work! I love my little device and it has all infos I’d could hope for at a glance or tap of a button!

TheSerapher, 2025-03-03:
Thirst sounds great! I actually am Ok with the current method of checking commits and fixing it manually but I wasn’t even sure there’s a better way. Packages does sound like a logical next step with this project. I’ll patiently wait for this to be available (breaking changes are also far and few between and often easily fixed 😉).

If you need a tester let me know!

badnetmask, 2025-03-06:
I'm on the same boat here: lots of customization, hard to keep track. I am also offering to help with testing.

raetha, 2025-03-22:
@BigBobbas You mentioned wanting to move to LVGL. I just finished most of that effort with my version based on yours. If helpful, you can find it at: https://github.com/raetha/esphome-configs/blob/main/esp32s3box3/config.yaml

Note that I'm not 100% happy with the Settings tabview widget as there is a bar at the bottom I haven't figured out how to get rid of which forces scrolling on some of the tabs. The media player is also not currently working for anything aside from announcements and I haven't figured out why. Due to those and other testing, the config is not clean and I still have a lot of commented out stuff that likely will go away.

Unfortunately for me, it doesn't lend towards breaking the configuration up too much though, as I had to embed LVGL commands into a lot of entities and other places to keep values in sync, or adjust the screen from HA changes. So I could break things up, but they are very dependent on being updated together. I do find it easier to comprehend what is going on, and 95% of the display stuff is now under the LVGL key.

Figured I'd share though, as I have many hours in on learning LVGL and converting things over.

BigBobbas, 2025-03-23:
Hi,
thanks for this, I will give it a flash and a test and see how it goes.  I do like working with LVGL and find it much easier to construct without having to create seperate binary sensors for every touch point.

I'll let you know how I get on over the next day or so.

Cheers

Rob

youkorr, 2025-03-23:
the bar at the bottom comes from the size, it needs to be reduced a little but overall it's very good

BigBobbas, 2025-03-23:
The media_player can't currently recieve audio to play due to their being no duplex support using native ESPHome speaker and media components. The mic cannot be listening at the same time as media playing, in order to overcome this you need to stop micro_wake_word: before media_player: will play.

I have been trying to figure this out with my config, but as it uses mww or streaming ww it makes it a little more complex in order to get it to work under all possible parameters defined by the user. 

However with a mww only config it should be much simpler.

raetha, 2025-03-24:
@youkoor do you know which size needs to be reduced to eliminate the bar? That seems easy, but the current sizes should match the screen dimensions exactly, unless I misread something.

@BigBobbas that makes sense about the media player, what confuses me is the fact that announcements work, but media doesn't. Since those both come from HA, I would think they'd hit the same issue if MWW wasn't stopped first. However the only places that are stopping MWW right now are the mute switch, and the timer ringing switch.

youkorr, 2025-03-24:
Hi, is the bar in the menu?

youkorr, 2025-03-24:
I had the same problem too when I put the real size of esp which was 320X240 I had these bars at the bottom or on the right so reducing them to 235X115 is the most bar

raetha, 2025-03-25:
> Hi, is the bar in the menu?

The bar is at the very bottom of the tabview, across from the final tab icon. It's probably 20-30px high and just solid black, but prevents the tab "contents" from flowing to the bottom of the screen, which forces a scroll bar in some cases where it shouldn't be needed. It basically looks like the tab "content" area is shorter than expected based on the heigh configuration in yaml.

closed 2025-04-01 by BigBobbas