BigBobbas/ESP32-S3-Box3-Custom-ESPHome #134: Compilation on ESPhome 2025.2.2 fails

Tobster77, 2025-03-07:
Dear all,

I copied the most recent stable `s3b.yaml` and updated to ESPhome 2025.2.2. Unfortunately, the compilation fails.
Is there a certain upgrade procedure that I should know?

```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
Failed config

image: [source /config/esphome/esp32-s3box-3.yaml:1120]
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/error.png
    id: error_img
    resize: 320x240
    
    Image type RGB24 is removed; replace with 'type: RGB'.
    type: RGB24
    
    [use_transparency] is an invalid option for [image]. Did you mean [transparency]?
    use_transparency: True
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/idle.png
    id: idle_img
    resize: 320x240
```

Tobster77, 2025-03-07:
Thanks. So only the s3b.yaml and that's it? No other ressources?

BigBobbas, 2025-03-07:
glad that sorted it, I have added a more detailed explenation of the issues with 2025.2.x here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/135

cheers
Rob

rissato, 2025-03-19:
Just want to confirm the dev configuration works, but you have to try to install twice. The first time I also got an error.

closed 2025-04-01 by BigBobbas