BigBobbas/ESP32-S3-Box3-Custom-ESPHome #136: I think yesterday's HA update broke this integration

janaxhell, 2025-03-12:
This morning I saw a notification for a HA core update and applied it. Since then no ESP32-S3 box is responsive anymore and if I try to plug/unplug, or tap the test icon, nothing happens, it just stays on idling. Also no reaction from wakeword, I've set it to turn on screen in presence, but nothing happens, I have to manually touch the screen to turn it on.

janaxhell, 2025-03-13:
Any feedback? Is it only me?
I've had 4 of these boxes working for a few weeks, then suddenly they've become unresponsive right after latest HA update.

BigBobbas, 2025-03-14:
Updating your version of ESPHome would have no affect on the devices, unless you have compiled and installed a new firmware to the devices using the updated version of ESPHome. 
To try and troubleshoot, if you can connect via USB and then visit https://web.esphome.io Connect to the devices com port and then view logs. You should see any errors. If you could post any errors or if you are unsure , you can copy and paste the full log. The logs are live so will constanly poll data, so if you just paste the first couple of minutes that would be good.

janaxhell, 2025-03-14:
I haven't recompiled anything, here is the log of the first 2 minutes:

```
[14:51:38]I (492) esp_image: segment 1: paddr=0028c110 vaddr=3fc997[I][logger:171]: Log initialized
[14:51:38][C][safe_mode:079]: There have been 2 suspected unsuccessful boot attempts
[14:51:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:51:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[14:51:38][I][app:029]: Running through setup()...
[14:51:38][D][spi:035]: Setting up SPI bus...
[14:51:38][C][i2c.idf:017]: Setting up I2C bus...
[14:51:38][I][i2c.idf:252]: Performing I2C bus recovery[D][sensor:094]: 'up_sens': Sending state 17.72600 s with 0 decimals of accuracy
[14:51:38][D][api:103]: Accepted 192.168.1.193
[14:51:38][D][api.connection:1446]: Home Assistant 2025.3.2 (192.168.1.193): Connected successfully
[14:51:38][D][light:036]: 'LCD Backlight' Setting:
[14:51:38][D][light:051]:   Brightness: 70%
[14:51:38][D][light:085]:   Transition length: 0.1s
[14:51:38][D][script:077]: Script 'saver_enabled' restarting (mode: restart)
[14:51:38][D][time:051]: Synchronized time: 2025-03-14 14:51:37
[14:51:38][D][ring_buffer:034]: Created ring buffer with size 2048
[14:51:38][D][micro_wake_word:399]: Resetting buffers and probabilities
[14:51:38][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[14:51:39][D][micro_wake_word:107]: Starting Microphone
[14:51:39][D][esp_adf_pipeline.microphone:025]: start request while ine state 0
[14:51:39][D][esp_adf_pipeline:060]: Starting request, current state UNINITIALIZED
[14:51:39][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[14:51:39][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from UNINITIALIZED to INITIALIZING. (REQ: 0)
[14:51:39][D][i2s_external_adc:024]: ES7210 chip found.
[14:51:39][D][i2s_audio:067]: Install driver requested by Reader
[14:51:39][D][esp-idf:000]: I (21998) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4
[14:51:39]
[14:51:39][D][esp-idf:000]: I (21999) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4
[14:51:39]
[14:51:39][D][i2s_audio:073]: Installing driver : yes
[14:51:39][D][esp-idf:000]: I (22001) I2S: I2S0, MCLK output by GPIO2
[14:51:39]
[14:51:39][D][esp_adf_pipeline:497]: pipeline tag 0, i2s_in
[14:51:39][D][esp_adf_pipeline:497]: pipeline tag 1, resampler
[14:51:39][D][esp_adf_pipeline:497]: pipeline tag 2, pcm_reader
[14:51:39][D][esp-idf:000]: I (22005) AUDIO_PIPELINE: link el->rb, el:0x3d049264, tag:i2s_in, rb:0x3d0496f8
[14:51:39]
[14:51:39][D][esp-idf:000]: I (22006) AUDIO_PIPELINE: link el->rb, el:0x3d04946c, tag:resampler, rb:0x3d049f38
[14:51:39]
[14:51:39][D][esp_adf_pipeline:507]: Setting up event listener.
[14:51:39][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from INITIALIZING to CREATED. (REQ: 0)
[14:51:39][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from CREATED to PREPARING. (REQ: 0)
[14:51:39][D][adf_audio_element:108]: Preparing [i2s_in]...
[14:51:39][D][esp-idf:000]: I (22013) AUDIO_ELEMENT: [i2s_in-0x3d049264] Element task created
[14:51:39]
[14:51:39][D][adf_audio_element:108]: Preparing [resampler]...
[14:51:39][D][esp-idf:000]: I (22019) AUDIO_THREAD: The resampler task allocate stack on external memory
[14:51:39]
[14:51:39][D][esp-idf:000]: I (22020) AUDIO_ELEMENT: [resampler-0x3d04946c] Element task created
[14:51:39]
[14:51:39][D][adf_audio_element:108]: Preparing [pcm_reader]...
[14:51:39][D][esp-idf:000]: I (22024) AUDIO_ELEMENT: [pcm_reader-0x3d0495b0] Element task created
[14:51:39]
[14:51:39][D][esp_audio_processors:108]: Received request from: I2S_Reader
[14:51:39][D][esp_audio_processors:113]: New settings: SRC: rate: 16000, ch: 2 bits: 16, DST: rate: 16000, ch: 1, bits 16
[14:51:39][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 2 bits: 16, DST: rate: 16000, ch: 1, bits 16
[14:51:39][D][esp_adf_pipeline:342]: wait for preparation, done
[14:51:39][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[14:51:39][D][adf_audio_element:165]: Resuming [i2s_in]...
[14:51:39][D][adf_audio_element:172]: [i2s_in] Sending resume command.
[14:51:39][D][esp-idf:000][i2s_in]: I (22032) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1
[14:51:39]
[14:51:39][D][adf_audio_element:191]: [i2s_in] Checking State, got 78
[14:51:39][I][esp_adf_pipeline:132]: [ i2s_in ] status: 12
[14:51:39][D][esp-idf:000][resampler]: I (22035) RSP_FILTER: sample rate of source data : 16000, channel of source data : 2, sample rate of destination data : 16000, channel of destination data : 1
[14:51:39]
[14:51:40][I][esp_adf_pipeline:132]: [ resampler ] status: 12
[... 6 lines cut ...]
[14:52:07][D][light:047]:   State: OFF
[14:52:07][D][light:085]:   Transition length: 0.1s
[14:52:10][D][sensor:094]: 'battery_voltage': Sending state 0.18772 V with 1 decimals of accuracy
[14:52:10][D][sensor:094]: 'Battery level': Sending state 0.00000 % with 0 decimals of accuracy
[14:52:13][D][sensor:094]: 'WiFi db': Sending state -42.00000 dBm with 0 decimals of accuracy
[14:52:13][D][sensor:094]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:52:19][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[14:52:19][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:52:19][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[... cut ...]
[14:53:13][D][sensor:094]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:53:35][D][sensor:094]: 'up_sens': Sending state 137.73500 s with 0 decimals of accuracy
[14:53:40][D][sensor:094]: 'battery_voltage': Sending state 0.22593 V with 1 decimals of accuracy
[14:53:40][D][sensor:094]: 'Battery level': Sending state 0.00000 % with 0 decimals of accuracy
[14:53:43][D][sensor:094]: 'WiFi db': Sending state -42.00000 dBm with 0 decimals of accuracy
[14:53:43][D][sensor:094]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
```

janaxhell, 2025-03-14:
After a few minutes I see some errors:

```
[15:04:58][D][adf_audio_element:165]: Resuming [http]...
[15:04:58][D][adf_audio_element:172]: [http] Sending resume command.
[15:04:58][D][adf_audio_element:165]: Resuming [decoder]...
[15:04:58][D][adf_audio_element:172]: [decoder] Sending resume command.
[15:04:58][D][esp-idf:000][decoder]: I (821807) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1
[15:04:58]
[15:04:58][D][esp-idf:000][decoder]: I (821809) MP3_DECODER: MP3 opened
[15:04:58]
[15:04:58][I][esp_audio_sources:033][http]: Receive http event: 2
[15:04:58][I][esp_audio_sources:033][http]: Receive http event: 4
[15:04:58][D][adf_audio_element:191]: [http] Checking State, got 72
[15:04:58][D][adf_audio_element:191]: [decoder] Checking State, got 72
[15:04:58][D][esp-idf:000][http]: I (821839) HTTP_STREAM: total_bytes=0
[15:04:58]
[15:04:58][D][esp-idf:000][http]: E (821841) HTTP_STREAM: Invalid HTTP stream, status code = 404
[15:04:58]
[15:04:58][D][esp-idf:000][http]: E (821842) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,-1
[15:04:58]
[15:04:58][D][esp-idf:000][http]: W (821843) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7
[15:04:58]
[15:04:58][D][esp-idf:000][decoder]: W (821846) AUDIO_ELEMENT: IN-[decoder] AEL_IO_ABORT
[15:04:58]
[15:04:58][D][esp-idf:000][decoder]: E (821848) MP3_DECODER: failed to read audio data (line 117)
[15:04:58]
[15:04:58][D][esp-idf:000][decoder]: W (821849) AUDIO_ELEMENT: [decoder] AEL_IO_ABORT, -3
[15:04:58]
[15:04:58][D][esp-idf:000][decoder]: I (821850) MP3_DECODER: Closed
[15:04:58]
[15:04:58][E][esp_adf_pipeline:223]: HTTPStreamReader got in error state while PREPARING. Stopping pipeline!
[15:04:58][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from PREPARING to ABORTING. (REQ: 1)
[15:04:58][I][adf_media_player:192]: got new pipeline state: 10, while in MP state PLAYING
[15:04:58][I][adf_media_player:256]: current mp state: PLAYING
[15:04:58][I][adf_media_player:257]: anouncement: false
[15:04:58][I][adf_media_player:258]: play_intent: false
[15:04:58][I][adf_media_player:259]: current_uri_: yes
[15:04:58][E][esp_adf_pipeline:246]: Timeout while PREPARING. Stopping pipeline!
[15:04:58][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from ABORTING to ABORTING. (REQ: 1)
[15:04:58][I][adf_media_player:192]: got new pipeline state: 10, while in MP state PLAYING
[15:04:58][I][adf_media_player:256]: current mp state: PLAYING
[15:04:58][I][adf_media_player:257]: anouncement: false
[15:04:58][I][adf_media_player:258]: play_intent: false
[15:04:58][I][adf_media_player:259]: current_uri_: yes
[15:04:58][D][adf_audio_element:324]: [http] Checking State for stopping, got 7
[15:04:59][D][esp-idf:000]: E (821867) AUDIO_ELEMENT: [http] Element already stopped
[15:04:59]
[15:04:59][E][esp_adf_pipeline:223]: HTTPStreamReader got in error state while STOPPING. Stopping pipeline!
[15:04:59][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from ABORTING to DESTROYING. (REQ: 4)
[15:04:59][I][adf_media_player:192]: got new pipeline state: 11, while in MP state PLAYING
[15:04:59][I][adf_media_player:256]: current mp state: PLAYING
[15:04:59][I][adf_media_player:257]: anouncement: false
[15:04:59][I][adf_media_player:258]: play_intent: false
[15:04:59][I][adf_media_player:259]: current_uri_: yes
[15:04:59][E][esp_adf_pipeline:246]: Timeout while STOPPING. Stopping pipeline!
[15:04:59][D][esp_adf_pipeline:448]: [MediaPlayer] Pipeline changed from DESTROYING to DESTROYING. (REQ: 4)
[15:04:59][I][adf_media_player:192]: got new pipeline state: 11, while in MP state PLAYING
[15:04:59][I][adf_media_player:256]: current mp state: PLAYING
[15:04:59][I][adf_media_player:257]: anouncement: false
[15:04:59][I][adf_media_player:258]: play_intent: false
[15:04:59][I][adf_media_player:259]: current_uri_: yes
[15:04:59][D][esp_adf_pipeline:518]: Called deinit_all
[15:04:59][D][esp-idf:000]: I (821884) AUDIO_PIPELINE: audio_pipeline_unlinked
[15:04:59]
[15:04:59][D][esp-idf:000]: W (821884) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
[15:04:59]
[15:04:59][D][esp-idf:000]: W (821885) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE
[15:04:59]
[15:04:59][D][esp-idf:000]: W (821886) AUDIO_ELEMENT: [resampler] Element has not create when AUDIO_ELEMENT_TERMINATE
[... 7 lines cut ...]
[15:04:59][I][adf_media_player:257]: anouncement: false
[15:04:59][I][adf_media_player:258]: play_intent: false
[15:04:59][I][adf_media_player:259]: current_uri_: yes
[15:04:59][D][media_player:065]: 's3-box_media_player' - Setting
[15:04:59][D][media_player:069]:   Command: STOP
[15:04:59][D][esp_adf_pipeline:070]: Called 'stop' while in UNINITIALIZED state.
[15:04:59][D][voice_assistant:512]: State changed from IDLE to START_MICROPHONE
[15:04:59][D][voice_assistant:518]: Desired state set to START_PIPELINE
[15:04:59][D][voice_assistant:222]: Starting Microphone
[15:04:59][D][ring_buffer:034]: Created ring buffer with size 16384
[15:04:59][D][esp_adf_pipeline.microphone:025]: start request while ine state 0
[15:04:59][D][esp_adf_pipeline:060]: Starting request, current state STOPPED
[15:04:59][D][voice_assistant:512]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[15:04:59][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from STOPPED to PREPARING. (REQ: 0)
[15:04:59][D][adf_audio_element:108]: Preparing [i2s_in]...
[15:04:59][D][adf_audio_element:108]: Preparing [resampler]...
[... cut ...]
[15:04:59][D][esp_audio_processors:124]: Current settings: SRC: rate: 16000, ch: 2 bits: 16, DST: rate: 16000, ch: 1, bits 16
[15:04:59][D][esp_adf_pipeline:342]: wait for preparation, done
[15:04:59][D][esp_adf_pipeline:448]: [ADFMicrophone] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[15:04:59][D][adf_audio_element:165]: Resuming [i2s_in]...
[15:04:59][D][adf_audio_element:172]: [i2s_in] Sending resume command.
[15:04:59][D][esp-idf:000][i2s_in]: I (822136) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1
[15:04:59]
```

janaxhell, 2025-03-19:
Hello?

janaxhell, 2025-03-19:
I'm using this this totally as-is according to your tutorial, I haven't changed a single thing, just installed the firmware. I've hooked Voice assistant to OpenAI api Italian and as TTS I've set Google Translate en com / Italian. No TTS set.

BigBobbas, 2025-03-19:
in voice settings on the device do you have this set to 'On Device` or `Home Assistant` 

also if you manually type a command in HA from the chat icon

[image]

does HA respond correctly.?

janaxhell, 2025-03-19:
ESP32-S3-Box is set to On Device and HA chat works fine:

[image]

BigBobbas, 2025-03-19:
could you watch the device logs and say the wake word and see if anything changes in the logs.  It is as if the device is on mute. Also make sure that the orange light on top of the box is not on and that the mic icon at the top of the screen is not showing as muted.

janaxhell, 2025-03-19:
Light is green, mic is ON and as I said in first post, test is unresponsive, if I tap the icon nothing changes. Only other icons do work, test doesn't. Wakeword is completely ignored and screen doesn't turn on on presence (both thing did work before). This is true for all my 4 boxes, they all stopped working at the same time. 

[image]

BigBobbas, 2025-03-19:
It doesn't really make any sense that all 4 boxes would have the same issue. This points to it being an issue on the HA side of things. and not with the device.

Can you look in HA >> settings >> system>>logs 
and look for any logs that may be related.

Unfortunatley it is like looking for a needle in a haystack.  There are many different things that could be causing it.

janaxhell, 2025-03-19:
In fact it happened after ESPHome update. What should I look for in the logs? voice and esp show nothing.

janaxhell, 2025-03-19:
I don't see anything like that, plus the logs are moving and partial, so how could I find anything?
HA is current, I update everything as I see notifications.

```
Core 2025.3.3
Supervisor 2025.03.3
Operating System 15.0
Frontend 20250306.0
```

BigBobbas, 2025-03-20:
are you looking at the logs in the right place?  They shouldn't be moving too fast, you can click on an entry for more information.

Can you also paste an image of the ESPHome integration page for one of the devices. from HA>>settings>>devices and services>>ESPHome 

The details I need to see are from the centre column like the screen recording below.

https://github.com/user-attachments/assets/f09d5ec0-3f77-48a9-8285-9f4c8f1bd7bd

Also can you paste a screen shot of your assist settings as per the screen shot below, from settings>>voice assistants

https://github.com/user-attachments/assets/988b996d-f251-4e5a-b1d9-f968c1f36a14

BigBobbas, 2025-03-20:
If you do reflash 1 of the devices can you use the config from here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/dev/s3b.yaml

BigBobbas, 2025-03-20:
Thanks for the screenshots , this helps greatly.

2 things I notice.

you have nothing set to process voice commands , therefore nothing will happen. You have to have an STT service selected for voice to function.

[image]

also you have mic muted.
therefore no audio will be passed to HA to process

[image]

janaxhell, 2025-03-20:
Solved, it was exactly that, but I have no idea why those settings were like that, I never opened that tab before you told me to do so today. I have now unmuted mic and set STT to Whisper (the only one I have) and everything is back to working. Thank you very much for sorting this out!

BigBobbas, 2025-03-20:
great... glad we sorted it in the end. Maybe an HA update had something to do with it. There has been quite a few changes in recent updates with regards to voice, so not entirely sure, but at least you are now back up and running 👍

janaxhell, 2025-03-20:
I have tried a few blueprints found on reddit, maybe one of those messed with some settings without me noticing. It's the only thing I can think of besides updates.

BigBobbas, 2025-04-01:
> i was waiting for the resolution to the above errors: src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp: In member function 'virtual void esphome::esp_adf::ADFMediaPlayer::control(const esphome::media_player::MediaPlayerCall&)': src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_ENQUEUE' not handled in switch [-Werror=switch] switch (call.get_command().value()) { ^ src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_ONE' not handled in switch [-Werror=switch] src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_OFF' not handled in switch [-Werror=switch] src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:99:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_CLEAR_PLAYLIST' not handled in switch [-Werror=switch] cc1plus: some warnings being treated as errors *** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/media_player/adf_media_player.o] Error 1

PLease see the announcment here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/135

closed 2025-04-01 by BigBobbas