BigBobbas/ESP32-S3-Box3-Custom-ESPHome #137: Touchscreen needs a reboot to work after uploading code / powercylce device

Tobi9111, 2025-03-16:
Hello,
First of all, @BigBobbas : really great work and thank you for providing the code!

I've already made a lot of adjustments and overall everything is working as it should. However, I still need to fix one thing.

After uploading a code or powercycle the device and start it again, the touchscreen doesn't work. It only works after another reboot.

The log shows the following:

[09:18:57][C][gt911.touchscreen:127]: GT911 Touchscreen:
[09:18:57][C][gt911.touchscreen:128]:   Address: 0x5D
[09:18:57][C][gt911.touchscreen:129]:   Interrupt Pin: GPIO3
[09:18:57][E][component:082]:   Component touchscreen is marked FAILED

The full log:

[logs_esp32-s3box-3_run.txt](https://github.com/user-attachments/files/19270074/logs_esp32-s3box-3_run.txt)

I could imagine that this error is triggered because the touchscreen uses a strapping pin.  

Best regards 

Tobi

BigBobbas, 2025-03-16:
Hi Tobi,
this has been an ongoing issue that has existed since the device was launched. It has been looked into many times and as yet no solution has been found, so it's not an issue with your particular device and config. 
Some people don't have the problem and yet others do, although using the exact same hardware and config, and for some time using an external audio component this appeared to fix the issue for some people, however that component is no longer supported and I have had to change in order to maintain compatability with the latest versions of ESPHome.
With the current firmware in Dev branch , I have added a message to the firmware upload screen to advise rebooting after flashing is complete to enable the touch screen.
I realise that this isn't ideal, however it does only take a few seconds to reboot.
As soon as there is a fix available I will be sure to make the required changes. 

Sorry this doesn't provide a solution but hopefully confirms that this is not something specific to your own setup.

Many thanks for your support 

Cheers 
Rob 👍

Tobi9111, 2025-03-16:
> Hi Tobi, this has been an ongoing issue that has existed since the device was launched. It has been looked into many times and as yet no solution has been found, so it's not an issue with your [...]

Hi Rob,

thank you very much for your fast answer. It is not that big thing and perhephs it can be fixed in the future. 

Best regards
Tobi

youkorr, 2025-03-16:
It should work from the first flash but it's stuck for the moment we haven't found a solution you just have to restart the box for the touchscreen to work, but it depends on each person who doesn't have this problem

Tobi9111, 2025-03-22:
Since ESPHome Update 2025.3 the issue is fixed. After uploading firmware the touchscreen works instantly.

Best regards
Tobi

closed 2025-04-01 by BigBobbas