BigBobbas/ESP32-S3-Box3-Custom-ESPHome #143: esphome  2025.3.2   adf_media_player.cpp. issue

carlinkkit, 2025-03-27:
since HA upgraded to esphome  2025.3.2 , I'm getting the following error when try to compile. anyone experiencing the same or found a solution ?

Compiling .pioenvs/s3box/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.o
Compiling .pioenvs/s3box/src/esphome/components/gt911/binary_sensor/gt911_button.o
Compiling .pioenvs/s3box/src/esphome/components/gt911/touchscreen/gt911_touchscreen.o
Compiling .pioenvs/s3box/src/esphome/components/homeassistant/text_sensor/homeassistant_text_sensor.o
Compiling .pioenvs/s3box/src/esphome/components/homeassistant/time/homeassistant_time.o
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp: In member function 'virtual void esphome::esp_adf::ADFMediaPlayer::control(const esphome::media_player::MediaPlayerCall&)':
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:100:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_ENQUEUE' not handled in switch [-Werror=switch]
     switch (call.get_command().value()) {
            ^
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:100:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_ONE' not handled in switch [-Werror=switch]
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:100:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_REPEAT_OFF' not handled in switch [-Werror=switch]
src/esphome/components/adf_pipeline/media_player/adf_media_player.cpp:100:12: error: enumeration value 'MEDIA_PLAYER_COMMAND_CLEAR_PLAYLIST' not handled in switch [-Werror=switch]
Compiling .pioenvs/s3box/src/esphome/components/i2c/i2c.o
Compiling .pioenvs/s3box/src/esphome/components/i2c/i2c_bus_arduino.o
Compiling .pioenvs/s3box/src/esphome/components/i2c/i2c_bus_esp_idf.o
Compiling .pioenvs/s3box/src/esphome/components/i2s_audio/adf_pipeline/adf_i2s_in.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/s3box/src/esphome/components/adf_pipeline/media_player/adf_media_player.o] Error 1
========================== [FAILED] Took 4.22 seconds ==========================

BigBobbas, 2025-03-27:
Hi please see https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/135 

if you need any help please let us know 👍 

cheers
Rob

carlinkkit, 2025-03-27:
@BigBobbas   thanks for the quick response. actually I was able to tinker with the template little bit and get it to work.
Not a perfect solution but at least it works on the current esphome release. (until a better solution is available)

1- copied the adf_pipline repo locally to my HA host.

2- update the adf_media_player.cpp abd added the missing cases to the switch statement.
`      case media_player::MEDIA_PLAYER_COMMAND_ENQUEUE:
      case media_player::MEDIA_PLAYER_COMMAND_REPEAT_ONE:
      case media_player::MEDIA_PLAYER_COMMAND_REPEAT_OFF:
      case media_player::MEDIA_PLAYER_COMMAND_CLEAR_PLAYLIST:
        // Do nothing
        break;
`

3- save and compile

**snippet of the updated yaml template:**
`external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components: [ i2s_audio ]
    refresh: 0s

  - source:
      type: local
      path: "my_components"
    components: [ adf_pipeline ]`

**snippet of the updated  adf_media_player.cpp:**

`if (call.get_command().has_value()) {
    switch (call.get_command().value()) {
      case media_player::MEDIA_PLAYER_COMMAND_PLAY:
        if (pipeline.getState() == PipelineState::STOPPED || pipeline.getState() == PipelineState::UNINITIALIZED) {
          pipeline.start();
        } else if (pipeline.getState() == PipelineState::PAUSED) {
          pipeline.restart();
        }
        else if (state == media_player::MEDIA_PLAYER_STATE_PLAYING || state == media_player::MEDIA_PLAYER_STATE_PAUSED){
          this->play_intent_ = true;
          pipeline.stop();
        }
        break;
      case media_player::MEDIA_PLAYER_COMMAND_PAUSE:
        if (pipeline.getState() == PipelineState::RUNNING) {
          pipeline.pause();
        }
        break;
      case media_player::MEDIA_PLAYER_COMMAND_STOP:
        this->current_track_.reset();
        this->current_uri_.reset();
        pipeline.stop();
        this->http_and_decoder_.set_fixed_settings(false);
        break;
      case media_player::MEDIA_PLAYER_COMMAND_MUTE:
        this->mute_();
        break;
      case media_player::MEDIA_PLAYER_COMMAND_UNMUTE:
        this->unmute_();
        break;
      case media_player::MEDIA_PLAYER_COMMAND_TOGGLE:
        if (pipeline.getState() == PipelineState::STOPPED || pipeline.getState() == PipelineState::PAUSED) {
          state = media_player::MEDIA_PLAYER_STATE_PLAYING;
        } else {
          state = media_player::MEDIA_PLAYER_STATE_PAUSED;
        }
        break;
      case media_player::MEDIA_PLAYER_COMMAND_VOLUME_UP: {
        float new_volume = this->volume + 0.1f;
        if (new_volume > 1.0f)
          new_volume = 1.0f;
        set_volume_(new_volume);
        break;
      }
      case media_player::MEDIA_PLAYER_COMMAND_VOLUME_DOWN: {
        float new_volume = this->volume - 0.1f;
        if (new_volume < 0.0f)
          new_volume = 0.0f;
        set_volume_(new_volume);
        break;
      }
      case media_player::MEDIA_PLAYER_COMMAND_ENQUEUE:
      case media_player::MEDIA_PLAYER_COMMAND_REPEAT_ONE:
      case media_player::MEDIA_PLAYER_COMMAND_REPEAT_OFF:
      case media_player::MEDIA_PLAYER_COMMAND_CLEAR_PLAYLIST:
        // Do nothing
        break;
    }
}`

BigBobbas, 2025-03-27:
nice job and thank you for sharing.
The reason we have moved away from using gnumpi's ADF pipeline is that the project has not received any support or updates to facilitate changes made in ESPHome from the author.  ESPHome has now removed all ADF dependencays from the project which is why we have changed to a native ESPHome solution.
On the whole the version currently in dev branch now appers to be working well, albeit without duplex support (which i'm assured is coming). However we have worked around this for the time being.

carlinkkit, 2025-03-27:
great ! looking forward for it

closed 2025-04-01 by BigBobbas