BigBobbas/ESP32-S3-Box3-Custom-ESPHome #144: Can't upgrade firmware error

wappinghigh, 2025-03-29:
failed config

image: [source /config/esphome/bigbobbass3box.yaml:515]
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/error.png
    id: error_img
    resize: 320x240
    
    Image type RGB24 is removed; replace with 'type: RGB'.
    type: RGB24
    
    [use_transparency] is an invalid option for [image]. Did you mean [transparency]?
    use_transparency: True
  - file: |-
      https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/images/idle.png
    id: idle_img
    resize: 320x240

Any idea what I am doing wrong?

thanks!

BigBobbas, 2025-03-29:
https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/discussions/135

wappinghigh, 2025-03-29:
Ah ok. Cheers for that. Didn't know. Will wait for the stable version.

closed 2025-04-01 by BigBobbas