BigBobbas/ESP32-S3-Box3-Custom-ESPHome #145: Issue playback on alexa echo

mininja2131, 2025-03-30:
I was attempting to use this firmware and seen that it allows playback of responses to alexa echo. I attempted to send the response to the echo and the echo give me a error tts saying direct playback is not supported blah blah. But on the old firmware I had from the guides on homeassistant worked with said playback. I should also put in that while doing this change I did finally update everything (normally do not update for LONG periods of time on HA). Is this a limitation of this firmware or something to do with the firmware? I can tell it it attempting to send something to the echo due to the echo responds with that error. Something is not allowing the "turned on the light" response to happen just some direct playback not supported error. Just would like to use my echo(s) to respond for my s3boxs....I had changed the opening part of the code to 

```
## You can change the friendly name if you wish on line 7 below. The friendly name is how the device will show in HA 
## However the <name:> is how HA connects to the device eg. esp32-s3-box-3.local is the mDNS hostname for the device.
## Once added to HA changing the device name can lead to connection issues, between HA and the device.

substitutions:
  name: tj_desk
  friendly_name: ESP32-S3-Box-TJ
  
  external_media_player: livingroom ##change this to your external media player enity_id: do not include media_player.
  home_assistant_host: http://homeassistant.local:8123 ##change to the full url or IP of your HA server including port
  tts_voice_speed: "24000" #options #nabu casa "24000" #piper "16000" elevenlabs "44100" must include quotes
```

BigBobbas, 2025-04-01:
sorry for the delay in replying.
Are you able to send TTS to the echo directly from HA using developer tools? I apologise I do not have any amazon devices so cannot test.
Maybe someone on our discord can help, you could also ask there https://discord.gg/U3SRYCG6Kp

mininja2131, 2025-04-01:
yes I can. Already attempted that just to cover my behind lol. And I get it. Got them before home assistant and now dont want to have 20 voice devices. So wanted to reuse the echos. I was able to do it with some extra configs on the original firmware so I know it can be done. I will join the discord to see if anyone else has info. And if I get it I will post it here for anyone else using google fu lol.

geddyleefan, 2025-04-13:
I'm trying to do the same thing, but I don't even get an error, just nothing happens. Hoping you get it to work :)

mininja2131, 2025-04-14:
Not going to lie i have about given up. I can not figure it out and others have not seemed to try.