BigBobbas/ESP32-S3-Box3-Custom-ESPHome #151: Voice Assistant doesn't work in stable version

Mis1ify, 2025-04-16:
Hi. I installed the newest version from the main branch. When I say "Okay Nabu" nothing happens. Default official hass config works, but I really like the screensaver in your config. I had previously older version of your config and it was working good - screensaver worked and the voice assistant was working fine, however the TTS often just didn't say anything. That's why I wanted to update, I thought maybe this was fixed. 

The following is a clean install, after I removed the integration from Home Assistant, installed default config by the hass site (so it erased everything) and then put your newest config into the device using ESPhome integration.

Here are the logs:
```
[23:11:31][D][api.connection:1801]: Home Assistant 2025.3.4 (192.168.1.100): Connected successfully
[23:11:31][D][light:036]: 'LCD Backlight' Setting:
[23:11:31][D][light:051]:   Brightness: 100%
[23:11:31][D][light:085]:   Transition length: 0.1s
[23:11:31][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[23:11:31][D][voice_assistant:512]: Desired state set to IDLE
[23:11:31][D][time:050]: Synchronized time: 2025-04-16 23:11:31
[23:11:31][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[23:11:31][D][voice_assistant:505]: State changed from STOPPING_MICROPHONE to IDLE
[23:11:32][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[23:11:32][D][voice_assistant:512]: Desired state set to START_PIPELINE
[23:11:32][D][voice_assistant:223]: Starting Microphone
[23:11:32][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[23:11:32][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[23:11:32][D][voice_assistant:277]: Requesting start...
[23:11:32][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[23:11:32][D][voice_assistant:527]: Client started, streaming microphone
[23:11:32][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[23:11:32][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
```

Version of my ESPHome Device Builder is 2025.3.3, I'm wondering if that is the problem, but I tried reverting to old one I had (2025.2.1) and it still wasn't working.
My HASS is 2025.3.4 and OS is 15.0.

BigBobbas, 2025-04-17:
Hi there ,
on the device screen if you go to the settings cogs and then 'voice'  do you have wakeword: set to 'on device' ?  also, the 'microphone' icon on the top row should be green and not red.

Thanks
Rob

Mis1ify, 2025-04-17:
Thank you for responding! That was it, I had wakeword set to home assistant on the device itself. ESPHome integration showed that it is set to "On Device" so it just didn't sync with device.

Thanks again!

closed 2025-04-17 by Mis1ify