BigBobbas/ESP32-S3-Box3-Custom-ESPHome #152: Voice Assistant starts recording voice too late

Mis1ify, 2025-04-18:
Hi! In new stable version of the config I experience a problem with voice assistant. Previously, after I said 'Okay Nabu' and heard the awake sound, I said the command next, obviously. But now, I need to wait 1-2 sec after I hear the awake sound, because the voice assistant just starts recording my sound too lately. It's the delay that is very noticeable and wasn't present in the previous stable version.

I added "debug_recording_dir: /share/assist_pipeline" to my home assistant config to be able to hear what my voice assistant recorded. And yes, I hear only half of what I said, so for example if I say "Okay Nabu, turn on the light", it records only "the light", which I can confirm by the recordings.

I used the add-on to flash the configuration, both 2025.3.3 and 2025.4.0 versions of the ESPHome add-on.

Here are the logs while I'm talking to voice assistant:
```
[21:47:02][D][micro_wake_word:194]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[21:47:11][D][light:036]: 'LCD Backlight' Setting:
[21:47:11][D][light:051]:   Brightness: 50%
[21:47:11][D][light:085]:   Transition length: 0.1s
[21:47:22][D][sensor:093]: 'up_sens': Sending state 93.66700 s with 0 decimals of accuracy
[21:47:28][D][sensor:093]: 'WiFi db': Sending state -44.00000 dBm with 0 decimals of accuracy
[21:47:28][D][sensor:093]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[21:47:31][D][time:046]: Synchronized time: 2025-04-18 21:47:31
[21:47:31][D][time:046]: Synchronized time: 2025-04-18 21:47:31
[21:47:51][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:47:51][D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[21:47:52][D][sensor:093]: 'up_sens': Sending state 123.66700 s with 0 decimals of accuracy
[21:47:56][D][micro_wake_word:160]: The 'Okay Nabu' model sliding average probability is 0.975 and most recent probability is 1.000
[21:47:56][D][micro_wake_word:123]: Wake Word 'Okay Nabu' Detected
[21:47:56][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[21:47:56][D][micro_wake_word:129]: Stopping Microphone
[21:47:56][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[21:47:56][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[21:47:56][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[21:47:56][D][speaker_media_player:426]: State changed to ANNOUNCING
[21:47:56][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[21:47:56][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 16000 Hz sample rate, and 16 bits per sample
[21:47:56][D][ring_buffer:034][sample]: Created ring buffer with size 3200
[21:47:56][D][ring_buffer:034]: Created ring buffer with size 9600
[21:47:56][D][resampler_speaker:059]: Starting resampler task
[21:47:56][D][resampler_speaker:080]: Started resampler task
[21:47:56][D][speaker_mixer:310]: Starting speaker mixer
[21:47:56][D][speaker_mixer:318]: Started speaker mixer
[21:47:56][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[21:47:56][D][i2s_audio.speaker:111]: Starting Speaker
[21:47:56][D][i2s_audio.speaker:116]: Started Speaker
[21:47:57][D][speaker_media_player:426]: State changed to IDLE
[21:47:57][D][resampler_speaker:085]: Stopping resampler task
[21:47:57][D][resampler_speaker:090]: Stopped resampler task
[21:47:57][D][speaker_mixer:323]: Stopping speaker mixer
[21:47:57][D][voice_assistant:505]: State changed from IDLE to START_MICROPHONE
[21:47:57][D][voice_assistant:512]: Desired state set to START_PIPELINE
[21:47:57][D][voice_assistant:223]: Starting Microphone
[21:47:57][D][voice_assistant:505]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[21:47:57][D][i2s_audio.speaker:123]: Stopping Speaker
[21:47:57][D][i2s_audio.speaker:129]: Stopped Speaker
[21:47:57][D][voice_assistant:505]: State changed from STARTING_MICROPHONE to START_PIPELINE
[21:47:57][D][voice_assistant:277]: Requesting start...
[21:47:57][D][voice_assistant:505]: State changed from START_PIPELINE to STARTING_PIPELINE
[21:47:57][D][voice_assistant:527]: Client started, streaming microphone
[21:47:57][D][voice_assistant:505]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[21:47:57][D][voice_assistant:512]: Desired state set to STREAMING_MICROPHONE
[21:47:57][D][voice_assistant:642]: Event Type: 1
[21:47:57][D][voice_assistant:645]: Assist Pipeline running
[21:47:57][D][voice_assistant:642]: Event Type: 3
[21:47:57][D][voice_assistant:656]: STT started
[21:47:57][D][light:036]: 'LCD Backlight' Setting:
[21:47:57][D][light:051]:   Brightness: 50%
[21:47:57][D][light:085]:   Transition length: 0.1s
[21:47:58][D][sensor:093]: 'WiFi db': Sending state -44.00000 dBm with 0 decimals of accuracy
[21:47:58][D][sensor:093]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
[21:47:59][D][voice_assistant:642]: Event Type: 11
[21:47:59][D][voice_assistant:805]: Starting STT by VAD
[21:48:01][D][voice_assistant:642]: Event Type: 12
[21:48:01][D][voice_assistant:809]: STT by VAD end
[21:48:01][D][voice_assistant:505]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[21:48:01][D][voice_assistant:512]: Desired state set to AWAITING_RESPONSE
[21:48:01][D][voice_assistant:505]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[21:48:01][D][voice_assistant:505]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[21:48:06][D][voice_assistant:642]: Event Type: 4
[21:48:06][D][voice_assistant:670]: Speech recognised as: " Dzięki za uwagę."
[21:48:06][D][voice_assistant:642]: Event Type: 5
[21:48:06][D][voice_assistant:675]: Intent started
[21:48:06][D][voice_assistant:642]: Event Type: 6
[21:48:06][D][voice_assistant:642]: Event Type: 7
[21:48:06][D][voice_assistant:698]: Response: "Zawsze do usług."
[21:48:06][D][voice_assistant:642]: Event Type: 8
[21:48:06][D][voice_assistant:720]: Response URL: "http://192.168.1.100:8123/api/tts_proxy/Ax2Kxo5EFS5OHGn628TPKQ.flac"
[21:48:06][D][voice_assistant:505]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[21:48:06][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[21:48:06][D][media_player:074]: 'Media Player' - Setting
[21:48:06][D][media_player:081]:   Media URL: http://192.168.1.100:8123/api/tts_proxy/Ax2Kxo5EFS5OHGn628TPKQ.flac
[21:48:06][D][media_player:087]:  Announcement: yes
[21:48:07][D][speaker_media_player:426]: State changed to ANNOUNCING
[21:48:07][D][voice_assistant:642]: Event Type: 2
[21:48:07][D][voice_assistant:736]: Assist Pipeline ended
[21:48:07][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[21:48:07][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[21:48:07][D][ring_buffer:034]: Created ring buffer with size 9600
[21:48:07][D][speaker_mixer:310]: Starting speaker mixer
[21:48:07][D][speaker_mixer:318]: Started speaker mixer
[21:48:07][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[21:48:07][D][i2s_audio.speaker:116]: Started Speaker
[21:48:07][D][light:036]: 'LCD Backlight' Setting:
[21:48:07][D][light:051]:   Brightness: 50%
[21:48:07][D][light:085]:   Transition length: 0.1s
[21:48:08][D][speaker_media_player:426]: State changed to IDLE
[21:48:08][D][voice_assistant:505]: State changed from STREAMING_RESPONSE to IDLE
[21:48:08][D][voice_assistant:512]: Desired state set to IDLE
[21:48:08][D][micro_wake_word:399]: Resetting buffers and probabilities
[21:48:08][D][micro_wake_word:194]: State changed from IDLE to START_MICROPHONE
[21:48:08][D][number:054]: 'Scr/save delay' - Setting number value
[21:48:08][D][number:113]:   New number value: 10.000000
[21:48:08][D][number:012]: 'Scr/save delay': Sending state 10.000000
[21:48:08][D][light:036]: 'LCD Backlight' Setting:
[21:48:08][D][light:051]:   Brightness: 50%
[21:48:08][D][light:085]:   Transition length: 0.1s
[21:48:08][D][micro_wake_word:107]: Starting Microphone
[21:48:08][D][micro_wake_word:194]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[21:48:08][D][speaker_mixer:323]: Stopping speaker mixer
[21:48:09][D][i2s_audio.speaker:123]: Stopping Speaker
[21:48:09][D][i2s_audio.speaker:129]: Stopped Speaker
[21:48:09][D][micro_wake_word:194]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[21:48:18][D][light:036]: 'LCD Backlight' Setting:
[21:48:18][D][light:051]:   Brightness: 50%
[21:48:18][D][light:085]:   Transition length: 0.1s
[21:48:22][D][sensor:093]: 'up_sens': Sending state 153.67500 s with 0 decimals of accuracy
[21:48:28][D][sensor:093]: 'WiFi db': Sending state -43.00000 dBm with 0 decimals of accuracy
[21:48:28][D][sensor:093]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
```

BigBobbas, 2025-04-19:
Hi there 
thanks for the feedback. I will have a look into this. Which service are you using for your STT engine?

Thanks
Rob

BigBobbas, 2025-04-19:
I'm looking at the config s3b.yaml for reference as I'm not sure if you are using with or without the sensor dock.

Ive highlighted lines 1044 and 1060 in the screenshot below. Both of these contain a delay. You could try reducing these maybe as low as 25ms to see if that helps.  It would be a little trial and error to see how low you can set them and it still works for you.

Because I have written the firmware in a way that gives options to varying user requirements, there will be parts of the configuration which aren't applicable in your use case. For example the option to use on device or HA processed wake words, also the option to use an external media player to output tts rather than the internal speaker.  These 'options' all add to the compute time and we can remove these if neccessary to see if these also help with performance. 

Start by changing the delays and we can look at making a bespoke config to suit you if needed. 

good luck !

Mis1ify, 2025-04-19:
Thanks for responding! I changed the delays, however it didn't helped. I'm using official faster-whisper add-on for the STT and the s3b_no_sensors config.
I always used this config because it has a screensaver with built-in digital clock. The main use case for me is this clock and voice assistant with internal speaker.

Mis1ify, 2025-05-05:
I did my own config from the scratch using official config, and added your screensaver to the config, i think it's great. I also added outdoor temperature on the screensaver.

[esp config.txt](https://github.com/user-attachments/files/20048104/esp.config.txt)