BigBobbas/ESP32-S3-Box3-Custom-ESPHome #153: TTS audio playback failure: no sound and cannot be stopped.

zminde, 2025-04-20:
This is a fantastic fork, far more exciting than the original repository. I can use almost all the features: voice wake-up works normally, media file playback is fine, and screen sleep functions properly. However, when I try to play TTS audio through the media player, the device enters a frozen state, Media player status: "playing" and cannot be changed, and the background log file endlessly loops with the same error message. I’ve reviewed nearly all the issues submitted by others in this fork and couldn’t find a case similar to mine. When using the firmware from the original main repository, the device does not exhibit any abnormalities. Playing TTS audio messages generated by the voice agent is critical for my use case. However, I cannot give up this incredibly impressive customized firmware because of this issue. Below, I’ve attached a portion of the log file for reference. Additionally, I submitted the error code to ChatGPT for analysis, and it seems to point to potential causes like GPIO conflicts or other hardware issues. Since I lack expertise in these areas, I’m unable to troubleshoot and resolve the issue on my own.
（Using a sampling rate of 2400 or 4800 did not help with the issue.）

system info:
homeassistant os  by esxi
Core 2025.4.3 
Supervisor 2025.04.0 
Operating System 15.2 
Frontend 20250411.0

log：
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Detected timezone 'Asia/Shanghai'
INFO Starting log output from 192.168.0.111 using esphome API
INFO Successfully connected to esp32-s3box-3 @ 192.168.0.111 in 0.075s
INFO Successful handshake with esp32-s3box-3 @ 192.168.0.111 in 0.091s
[22:28:14][I][app:100]: ESPHome version 2025.4.0 compiled on Apr 20 2025, 21:51:45
[22:28:14][I][app:102]: Project BigBobbas.s3box version 2024.09.02
[22:28:14][C][wifi:600]: WiFi:
[22:28:14][C][wifi:428]:   Local MAC: 30:30:F9:5A:C0:9C
[22:28:14][C][wifi:433]:   SSID: 'mi.mao'[redacted]
[22:28:14][C][wifi:436]:   IP Address: 192.168.0.111
[22:28:14][C][wifi:439]:   BSSID: F0:B4:29:1A:D9:CB[redacted]
[22:28:14][C][wifi:441]:   Hostname: 'esp32-s3box-3'
[22:28:15][C][wifi:443]:   Signal strength: -51 dB ▂▄▆█
[22:28:15][C][wifi:447]:   Channel: 7
[22:28:15][C][wifi:448]:   Subnet: 255.255.255.0
[22:28:15][C][wifi:449]:   Gateway: 192.168.0.1
[22:28:15][C][wifi:450]:   DNS1: 192.168.0.1
[22:28:15][C][wifi:451]:   DNS2: 0.0.0.0
[22:28:15][C][logger:177]: Logger:
[22:28:15][C][logger:178]:   Max Level: DEBUG
[22:28:15][C][logger:179]:   Initial Level: DEBUG
[22:28:15][C][logger:181]:   Log Baud Rate: 115200
[22:28:15][C][logger:182]:   Hardware UART: USB_SERIAL_JTAG
[22:28:15][C][logger:186]:   Level for 'component': ERROR
[22:28:15][C][logger:186]:   Level for 'text_sensor': WARN
[22:28:15][C][spi:068]: SPI bus:
[22:28:15][C][spi:069]:   CLK Pin: GPIO7
[22:28:15][C][spi:070]:   SDI Pin: 
[22:28:15][C][spi:071]:   SDO Pin: GPIO6
[22:28:15][C][spi:076]:   Using HW SPI: SPI2_HOST
[22:28:15][C][i2c.idf:083]: I2C Bus:
[22:28:15][C][i2c.idf:084]:   SDA Pin: GPIO8
[22:28:15][C][i2c.idf:085]:   SCL Pin: GPIO18
[22:28:15][C][i2c.idf:086]:   Frequency: 100000 Hz
[22:28:15][C][i2c.idf:092]:   Recovery: bus successfully recovered
[22:28:15][I][i2c.idf:102]: Results from i2c bus scan:
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x14
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x18
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x40
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x68
[22:28:15][C][i2c.idf:083]: I2C Bus:
[22:28:15][C][i2c.idf:084]:   SDA Pin: GPIO41
[22:28:15][C][i2c.idf:085]:   SCL Pin: GPIO40
[22:28:15][C][i2c.idf:086]:   Frequency: 50000 Hz
[22:28:15][C][i2c.idf:092]:   Recovery: bus successfully recovered
[22:28:15][I][i2c.idf:102]: Results from i2c bus scan:
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x28
[22:28:15][I][i2c.idf:108]: Found i2c device at address 0x38
[22:28:15][C][template.select:065]: Template Select 'Wake word engine location'
[22:28:15][C][template.select:066]:   Update Interval: 60.0s
[22:28:15][C][template.select:069]:   Optimistic: YES
[22:28:15][C][template.select:070]:   Initial Option: In Home Assistant
[22:28:15][C][template.select:071]:   Restore Value: YES
[22:28:15][C][template.select:065]: Template Select 'Time Format'
[22:28:15][C][template.select:066]:   Update Interval: 60.0s
[22:28:15][C][template.select:069]:   Optimistic: YES
[22:28:15][C][template.select:070]:   Initial Option: 24 Hr
[... 419 lines cut ...]
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:111]: Starting Speaker
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:111]: Starting Speaker
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:111]: Starting Speaker
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:111]: Starting Speaker
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE
[22:31:21][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[22:31:21][D][i2s_audio.speaker:111]: Starting Speaker
[22:31:21][D][i2s_audio.speaker:129]: Stopped Speaker
[22:31:21][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE.......

BigBobbas, 2025-04-21:
this is possibly related to your network settings in HA
check that your local address in HA is correct and that the s3box and HA are on the same subnet

thank you
Rob

chrisdunnname, 2025-04-21:
I am getting this too and it seems to only occur with TTS on 2025.4.x. Media plays fine and it only started post upgrade. Sending an satellite announcement or any TTS action causes the symptoms. 
It appears related to recent TTS changes.  There are many issues floating around but the best is raised with HA core. 
[https://github.com/home-assistant/core/issues/142022](https://github.com/home-assistant/core/issues/142022)

zminde, 2025-04-23:
> I am getting this too and it seems to only occur with TTS on 2025.4.x. Media plays fine and it only started post upgrade. Sending an satellite announcement or any TTS action causes the symptoms. It appears related to recent TTS changes. There are many issues floating around but the best is raised with HA core. [home-assistant/core#142022](https://github.com/home-assistant/core/issues/142022)

Thank you for your attention and the information you provided.

chrisdunnname, 2025-05-22:
I have suggested some changes to refactor the code to support ESPHOME 2025.5.0 and these seem to have resolved this for me. I believe the updated i2s driver and some changes to way audio is handled solve this. It is working for me fine now.