BigBobbas/ESP32-S3-Box3-Custom-ESPHome #155: Suggestion: reboot when the touchscreen fails upon boot

igorsantos07, 2025-05-09:
Hi Bob!

I thought the touchscreen issue was specific to some other firmware I tried, but then I realized it happened here as well...

However, the guy on the other firmware had an interesting idea: forcing reboot when that happens. I wonder if you'd be interested in implementing it here as well, as a way to circumvent the issue for the time being :)

Look for `check_touchscreen_failure` at https://github.com/AlmostInteractive/ESP32-S3-Box-3-Voice-Assistant-Sensor-Dock/blob/master/esp32-s3-box-3.yaml

Thanks for such awesome documentation (and firmware as well, of course, but it would be a nightmare if it wasn't for the docs you wrote!!!!)

> re: #25, #127

youkorr, 2025-05-09:
Hi, I have your comment but it doesn't work, but in my case I noticed that if you use an 1850 battery in the box I have less problem with the touchscreen

igorsantos07, 2025-05-09:
@youkorr what doesn't work from my comment?

I've used that other firmware and the automatic reboot does happen, I've seen it a couple of times.

youkorr, 2025-05-09:
but for me it doesn't work, I tested it, it all comes from a problem with the power supply of the toushsreeen pullup

BigBobbas, 2025-05-09:
Sorry,  I'm away until next Friday with no access to anything useful. 
I will take a look when I'm back to reality. 👍

youkorr, 2025-05-09:
The author is to detect touchscreen errors, display an error message, and schedule an automatic reboot with a countdown. The user can cancel this reboot by pressing the upper left physical button, but it does not solve the touchscreen problem.

BigBobbas, 2025-05-10:
> The author is to detect touchscreen errors, display an error message, and schedule an automatic reboot with a countdown. The user can cancel this reboot by pressing the upper left physical button, but it does not solve the touchscreen problem.

Unfortunately still not a fix but at least this would automate the process. At the moment this would be better than what we do have. So certainly worth implementing to make it less of an issue for end users.

BigBobbas, 2025-05-10:
> Hi Bob!
>
> I thought the touchscreen issue was specific to some other firmware I tried, but then I realized it happened here as well...
>
> However, the guy on the other firmware had an interesting idea: [...]

Thank you so much for your support and coffee. New suggestions and ideas are always gratefully received. I will take a look at the touchscreen workaround when I am back home next week.

Cheers
Rob