BigBobbas/ESP32-S3-Box3-Custom-ESPHome #159: ESPHome 2025.5.0 S3 Box reboots on end of voice assistant announcement

chrisdunnname, 2025-05-21:
Since updating to ESPHome 2025.0 the S3 box reboots on the end of any conversation when it stops the mixing speaker. 

```
[01:13:50][D][voice_assistant:626]: Speech recognised as: "Hi."
[01:13:50][D][voice_assistant:456]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[01:13:50][D][voice_assistant:598]: Event Type: 5
[01:13:50][D][voice_assistant:631]: Intent started
[01:13:50][D][i2s_audio.microphone:467]: Task is stopping, attempting to unload the I2S audio driver
[01:13:50][D][i2s_audio.microphone:472]: Task is finished, freeing resources
[01:13:50][D][voice_assistant:598]: Event Type: 6
[01:13:50][D][voice_assistant:598]: Event Type: 7
[01:13:50][D][voice_assistant:656]: Response: "Hello from Home Assistant."
[01:13:50][D][voice_assistant:598]: Event Type: 8
[01:13:50][D][voice_assistant:678]: Response URL: "http://192.168.210.124:8123/api/tts_proxy/N8NZ5AQTlsBHZM689FoQhQ.flac"
[01:13:50][D][voice_assistant:456]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[01:13:50][D][voice_assistant:463]: Desired state set to STREAMING_RESPONSE
[01:13:50][D][media_player:074]: 'Media Player' - Setting
[01:13:50][D][media_player:081]:   Media URL: http://192.168.210.124:8123/api/tts_proxy/N8NZ5AQTlsBHZM689FoQhQ.flac
[01:13:50][D][media_player:087]:  Announcement: yes
[01:13:50][D][speaker_media_player:408]: State changed to ANNOUNCING
[01:13:50][D][voice_assistant:598]: Event Type: 2
[01:13:50][D][voice_assistant:697]: Assist Pipeline ended
[01:13:50][D][sensor:093]: 'S3 Temperature': Sending state 21.09534 °C with 2 decimals of accuracy
[01:13:50][D][sensor:093]: 'S3 Humidity': Sending state 60.02083 % with 2 decimals of accuracy
[01:13:50][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[01:13:50][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[01:13:50][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[01:13:50][D][ring_buffer:034]: Created ring buffer with size 9600
[01:13:50][D][speaker_mixer:312]: Starting speaker mixer
[01:13:50][D][speaker_mixer:320]: Started speaker mixer
[01:13:50][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[01:13:50][D][i2s_audio.speaker:117]: Starting Speaker
[01:13:50][D][i2s_audio.speaker:122]: Started Speaker
[01:13:52][D][speaker_media_player:408]: State changed to IDLE
[01:13:52][D][voice_assistant:329]: Announcement finished playing
[01:13:52][D][voice_assistant:456]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[01:13:52][D][voice_assistant:463]: Desired state set to RESPONSE_FINISHED
[01:13:52][D][micro_wake_word:360]: Starting wake word detection
[01:13:52][D][micro_wake_word:378]: State changed from STOPPED to STARTING
[01:13:52][D][voice_assistant:456]: State changed from RESPONSE_FINISHED to IDLE
[01:13:52][D][voice_assistant:463]: Desired state set to IDLE
[01:13:52][D][ring_buffer:034][mww]: Created ring buffer with size 3840
[01:13:52][D][micro_wake_word:261]: Inference task has started, attempting to allocate memory for buffers
[01:13:52][D][micro_wake_word:266]: Inference task is running
[01:13:52][D][micro_wake_word:378]: State changed from STARTING to DETECTING_WAKE_WORD
[01:13:52][D][i2s_audio.microphone:455]: Task has started, attempting to setup I2S audio driver
[01:13:52][D][speaker_mixer:325]: Stopping speaker mixer
WARNING esp32-s3-box-3-2e8c34 @ 192.168.210.49: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3-box-3-2e8c34 @ 192.168.210.49
WARNING Disconnected from API
INFO Successfully connected to esp32-s3-box-3-2e8c34 @ 192.168.210.49 in 0.009s
INFO Successful handshake with esp32-s3-box-3-2e8c34 @ 192.168.210.49 in 0.165s
[01:14:39][D][sensor:093]: 'WiFi db': Sending state -48.00000 dBm with 0 decimals of accuracy
[01:14:39][D][sensor:093]: 'WiFi Signal': Sending state 100.00000 % with 0 decimals of accuracy
```

I have a suspicion this is related to the new features that were just released.
[https://www.esphome.io/changelog/2025.5.0.html](https://www.esphome.io/changelog/2025.5.0.html)

I also noted that there is a new i2s_audio driver although I have tested the legacy mode and the issue still replicates:
[https://github.com/esphome/esphome/pull/8703](https://github.com/esphome/esphome/pull/8703)

I can see there is pending updates to the esphome wakewordvoiceassistants to support the new features which I suspect would resolve this:
[https://github.com/esphome/wake-word-voice-assistants/pull/110](https://github.com/esphome/wake-word-voice-assistants/pull/110)

It looks like a refactor of the way the voice assistant is configured as you can't assume the order of an interaction is wake word->stt->tts->done

I have been trying to handle how it ends the announcement but I haven't been able to mitigate this yet so I think the media_player and voice_assistant code may need to be updated similar to the pending update from esphome. I will wait until they release their changes.

I wanted to raise this as I expect others will start reporting it soon as it appears there is a breaking change in this release.

chrisdunnname, 2025-05-22:
I have created a pull request that addresses this and many other fixes and enhancements and aligns the code closer with the latest HA wake word voice assistant but keeps all the good stuff.
#160

ashimokawa, 2025-05-22:
@chrisdunnname 

For me also with your changes, the box reboots with 2025.5.0
I am using the Original S3 Box though with patches to the .yaml file to make it work.

Here is the stacktrace
```
[14:56:48][D][micro_wake_word:378]: State changed from STARTING to DETECTING_WAKE_WORD
[14:56:48][D][i2s_audio.microphone:455]: Task has started, attempting to setup I2S audio driver
[14:56:48][D][speaker_mixer:325]: Stopping speaker mixer
[14:56:49]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[14:56:49]
[14:56:49]Core  0 register dump:
[14:56:49]PC      : 0x403836d5  PS      : 0x00060030  A0      : 0x82085fa9  A1      : 0x3fce2d50  
WARNING Decoded 0x403836d5: xQueueSemaphoreTake at /home/xxx/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/queue.c:1657
[14:56:49]A2      : 0x00e800dc  A3      : 0xffffffff  A4      : 0x00060020  A5      : 0x00060023  
[14:56:49]A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x00000001  A9      : 0x00000000  
[14:56:49]A10     : 0x00060023  A11     : 0x00000000  A12     : 0xb33fffff  A13     : 0x00060023  
[14:56:49]A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
[14:56:49]EXCVADDR: 0x00e8011c  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[14:56:49]
[14:56:49]
[14:56:49]Backtrace: 0x403836d2:0x3fce2d50 0x42085fa6:0x3fce2d90 0x42012784:0x3fce2dc0 0x42012ac3:0x3fce2de0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x403836d2: xQueueSemaphoreTake at /home/xxxi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/queue.c:1657 (discriminator 1)
WARNING Decoded 0x42085fa6: i2s_channel_disable at /home/xxxi/.platformio/packages/framework-espidf/components/driver/i2s/i2s_common.c:1033 (discriminator 2)
WARNING Decoded 0x42012784: esphome::i2s_audio::I2SAudioMicrophone::stop_driver_() at /home/xxx/esp/esphome.s3box/.esphome/build/esp32-s3box/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:332
WARNING Decoded 0x42012ac3: esphome::i2s_audio::I2SAudioMicrophone::mic_task(void*) at /home/xxx/esp/esphome.s3box/.esphome/build/esp32-s3box/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:386

chrisdunnname, 2025-05-22:
Well that's frustrating.  I have two S3box3 both updated with TTS and va both working without crashing or getting stuck playing.

If you have the original box you could try changing the i2s_audio driver to prior version.
I think you just add use_legacy:true. It was mentioned in the pull request in the release notes. 

The adaptations i made came from the latest code from the esp home wake word voice assistants earlier today. You could test their new release and see if it crashes.

ashimokawa, 2025-05-22:
@chrisdunnname 
I tried that, it still reboots

aalaei, 2025-05-22:
I face the same problem (with or without legacy deriver) here is the stack trace:

> 0x4037bf2e: panic_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/panic.c:466
0x40382ea9: esp_system_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/port/esp_system_chip.c:84
0x40389e31: __assert_func at /config/.esphome/platformio/packages/framework-espidf/components/newlib/assert.c:81
0x403832f3: xQueueGenericSend at /config/.esphome/platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/queue.c:873
0x42031cf6: esphome::Mutex::unlock() at /config/.esphome/build/esp32-s3box-3/src/esphome/core/helpers.cpp:656
0x42013cdb: esphome::i2s_audio::I2SAudioComponent::unlock() at /config/.esphome/build/esp32-s3box-3/src/esphome/components/i2s_audio/microphone/../i2s_audio.h:92
0x42013cdb: esphome::i2s_audio::I2SAudioMicrophone::stop_driver_() at /config/.esphome/build/esp32-s3box-3/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:346
0x42013cdb: esphome::i2s_audio::I2SAudioMicrophone::stop_driver_() at /config/.esphome/build/esp32-s3box-3/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:305
0x42013fcf: esphome::i2s_audio::I2SAudioMicrophone::mic_task(void*) at /config/.esphome/build/esp32-s3box-3/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:386
0x4037fffe: esp_cpu_wait_for_intr at /config/.esphome/platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:121

ashimokawa, 2025-05-22:
@aalaei 
The original box? or the -3 ?
Did you use the config from the PR #160  ?

aalaei, 2025-05-22:
> [@aalaei](https://github.com/aalaei) The original box? or the -3 ? Did you use the config from the PR [#160](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/pull/160) ?

esp32-s3-box with sensor deck. Yes I used config from https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/pull/160

chrisdunnname, 2025-05-26:
I have had two s3-box-3 running with no problem on #160 (with HA 2025.5.3 and esphome 2025.5.0) since I shared it but If you are using the s3-box-3 and getting reboots you may find the updated config in #163 addresses this. It is a very significant refactor but I believe the problem with reboots may have been caused by the continued conversation behaviour and I think #163 addresses this better due to how it uses the voice phases configuration from the core repository. I don't have an original s3 box to test the other changes suggested in #161

ashimokawa, 2025-05-26:
@chrisdunnname 
Thanks, I tried your updated config with my changes on my original s3-box, and it still reboots

@aalaei 
How about you, does it fix things for you?

ashimokawa, 2025-05-26:
@chrisdunnname 
@aalaei 
For me this is fixed by decreasing buffer_duration from 500 to 100ms, like in the original firmware!

https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/pull/161/commits/ecb6163e262a0ac335f4eab2005a5727bf33249a

I also updated #161 accordingly

aalaei, 2025-05-26:
Thanks for the update! The issue still exists for me as well, though it seems a bit more stable now. I do get a response and it returns to main menu, but after a short time (like 100ms or so), it reboots.

aalaei, 2025-05-26:
Yes I can confirm, Decreasing buffer_duration from 500 to 100ms fixes the problem

chrisdunnname, 2025-05-26:
Doh!! My apologies. I have my devices running at 100ms but when I merged the code back i missed that line. 

I have updated #160 and #163.

chrisdunnname, 2025-06-06:
comparing the original to #163 I don't think it is just a parameter. Based on the change made to i2s microphone I think it is likely a timing issue with the microphone/wake word starting when it isn't ready or something like that.
I suspect it is in the on_state function of media player, on_tts_end function of voice_assistant or start_voice_assistant script. All other functions are now almost identical, except there are 2 i2c modules with the radar and two buses. I am thinking we may need to have some logger functions throughout to make it easier to diagnose the function where it falls over.

youkorr, 2025-06-07:
When I said original firmware, it's not Rob's custom firmware but that of home assistant and it works very well with the latest esphome version, but I think you're right that they must have modified the media player part because my last configuration in lvlgl that I just tested the multimedia player starts and stops quickly and no sound comes out but the text response part works except the mp3 which works, I'll look at this tomorrow

BigBobbas, 2025-06-07:
ok,  I have tested with @chrisdunnname config https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/cda36ede14271c42303b6593e2a24d0316eff659/s3b.yaml
I'm not getting a restart or crashing. I'm not getting any audio for responses and the wake sound plays after the action completes, there are quite a few errors in the logs. This is my first test, so will have a look through the logs and see what i can find.

youkorr, 2025-06-07:
Hi sorry I didn't also take care of the esp32s3box3 I was busy with the programming of the M5Stack Tab5 IoT Development Kit (ESP32-P4) I will take care of it in the afternoon to see where the problem comes from

youkorr, 2025-06-07:
Hello everyone, after several tests the problem comes from the media player, I solved it, I'll give you what I did.Virtual speakers no longer work in the latest update, it creates crashes,The Virtual Speakers no longer works in the latest update, it creates crashes, so I simplified it and it works very well without rebooting, you just have to modify Rob's custom with this simplified media player
```

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO45
    i2s_bclk_pin: GPIO17
    i2s_mclk_pin: GPIO2

audio_adc:
  - platform: es7210
    id: es7210_adc
    bits_per_sample: 16bit
    sample_rate: 16000
    i2c_id: bus_a

audio_dac:
  - platform: es8311
    id: es8311_dac
    bits_per_sample: 16bit
    sample_rate: 48000
    i2c_id: bus_a

microphone:
  - platform: i2s_audio
    id: box_mic
    sample_rate: 16000
    i2s_din_pin: GPIO16
    bits_per_sample: 16bit
    adc_type: external

speaker:
  - platform: i2s_audio
    id: box_speaker
    i2s_dout_pin: GPIO15
    dac_type: external
    sample_rate: 48000
    bits_per_sample: 16bit
    channel: left
    audio_dac: es8311_dac
    buffer_duration: 100ms

media_player:
  - platform: speaker
    name: Media Player
    id: speaker_media_player
    volume_increment: 0.05
    volume_min: 0.5
    volume_max: 0.8
    task_stack_in_psram: true
    announcement_pipeline:
      speaker: box_speaker
      format: FLAC
      sample_rate: 48000
      num_channels: 1  # S3 Box only has one output channel

    files:
      - id: timer_finished_sound
        file: https://github.com/esphome/home-assistant-voice-pe/raw/dev/sounds/timer_finished.flac
    on_announcement:
      # Stop the wake word (mWW or VA) if the mic is capturing
      - if:
          condition:
            - microphone.is_capturing:
          then:
            - script.execute: stop_wake_word
            # Ensure VA stops before moving on
            - if:
                condition:
                  - lambda: return id(wake_word_engine_location).state == "In Home Assistant";
                then:
                  - wait_until:
                      - not:
                          voice_assistant.is_running:
      # Since VA isn't running, this is user-intiated media playback. Draw the mute display
      - if:
          condition:
            not:
              voice_assistant.is_running:
          then:
            - lambda: id(voice_assistant_phase) = ${voice_assist_muted_phase_id};
            - script.execute: draw_display
    on_idle:
      # Since VA isn't running, this is the end of user-intiated media playback. Restart the wake word.
      - if:
          condition:
            not:
              voice_assistant.is_running:
          then:
            - script.execute: start_wake_word
            - script.execute: set_idle_or_mute_phase
            - script.execute: draw_display

    on_play:
      - lambda: ESP_LOGD("media_player", "on_play");
    on_pause:
      - lambda: ESP_LOGD("media_player", "on_pause")
      ```

youkorr, 2025-06-07:
[image]

igorsantos07, 2025-06-07:
@youkorr I dropped from a parachute on this issue, trying to report/help out with some issues I've seen in my S3... But I'm not able to "merge" your YAML into the one I'm using, which is a slightly modified version from this repo - but with no changes whatsoever to media stuff.

There are missing scripts (`start_wake_word`, `draw_display`, `set_idle_or_mute_phase`), `voice_assistant_phase` doesn't exist either, and when I tried to replace those or comment them, I got greeted by some extra errors - among other things, `media_mixing_input` doesn't exist anymore and [this](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b.yaml#L1195) section breaks.

Long story short, I wouldn't say "just have to modify Rob's with this" makes sense 😂

youkorr, 2025-06-07:
indeed the latest version of esphome has created problems in the media player but you can make it work as before, if you have made a backup of the old version of esphome

chrisdunnname, 2025-06-08:
Thanks @youkorr 
I removed the virtual speaker and I can confirm it resolves the problem with the voice assistant. I had to also add a delay to on_idle otherwise when casting media to the media player there was a timing issue where the speaker could go idle after the wake word started causing the same problem.

For Rob's original config [https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-1-2025-5-2/s3b.yaml](https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-1-2025-5-2/s3b.yaml) this has the fixes for 2025.5.0 and changes to bypass the virtual speaker. It is #160 with the additional changes. I haven't tested this one.

To more closely align with the latest esphome wake word voice assistant (which is admittedly much harder to merge) you can try [https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-2-2025-5-2/s3b.yaml](https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-2-2025-5-2/s3b.yaml) which is #163 with the additional changes. This works fine for me now on 2025.5.2.

I didn't update the existing pull requests as I can only hope this is a workaround until it is fixed.

youkorr, 2025-06-08:
I just solved it, it works perfectly
```
i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin:
      number: GPIO45
      ignore_strapping_warning: true
    i2s_bclk_pin: GPIO17
    i2s_mclk_pin: GPIO2

speaker:
  - id: i2s_audio_speaker
    platform: i2s_audio
    i2s_audio_id: i2s_audio_bus
    i2s_dout_pin: GPIO15
    dac_type: external
    sample_rate: 48000
    bits_per_sample: 16bit
    channel: left
    audio_dac: es8311_dac
    buffer_duration: 100ms
    
  # Virtual speakers to combine the announcement and media streams together into one output
  - id: mixing_speaker
    platform: mixer
    output_speaker: i2s_audio_speaker
    source_speakers:
      - id: announcement_mixing_input
      - id: media_mixing_input
  
  # Virtual speakers to resample each pipelines' audio, if necessary, as the mixer speaker requires the same sample rate
  - platform: resampler
    id: announcement_resampling_speaker
    output_speaker: announcement_mixing_input
    sample_rate: 48000

  - platform: resampler
    id: media_resampling_speaker
    output_speaker: media_mixing_input
    sample_rate: 48000

audio_adc:
  - platform: es7210
    id: es7210_adc
    i2c_id: bus_a  
    bits_per_sample: 16bit
    sample_rate: 16000

audio_dac:
  - platform: es8311
    id: es8311_dac
    i2c_id: bus_a
    bits_per_sample: 16bit 
    sample_rate: 48000

microphone:
  - platform: i2s_audio
    id: box_mic
    sample_rate: 16000
    i2s_din_pin: GPIO16
    adc_type: external
    pdm: false
    bits_per_sample: 16bit

media_player:
  - platform: speaker
    name: box3
    id: speaker_media_player
    volume_min: 0.5
    volume_max: 0.8
    task_stack_in_psram: true
    
    on_state:
      then:
        - if:
            condition:
              media_player.is_playing:
            then:
              - script.execute: stop_wake_word
        - if:
            condition:
              and:
                - switch.is_off: timer_ringing
                - not:
                    voice_assistant.is_running:
                - not:
                    media_player.is_announcing:
            then:
              - mixer_speaker.apply_ducking:
                  id: media_mixing_input
                  decibel_reduction: 0
                  duration: 1.0s                   
    on_play:
      - wait_until:
          - lambda: return id(global_is_assisting) == false;
    on_announcement:
      - if:
          condition:
            - microphone.is_capturing:
          then:
            - script.execute: stop_wake_word
            - if:
                condition:
                  - lambda: return id(wake_word_engine_location).state == "In Home Assistant";
                then:
                  - wait_until:
                      - not:
                          voice_assistant.is_running:
      - mixer_speaker.apply_ducking:
          id: media_mixing_input
          decibel_reduction: 20
          duration: 0.0s
      - if:
          condition:
            not:
              voice_assistant.is_running:
          then:
            - lambda: id(voice_assistant_phase) = ${voice_assist_muted_phase_id};
    on_idle:
      - if:
          condition:
            not:
              voice_assistant.is_running:
          then:
            - script.execute: start_wake_word          
       
    # OBLIGATOIRE en ESPHome 2025.5.x
    media_pipeline:
      speaker: media_resampling_speaker
      num_channels: 2
      sample_rate: 48000
      format: FLAC
        
    announcement_pipeline:
      speaker: announcement_resampling_speaker
      num_channels: 1  
      sample_rate: 48000
      format: FLAC

micro_wake_word: 
  id: mww
  vad:  
  models: 
    - model: ${micro_wake_word_model_3}
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
 
voice_assistant:
  id: va
  micro_wake_word: mww
  microphone: box_mic
  # FIX PRINCIPAL : Pointer directement vers le speaker announcement au lieu du media_player
  speaker: announcement_resampling_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 4.0
  
  on_start:
    - light.turn_on: led    
    - lvgl.page.show: page_lcars
  on_stt_end:
    - delay: 100ms
    - lvgl.page.show: page_lcars
    - text_sensor.template.publish:
        id: text_request
        state: !lambda return x;
  on_stt_vad_start:
    - light.turn_on: led
  on_stt_vad_end:
    - light.turn_on: led
  on_tts_start:
    - text_sensor.template.publish:
        id: text_response
        state: !lambda return x;  
  on_idle:
    - delay: 5s
    - lvgl.page.show: page_home
    - component.update: my_display       
  on_end:
    then:
      - voice_assistant.stop
      - wait_until:
          not:
            voice_assistant.is_running:
      - delay: 500ms
      - micro_wake_word.start
      - ```

youkorr, 2025-06-08:
> Thanks [@youkorr](https://github.com/youkorr) I removed the virtual speaker and I can confirm it resolves the problem with the voice assistant. I had to also add a delay to on_idle otherwise when [...]

test my solution it works perfectly with the latest update of esphome the Virtual speakers works

chrisdunnname, 2025-06-08:
that didn't resolve it for me. 
This was my test yaml [https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-2-youkorr/s3b.yaml](https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-2-youkorr/s3b.yaml) . Casting media is fine but it reboots on the end of voice announcements. Removing the virtual speakers works fine as earlier.

chrisdunnname, 2025-06-08:
I think there is a timing issue. I noticed to address the casting of media causing the reboot I needed to add a slight delay to the media_player on_idle which seems to ensure everything happens in order. I noticed the speaker was changing to idle after the first two scripts had run and this was causing the issue. So by adding the 1s delay it never fails. 
```
    on_idle:
      # Since VA isn't running, this is the end of user-intiated media playback. Restart the wake word.
      - if:
          condition:
            not:
              voice_assistant.is_running:
          then:
            - delay: 1s
            - script.execute: start_wake_word
            - script.execute: set_idle_or_mute_phase
            - script.execute: draw_display
```
I suspect that there is something similar occurring on the end of announcements and if the microphone/speaker are in the wrong state then it causes the microphone i2s error.

chrisdunnname, 2025-06-08:
I have managed to resolve the problem without needing @youkorr workaround. Most of the prior 2025.5.x changes were already in pull requests #160 and #163. The root cause of the new issue in 2025.5.2 appears to be related to the microphone changes. 

If the microphone attempts to start at the wrong time it causes an error because I assume i2s_audio be in the wrong state. My last post was close. The problem in the existing YAML is that at the end of playing media or an announcement we were checking the media player or voice assistant before we started the wake word. Just because the media player was not playing did not mean the 12s_audio module was not in use. I modified the on_idle of the media_player and the on_end of the voice assistant to confirm the i2s_audio had stopped before starting the wake word. This has resolved all the reboots at the end of audio.

I updated the pull requests for both #160 and #163 with the changes.

chrisdunnname, 2025-06-09:
It shouldn't reboot on media playback in #163 now only if the va tries to continue conversation does the mic error but the version i posted earlier which is same without the virtual speakers works fine.
I can see the issue rob mentioned with the wake word location select on device.

chrisdunnname, 2025-06-09:
The QR code for wifi can be viewed in #163 if you tap on the wifi page anywhere, tapping again returns to the wifi page. The wake word sound timing is odd as it seems to match the esphome wake word voice assistant code except for the external output option.
I updated #163 and the [non virtual speaker version](https://github.com/chrisdunnname/ESP32-S3-Box3-Custom-ESPHome/blob/patch-2-2025-5-2/s3b.yaml) to correct the wake word location displaying on the device. It wasn't the id but the updated select was not publishing the status to the template used for displaying.

youkorr, 2025-06-09:
I'll give you my code which works without the microphone or audio playback errors as I had from the beginning. I think the ducking for this esphome version is broken but there will always be a solution for us to find it.

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin:
      number: GPIO45
      ignore_strapping_warning: true
    i2s_bclk_pin: GPIO17
    i2s_mclk_pin: GPIO2

speaker:
  - id: i2s_audio_speaker
    platform: i2s_audio
    i2s_audio_id: i2s_audio_bus
    i2s_dout_pin: GPIO15
    dac_type: external
    sample_rate: 48000
    bits_per_sample: 16bit
    channel: left
    audio_dac: es8311_dac
    buffer_duration: 100ms
    
  # Virtual speakers to combine the announcement and media streams together into one output
  - id: mixing_speaker
    platform: mixer
    output_speaker: i2s_audio_speaker
    source_speakers:
      - id: announcement_mixing_input
      - id: media_mixing_input
  
  # Virtual speakers to resample each pipelines' audio, if necessary, as the mixer speaker requires the same sample rate
  - platform: resampler
    id: announcement_resampling_speaker
    output_speaker: announcement_mixing_input
    sample_rate: 48000

  - platform: resampler
    id: media_resampling_speaker
    output_speaker: media_mixing_input
    sample_rate: 48000

audio_adc:
  - platform: es7210
    id: es7210_adc
    i2c_id: bus_a  
    bits_per_sample: 16bit
    sample_rate: 16000

audio_dac:
  - platform: es8311
    id: es8311_dac
    i2c_id: bus_a
    bits_per_sample: 16bit 
    sample_rate: 48000

microphone:
  - platform: i2s_audio
    id: box_mic
    sample_rate: 16000
    i2s_din_pin: GPIO16
    adc_type: external
    bits_per_sample: 16bit
    channel: stereo

media_player:
  - platform: speaker
    name: box3
    id: speaker_media_player
    volume_min: 0.5
    volume_max: 0.8
    task_stack_in_psram: false
    on_state:
      then:
        - if:
            condition:
              media_player.is_playing:
            then:
              - micro_wake_word.stop:
              - voice_assistant.stop:
        - if:
            condition:
              media_player.is_idle:
            then:
              - delay: 2s
              - micro_wake_word.start:
                
    media_pipeline:
      speaker: media_resampling_speaker
      num_channels: 2
      sample_rate: 48000
      
    announcement_pipeline:
      speaker: announcement_resampling_speaker
      num_channels: 1  
      sample_rate: 48000

    files:
      - id: wake_sound_file
        file: https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/sounds/awake.flac

micro_wake_word: 
  microphone: box_mic
  models: 
    - model: ${micro_wake_word_model_3}
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
 
voice_assistant:
  id: va
  microphone: 
    microphone: box_mic
    gain_factor: 4
    channels: 1
  speaker: announcement_resampling_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  
  on_start:
    - light.turn_on: led    
    - lvgl.page.show: page_lcars
  on_stt_end:
    - delay: 100ms
    - lvgl.page.show: page_lcars
    - text_sensor.template.publish:
        id: text_request
        state: !lambda return x;
  on_stt_vad_start:
    - light.turn_on: led
  on_stt_vad_end:
    - light.turn_on: led
  on_tts_start:
    - text_sensor.template.publish:
        id: text_response
        state: !lambda return x;  
  on_idle:
    - delay: 5s
    - lvgl.page.show: page_home
    - component.update: my_display       
  on_end:
    then:
      - voice_assistant.stop
      - wait_until:
          not:
            voice_assistant.is_running:
      - delay: 500ms
      - micro_wake_word.start:
      - if:
          condition:
            lambda: "return !id(timer_active);"
          then:
            - delay: 6s
            - lvgl.page.show: page_home  

  on_timer_started:   
    - globals.set:
        id: timer_active
        value: 'true'
    - script.execute: timer_ending
    - lvgl.page.show: page_timer

  
  on_timer_cancelled:
    - globals.set:
        id: timer_active
        value: 'false'
    - lambda: id(time_remaining_0).publish_state ("0:00:00"); 
    - component.update: my_display
  on_timer_updated:
    - component.update: my_display
  on_timer_tick:
    - lambda: |-
        int seconds = timers[0].seconds_left;
        int hours = seconds / 3600;
        int minutes = (seconds % 3600) / 60;
        seconds %= 60;
        id(time_remaining_0).publish_state(str_sprintf("%d:%02d:%02d", hours, minutes, seconds));
        id(countdown_time) = timers[0].seconds_left;
        lv_arc_set_value(id(arc_timer), id(countdown_time) % 60);
        lv_meter_set_indicator_value(id(timer_meter), id(needle_indicator), id(countdown_time) % 60);
        char buf[10];
        sprintf(buf, "%d\nsec", id(countdown_time));
        lv_label_set_text(id(timer_label), buf);
    - component.update: my_display

  on_timer_finished:
    - globals.set:
        id: timer_active
        value: 'false'
    - script.stop: timer_ending
    - lvgl.page.show: page_timer_finished
    - component.update: my_display
    - if:
        condition:
          - switch.is_on: output_audio
        then:
          - switch.turn_on: timer_ringing
          - script.execute: timer_end_alert
        else:
          - if:
              condition:
                - switch.is_off: output_audio
              then:
                - switch.turn_on: timer_ringing
                - script.execute: timer_end_alert

    - component.update: my_display

youkorr, 2025-06-09:
Here is the link that helped me

https://github.com/esphome/esphome/pull/8645