BigBobbas/ESP32-S3-Box3-Custom-ESPHome #162: Cannot get .yaml install to complete

budpi, 2025-05-23:
After adoption in home assistant, esphome builder, install just freezes at this point.

INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...

tylxr59, 2025-06-06:
@budpi Were you able to resolve this?

kicker10BOG, 2025-08-02:
I'm having this same issue. Except, I am on 2025.7.4 and I see the following log in the ESP Home Builder logs:
```
 INFO Process exited with return code -9
```

BigBobbas, 2025-08-03:
@kicker10BOG  could you post your compile log please.

What hardware are you running ESPHome on and how much RAM does it have ? also how is ESPHome installed? eg. HA Add-on , Docker, etc

The issue won't have anything to do with external media player, as ESPHome doesn't know if the defined entity exists or not and just sends commands to it. If you are not using an external media player, you can leave the substitution unchanged and just don't enable the feature in the settings.

cheers
Rob

kicker10BOG, 2025-08-03:
Before trying that, I rebooted the machine and gave it another try because I noticed quite high resource usage, This time, I got some more helpful logs. 
[logs_esp32-s3box-3_compile (after reboot).txt](https://github.com/user-attachments/files/21564059/logs_esp32-s3box-3_compile.after.reboot.txt)

I had tried what you suggested before and it complained about the wifi qr code. 
[logs_esp32-s3box-3_compile (qr).txt](https://github.com/user-attachments/files/21564060/logs_esp32-s3box-3_compile.qr.txt)

So, I tried again after commenting out all of the qr code lines and pages. 
[logs_esp32-s3box-3_compile (no qr).txt](https://github.com/user-attachments/files/21564074/logs_esp32-s3box-3_compile.no.qr.txt)

It appears to be an issue with git. This is the error in both attempts that made it past just a syntax check. 
```
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:304 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:311 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:607 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)
``` 

Thank you for looking into this.

kicker10BOG, 2025-08-03:
That fixed the install. Thanks again! It's not responding to commands yet, but I'll get that working later. It actually worked great for me a year or two ago, shortly after the initial launch, but then it started too quickly selecting a command, sometimes before I even tried to say it. That's not the ESP32 code though, I'm pretty sure of that. I think it is probably a setting in vosk or something. Yes, I'm using Vosk because I want custom commands like "nox" and "lumos" in different parts of the house. That was the dream anyways. And my speech isn't great, so having a limited number of commands it could listen for seemed to help until that issue arose.