BigBobbas/ESP32-S3-Box3-Custom-ESPHome #164: Can't display simplified Chinese characters.

zminde, 2025-06-01:
Can't display simplified Chinese characters, the same problem exists in esp32 original configuration. Is this a solvable problem?

tylxr59, 2025-06-06:
Theoretically yes, this is solvable. You'd just need a font that supports Chinese characters and add all characters to the font_glyphs section.

zminde, 2025-06-28:
> Theoretically yes, this is solvable. You'd just need a font that supports Chinese characters and add all characters to the font_glyphs section.

Thank you for your reply and suggestions. I have tried the method you mentioned with the help of chatgpt, but the modified configuration file cannot pass the verification and an error occurs when flashing the firmware. Due to my lack of knowledge and experience in this area, I have not been able to solve this problem. In addition, I don't understand why such a simple problem as the display of Chinese characters has not been solved by the official firmware for such a long time.