BigBobbas/ESP32-S3-Box3-Custom-ESPHome #169: Can only use "okay nabu" one time, and some erry about a api update,

JyskMakerDK, 2025-07-23:
Hey.
Grate setup but with some compiling errors with the news update, and i can only say Okay nabo ones, and then its loos the "microfone" and there is no sound comming from it and no text if that is turned on.

I dont know how to fix it, not that into yaml. Sorry

BigBobbas, 2025-07-26:
Hi can you paste the actual error please.
Thanks

mbeilste, 2025-12-14:
I did some more digging and the single wake issue was in HA with the voice assistant.  It took several tries to get it setup and it is now working correctly.  I believe this issue can be closed and is related to the HA side, not the firmware in the ESP32.  I do know that the mic driver error causes some strange/awkward timing issues when talking with the device.  If you don't pause to let the error clear, the device will not catch what you say.  The stock HA firmware does not throw this error and you can rattle off commands much faster and more reliably.  Just load up the wireless log and give commands to see the errors in real time.  If you talk too quick, it will miss your commands since the error has not yet cleared.

galsom, 2026-01-05:
Thank you for the feedback. I'll continue to play around as well.