BigBobbas/ESP32-S3-Box3-Custom-ESPHome #170: Unusable custom firmware with latest ESPHome version (2025.7.3)

amarok45, 2025-07-25:
I'm running Home Assistant, which proposed to update ESPHome to 2025.7.3 but this version is no longer compatible with ESP-IDF 4.4.8 (it uses instead 5.3.2) which prevents your firmware from being installed on my ESP32-S3-Box-3.

Is there a way to make it work? Do I have to modify some data in the yaml file?

Thank you for your answer.

BigBobbas, 2025-07-26:
can you try one of the versions in the dev branch. Either s3b.yaml if you have sensor dock etc or s3b_n0_sensors.yaml if you don't.

thanks
Rob

amarok45, 2025-07-26:
Thank you again.

Now, when trying to install the configuration, compiling fails and I get this error message:

xtensa-esp-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/esp32-s3box-3/src/main.cpp.o] Error 1

Sorry for so many problems, but whereas my device worked fine before upgrading HA, now it's a nightmare...

Best regards

BigBobbas, 2025-07-26:
this will because the machine you are compiling on is low in ram.

FirstRulez, 2025-07-27:
> Thank you again.
>
> Now, when trying to install the configuration, compiling fails and I get this error message:
>
> xtensa-esp-elf-g++: fatal error: Killed signal terminated program cc1plus compilation [...]

I also had this issue building many esphome devices, and found that sometimes if you restart esphome add-on just before doing the compile it will work (assuming you are not able to make more RAM available to the build machine. I thankfully upgraded recently so resolved by allocating more ram to my virtual HASS machine.

JyskMakerDK, 2025-07-30:
Hey
Get this when i try to compile the s3b.yaml from the dev sectien.

```

Dependency Graph
|-- qr-code-generator-library @ 1.7.0
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.4
Compiling .pioenvs/esp32-s3box-3/src/main.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:9,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
```