BigBobbas/ESP32-S3-Box3-Custom-ESPHome #177: Failed to resolve component 'esp_adc_cal' required by component   'esp_peripherals': unknown name.

blind-coder, 2025-09-02:
Hello

I'm trying to rebuild your firmware due to the ESPHome security issue, but getting this error with the dev branch on latest HA/ESPHome:

```
INFO ESPHome 2025.8.2
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Detected timezone 'Europe/Berlin'
WARNING The 'publish_initial_state' option has been replaced by 'trigger_on_initial_state' and will be removed in a future release
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
Processing 3 dependencies:
[1/3] espressif/mdns (1.8.2)
[2/3] zorxx/multipart-parser (1.0.1)
[3/3] idf (5.4.2)
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:328 (message):
  Failed to resolve component 'esp_adc_cal' required by component
  'esp_peripherals': unknown name.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:371 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:378 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:675 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:718 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 7.04 seconds ==========================
```

I've tried search for this error far and wide and all I see is that esp_adc_cal is deprecated but no instructions how to remove it as a dependency from a YAML file, only from a CMake file which seems inaccessible?

Kelerion-UK, 2025-09-02:
Have you tried doing a "Clean Build".

blind-coder, 2025-09-02:
Yes, I clicked "Clean Build Files" several times.

blind-coder, 2025-09-02:
Did "Clean Build Files" a few times in a row, but still:

```
INFO ESPHome 2025.8.2
INFO Reading configuration /config/esphome/esp32-s3box-3.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/9100/head
INFO Detected timezone 'Europe/Berlin'
WARNING The 'publish_initial_state' option has been replaced by 'trigger_on_initial_state' and will be removed in a future release
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-s3box-3 (board: esp32s3box; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
Library Manager: Installing wjtje/qr-code-generator-library @ ^1.7.0
INFO Installing wjtje/qr-code-generator-library @ ^1.7.0
Unpacking  [####################################]  100%
Library Manager: qr-code-generator-library@1.7.0 has been installed!
INFO qr-code-generator-library@1.7.0 has been installed!
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
Library Manager: Installing esphome/esp-audio-libs @ 1.1.4
INFO Installing esphome/esp-audio-libs @ 1.1.4
Unpacking  [####################################]  100%
Library Manager: esp-audio-libs@1.1.4 has been installed!
INFO esp-audio-libs@1.1.4 has been installed!
INFO Package configuration completed successfully
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 3 dependencies:
[1/3] espressif/mdns (1.8.2)
[2/3] zorxx/multipart-parser (1.0.1)
[3/3] idf (5.4.2)
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:328 (message):
  Failed to resolve component 'esp_adc_cal' required by component
  'esp_peripherals': unknown name.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:371 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:378 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:675 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:718 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 11.45 seconds =========================
```

blind-coder, 2025-09-02:
How do I install the beta version? Is it "ESPHome Device Builder (dev) 2025.9.0-dev20250902" in the HomeAssistant AddOns?

closed 2025-09-02 by blind-coder